wmbusmeters/wmbusmeters #711: HA-Addon mqtt discovery is looking for the wrong file

autega, 2022-12-01:
When using the addon with mqtt discovery on the script ist looking for the wrong file.
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
MQTT Discovery ...
 Adding meter: Sensostar_Heatmeter ...
  File /config/wmbusmeters/etc/mqtt_discovery/sensostar:MAIN:mbus.json not found.
MQTT Discovery cleanup...
`

Here  my working config:
`data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: |-
  loglevel=debug
  device=MAIN=/dev/ttyUSB0:mbus:2400
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=Sensostar_Heatmeter
    driver=sensostar:MAIN:mbus
    pollinterval=5s
    id=21950207
    #id=4
    key=NOKEY
mqtt: {}
`

autega, 2022-12-01:
Solution might be to change this line:
https://github.com/weetmuts/wmbusmeters/blob/9a8cd21628d4c01d3d2822218a616cba61fb8c8e/ha-addon/mqtt_discovery.sh#L49

akanegally, 2023-07-15:
It seems that mqtt discovery doesn't work.
I have the same problem.
Here is the log :
MQTT Discovery ...
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
[23:01:44] INFO:  Adding meter: cu_compteur_eau ...
[23:01:44] INFO:   File /config/wmbusmeters/etc/mqtt_discovery/apator172.json not found.
[23:01:44] INFO: MQTT Discovery cleanup...

Seger85, 2023-07-16:
I have the same problem, I just can't get any further.

spectrality-github, 2023-09-28:
I fixed that in my private fork with additional line in mqtt_discovery.sh
`aryKV["driver"]=${aryKV["driver"]%%:*}`
https://github.com/spectrality-github/wmbusmeters-ha-addon/blob/9351b966d42ccfe838aa140dc314b67b0fb45570/wmbusmeters-ha-addon-edge/mqtt_discovery.sh#L42 
As colons are required in driver name (`amiplus:MAIN:mbus`) if you using Mbus devices and not Wireless Mbus, but should be removed to match discovery template files.

But, still need to create new driver and discovery template for my NZR electricity meters as amiplus driver is not fully matching.

Also I'm interested in Mbus TCP support in wmbusmeters, as I have my Mbus USB adapter connected to OpenWrt device with `ser2net `running.  
Currently trying to workaround lack of Mbus TCP support by adding and starting `socat `inside wmbusmeters-ha-addon container. 
`socat -d -d pty,group-late=tty,link=/root/ttyMBUS0,mode=660,rawer,echo=0,b2400,waitslave,ignoreeof tcp:192.168.x.x:2003 &`
And using `/root/ttyMBUS0` as device in wmbusmeters config.
But `socat `is tend to stop if no data is transferred and whole container need to be restarted to restore connectivity.
It whoudbe nice if somebody implement native Mbus TCP support in wmbusmeters as it is done in `libmbus`.
`mbus-tcp-request-data`
http://www.rscada.se/libmbus/

BIBOLV, 2023-11-04:
Fixed with https://github.com/wmbusmeters/wmbusmeters-ha-addon/commit/1e66d6876e30e0631b43d9c9652e688c0d47d3e0

closed 2023-11-04 by BIBOLV