wmbusmeters/wmbusmeters #712: Support for GWF/Honeywell/Elster gas meter BK-G4-A with BGZcoder

marcelstoer, 2022-12-01:
This is a (more specific) follow-up to #703.

I got this gas meter installed

[image]

Apart from a [product description](https://products.gwf.ch/wp-content/uploads/2018/11/BGZcoder_MP_AE5_EPd30115.pdf) and some [installation instructions](https://gwf.ch/wpfd_file/rcmsplit_montageanleitung-auf-bgzcoder-mp-ae5_bad3030-3/) I haven't found any information online yet. Inquiry with GWF is pending.

It appears to be the same or a similar device to what @awor asked about in #435.

The 'device type' byte in the A-field incorrectly contains 0x06 instead of 0x03; likely a misconfiguration by the utility provider.

```
pi@white-rose:~ $ wmbusmeters/build/wmbusmeters --logtelegrams /dev/ttyUSB0:im871a:t1,c1
Started config im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 875103bf
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x00
                device: im871a[00103852]
                  rssi: -88 dBm
                driver: unknown!
telegram=|_1944E61EBF0351870006B2010654168A3F737CE223319854BE6E|+7
Received telegram from: 875103bf
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x00
                device: im871a[00103852]
                  rssi: -89 dBm
                driver: unknown!
telegram=|_1944E61EBF0351870006B2110654168A285C354902E2418FA76C|+16
```
@weetmuts @BIBOLV is there anything you can derive from this information towards supporting this family of devices?

BIBOLV, 2022-12-02:
Based on [this](https://loewener.dk/wp-content/files/3871_20.11.2007_gwf_radiomodul_rcmpi_2_dk.pdf), this module uses PRIOS protocol similar as izar devices. So it makes things more complex probably. It`s doubtful but can you check if izar driver gives correct reading for your meter?

marcelstoer, 2022-12-02:
Oh, good catch! I will certainly give the izar driver a shot once I get the AES key from the gas provider.

marcelstoer, 2022-12-09:
@ZeWaren (you appear to be the original contributor of the PRIOS implementation) is there anything obvious you conclude from the telegram and the output above?

marcelstoer, 2023-03-05:
Yes bot, I know. It's the nature of "new driver" issues that they remain open until implemented. I'd like to help implementing this but I don't know how.

thecem, 2023-04-01:
@marcelstoer did you get your sensus water meter to work, I have a similar version. But not able to receive a telegram.

marcelstoer, 2023-04-01:
@thecem this issue is about reading / decoding the gas meter. My Sensus meter was (briefly) reported here https://github.com/wmbusmeters/wmbusmeters/issues/703#issuecomment-1330546866. It's encrypted but I didn't get the key yet.

marcelstoer, 2023-04-17:
For anyone following along, the offset between the reported and the effective value does not appear to be constant.

| reported | effective |
| --- | --- |
| 11.092 | 11095.205 |
| 11.367 | 11377.347 |

I might amend this table over time. Maybe a pattern emerges.

marcelstoer, 2023-07-06:
Confirmed the meter's communication device ran out of batteries. Utility needs to replace it first.

marcelstoer, 2023-09-06:
Is there a better way to keep this open for others to see other than commenting occasionally to fight the bot?

closed 2024-02-06 by marcelstoer