wmbusmeters/wmbusmeters #713: Cannot find driver for mfct: (FLM) Unknown (0x198d) type: Heat meter (0x04) ver: 0x01 and type: Water meter (0x07) ver: 0x01

zdzichu, 2022-12-02:
Hi,

I have im871a dongle for reading meters in my apartment building. Unfortunately they all came as unknown:
```
[2022-11-23_14:41:53] (meter) licznik82007704: meter detection could not find driver for id: 82007704 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01
[2022-11-23_14:41:53] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-11-23_14:41:53] (meter) to add support for this unknown mfct,media,version combination
```
My reader is `(im871a/im170a) info: type=im871a mode=collector firmware=16 hci=01 uid=00003c25`

I'm receiving around 6 water meters as: `(meter) licznik82007410: meter detection could not find driver for id: 82007410 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01`
And around 3 heat meters: `(meter) licznik72012567: meter detection could not find driver for id: 72012567 mfct: (FLM) Unknown (0x198d) type: Heat meter (0x04) ver: 0x01`

Radio addon on water meter have a pretty generic name "FLOW meters pro" with "radio module OMS", I wasn't able to google anything. 
Similarly, the heat meter have a transmitter with "FLOW meters PULSE". 
Serial number printed on transmitters match what I receive in wmbusmeters. I've blurred apartment numbers on meters.

[image]
[image]
[image]

```
(meter) licznik72012563(2) auto  handling telegram from 72012563
(meter) licznik72012563 72012563 "7E448D196325017201047A150070A51224D17E0390858ADBDA5950FA99A3FC9738BD74601FF0173E785553DB617712D51BE9D232C933FC8580DA61BEE048DCE2095D4182211BEA6C84DDB9C6D9FCDFAC97BA74D7A70CDC2673599F77B8418FA832613A91EA5EBB819CECAA8A021FD2764C0C21934AEF2AFA56797B063E0961"
(wmbus) parseDLL @0 127
(telegram) DLL L=7e C=44 (from meter SND_NR) M=198d (FLM) A=72012563 VER=01 TYPE=04 (Heat meter) (driver unknown!) DEV=im871a[00103843] RSSI=-88
(wmbus) parseELL @10 117
(wmbus) parseNWL @10 117
(wmbus) parseAFL @10 117
(wmbus) parseTPL @10 117
(TPL) num encrypted blocks 7 (112 bytes and remaining unencrypted 0 bytes)
[2022-12-01_15:33:31] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 72012563 mfct: (FLM) Unknown (0x198d) type: Heat meter (0x04
) ver: 0x01
(telegram) TPL CI=7a ACC=15 STS=00 CFG=a570 (bidirectional synchronous AES_CBC_IV nb=7 cntn=0 ra=0 hc=0)
[2022-12-01_15:33:31] (meter) newly created meter (licznik72012563 72012563 auto) did not handle telegram!
(wmbus) telegram from 72012563 ignored by all configured meters!
```

Where should I start to add support to them?

BIBOLV, 2022-12-02:
Hi, you need to wait for telegrams for your meters - telegrams you provided does not correspond to ID`s displayed on devices. Then you can check if telegrams are encrypted, if they are - you need decryption key, to decrypt them. When you have collected decrypted telegrams, you can post them here and we will take a look.

zdzichu, 2022-12-04:
I've put the photos of single meters, as others are the same (mysterious) make and model. I get other telegrams, all unreadable:
```
(meter) licznik82007410 82007410 "80448D191074008201077A290070A58B5F10B654A80B4B791DFB39C8F6E36F9372F0DFD0D38C09AA10F2B7C1E2D3EF92B3186E086223557A69DF139FEFCD216FB9CB300E335CD074DE9E43A14BB8336C1D6A9CC24FE33DD33EEA35444F9486CA74D72F42145803CB05E747DF4182FDBAEE10FFE2336D3BC50E6430A3A76B884FCD"
[2022-11-25_16:11:42] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 82007410 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01

(meter) licznik82007418 82007418 "7E448D191874008201077A070070A5349F9E0D8ED986E069D09A392C7A59AF1FE6C4A1B6B4D15A951EEC066DB6F9D87729F2F10524854DD3165FDE3ACB1E46710524819B03B5696FCE2DB6F3907177F85EB450505FF616D9D02C0C009EA2C0493FBB216C4E1C0B3454AFEC25AC1188F3B9C261BF54C547388442DF2E7FCC4F"
[2022-12-01_11:50:32] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 82007418 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01

(meter) licznik82007704 82007704 "80448D190477008201077AD90070A56A8616E803B19CCAD87EC47CC1D62C4FE439AC5489B29C8316F6AD8719EA98BED467EA75ABDFCF01AC518808D86CEED956881EA4E58E729069B3EC485D4ED482AE5473AB360E8AE7372BAC91CDA951EAD3D3C9B7C76BEE8C64504D38A9E9D37BE88EE5B18506F4D748CD1EFB7AC73098AE13"
[2022-11-23_14:41:53] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 82007704 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01

(meter) licznik82008732 82008732 "80448D193287008201077A290070A5A8C77E687BFAC03071B6C54BE27F04CF89E65012D52E6EFDE8C0D4DAC8FB83F8616B00F0EC50B9C9E876ED167E74685AB36DF3
6B74BF4076BAC28098A8BA4143F0355D917159E66E54B032DC59EE43DA5BB6D4617986D424EA8355B4060FC64652F55D26D597EF0ABA71EE50560BDDE67453"
[2022-11-24_09:10:43] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 82008732 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01

(meter) licznik82008743 82008743 "7E448D194387008201077AF00070A51DC234B6759F605367A4A8E6184D2ACD7A5058DF7B46D3C9F714B7172FD6F62D5E0F81C19E9EA805D1B7A548447DA488C7336BD83FEFB0CD311D2E3D0ABE73E0B80B17C4B4586A42BD20ACD9793EB588F66152B27B7D410FFAD3BB6573F3B42D9F2D5D88577ABF2BA89AFCB5326F1A67"
[2022-12-01_11:46:44] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 82008743 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01

(meter) licznik82008859 82008859 "80448D195988008201077A380070A5445E27525C4F153E177D7DC52D34EBD97730A6240F660F262205C46609E01BE3E85BB76097ADDE8B1AFE81ABECD88F76CD9D194890F6F44133291FD82FBFA63F444C593305C4D8C526473B5234A71E4711606410AF86CE667D787711054A7F2B7A0709AB3753E3EC07AC059F0EA37326967C"
[2022-11-23_14:56:19] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 82008859 mfct: (FLM) Unknown (0x198d) type: Water meter (0x07) ver: 0x01

(meter) licznik72012563 72012563 "7E448D196325017201047A150070A51224D17E0390858ADBDA5950FA99A3FC9738BD74601FF0173E785553DB617712D51BE9D232C933FC8580DA61BEE048DCE2095D4182211BEA6C84DDB9C6D9FCDFAC97BA74D7A70CDC2673599F77B8418FA832613A91EA5EBB819CECAA8A021FD2764C0C21934AEF2AFA56797B063E0961"
[2022-12-01_15:33:31] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 72012563 mfct: (FLM) Unknown (0x198d) type: Heat meter (0x04) ver: 0x01

(meter) licznik72012565 72012565 "7E448D196525017201047A9A0070A563E835D149E52FF082BDB3A6C400ED2C89AE134CF8D75A4EB6A50F6533DC49061CAB198E7828E6284347AED7E79B43F195882DF10400ACA20AB56173AB3BC52ABAFABE80DD4E7AEDF3F8770BBE4FFBE7A8E347425688E510BEA3488D286FB93023DDB1D91FE711CAD3F1657A26446B6B"
[2022-12-01_11:46:28] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 72012565 mfct: (FLM) Unknown (0x198d) type: Heat meter (0x04) ver: 0x01

(meter) licznik72012567 72012567 "7E448D196725017201047A2D0070A58E0CB83152FE4C845EB8C9B5CA693C3E39BAABFBF5BE1F0DF19E07370AD88CD513AFEB2568859AF2A03CD18EB115FEACF910A6362B1BAA6C52722E43B9E5FF2150FC8F542C65FCCFEF0FB7E322B997C8AEC067C19A73C1C2A9E12C0E75B63336568F4FCFA00B8F628DECEAC312AED8E4"
[2022-12-01_12:13:29] (wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 72012567 mfct: (FLM) Unknown (0x198d) type: Heat meter (0x04) ver: 0x01

(and the last one from another reader)
(rtl433) checkRTL433Frame "2022-11-25 16:36:29,,,Wireless-MBus,,72012574,,,,CRC,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,7c448d197425017201047a800070a585093905b0dfa57328
0a864f92030e60894ba28e0346d11a53d43c6f66728dff59bff9423b2b2ae34309c15b43269342d35642c4d3c02f203c15572740d88c29d181839df90ca2369341512ccc2a302c04e597e2eff6bf76a070a94a
0671a02f9d895956e8f5e09d38222175875a7d295ff3,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,4,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,T,1,Hea
t,122,128,0,42352,,,,,,,,,FLM,68,127,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,<0A>"
```

I am aware of key brute-forcers for some meters (like  https://github.com/ZeWaren/izar-prios-smart-meter-collector/tree/master/PC), but not for the ones I hear over the radio.
I've asked building administrator for keys few days ago, but I was met with silence :(

zdzichu, 2022-12-13:
Got some more information. Heat meter is Ploumeter RC12, using protocol AGFW NOWA.
Water meter has a FLOWIS+ addon (https://metering.com.pl/oferta/systemy-zdalnego-odczytu-danych/system-zdalnego-odczytu-librars/modul-flowis/)
I'm still trying to get decryption keys from the contractor, who installed them

wuub, 2023-02-14:
@zdzichu did you make any progress on decoding the water meter readings?

zdzichu, 2023-02-14:
Not yet, I'm trying to get the decryption keys. I've already talked with three companies in the supply chain, but didn't get the keys yet. Which is infuriating, as I paid dearly for those meters.
The companies are eager to sell me a *readout service*, where they come, read the meter and tell me the number 🤦🏻‍♂️

Szewcson, 2023-04-30:
I was able to track the vendor (https://www.wasser-geraete.de/en/categories/getPrm/prod/198/), and software used to readout https://play.google.com/store/apps/details?id=pl.globalrnd.libra.LibraRS&hl=pl&gl=US .

zdzichu, 2023-04-30:
Yeah, I saw it too. Even decompiled the app looking for the keys. Unsuccessfully.
It looks like the Android App uses Bluetooth to communicate with external radio module. This module is where WMBUS transmission is received and decrypted Libra RS Android app has no knowledge of wmbus protocol.
External receiver looks like this:
https://metering.com.pl/wp-content/uploads/2022/05/ule_skaner.png

closed 2023-07-31 by github-actions[bot]

Szewcson, 2026-04-24:
> Looks like Flowis is not wmbus → https://github.com/merbanan/rtl_433/issues/2317#issuecomment-1384567404

It looks like: yes and no. As far as I understand Flowis uses their proprietary protocol on some devices and that devices are even not visible in wmbusmeters since it is not wmbus. But it also looks like they uses OMS with manufacturer set to FLM on other devices which are decoded by rtl_433 as wmbus not flowis.

So still, support for FLM will be nice to have.