wmbusmeters/wmbusmeters #720: HA-addon: Possible to set pause between telegrams to reduce cpu-load?

CptBananaPeel, 2022-12-05:
As the title suggest, is it possible to set a timeout, pause or similar between telegrams? 
I've noticed that the addon uses around 7% of constant cpu-load which i'd very much like to bring down to minimize heat and load on the system. 
I was hoping to do something like: receive a telegram from a meter, wait 600s, listen and wait for new telegram, wait 600s - repeat.

My config:
```
conf: |-
  loglevel=normal  
  device=rtlwmbus[00000001]:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)  
  donotprobe=/dev/ttyAMA0 
  logtelegrams=false
  format=json   
  logfile=/dev/stdout  
  shell=/wmbusmeters/mosquitto_pub.sh  wmbusmeters/$METER_NAME "$METER_JSON"
```

Thanks a lot in advance.
And sorry if this is the wrong place to ask.

BIBOLV, 2022-12-18:
There are different features available, but non of them would help in your case - with ha-addon. My suggestions would be to use separate device where rldsdr is connected or get native wmbus dongle that eliminates any load on system.

KrzysztofHajdamowicz, 2023-02-05:
Try to change rtlwmbus to rtl_433, in my case it caused much less CPU load
```
device=rtl433:868.95M:t1
```

denissalanta-codecrafters, 2025-04-09:
> Try to change rtlwmbus to rtl_433, in my case it caused much less CPU load
> 
> ```
> device=rtl433:868.95M:t1
> ```
Thanks for the tip 🙇🏻‍♂️this fixed my CPU temperature [🤣]

[image] that made me look to replace my current setup. Really appreciate it

rkotulan, 2025-09-28:
> Try to change rtlwmbus to rtl_433, in my case it caused much less CPU load
> 
> ```
> device=rtl433:868.95M:t1
> ```

This helped me reduce cost o my synology from 15 % of cpu load to 3 %.
Thanks.