wmbusmeters/wmbusmeters #724: 1.10.0 does not find devices anymore

marcelstoer, 2022-12-05:
Did I maybe just walk into a storm? By coincidence I am now running the bleeding edge 1.10.0 (or 1.9.0 before the tag was created), commit 34b0cc5.

Contrary to a couple of days ago, `wmbusmeters` now doesn't find any devices anymore. The im871a dongle _is_ plugged in and mounted at `/dev/ttyUSB0`; the only attached USB device.

```
pi@white-rose:~ $ wmbusmeters/build/wmbusmeters --debug --format=json  --logtelegrams /dev/ttyUSB0:im871a:t1,c1 MyGasMeter izar 875103bf NOKEY
(wmbusmeters) version: 1.9.0-175-g34b0cc5
(config) using device: /dev/ttyUSB0:im871a:t1,c1 
(config) number of meters: 1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 1498 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>1495
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 1499 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
^C(serial) stopping manager
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
pi@white-rose:~ $ ls -al /dev/ttyUSB0 
crw-rw---- 1 root dialout 188, 0 Dec  5 21:40 /dev/ttyUSB0
pi@white-rose:~ $ wmbusmeters/build/wmbusmeters --logtelegrams /dev/ttyUSB0:im871a:t1,c1
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!

```

Strangely, if I use `auto` then the very device I configured explicitly in the previous command is found and all seems well.

```
pi@white-rose:~ $ wmbusmeters/build/wmbusmeters --debug --logtelegrams --donotprobe=/dev/ttyAMA0 auto:t1,c1
(wmbusmeters) version: 1.9.0-175-g34b0cc5
(config) number of meters: 0
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465/3665) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465/3665) reading response... 1
(serial) received binary ""
(amb8465/3665) reading response... 2
(serial) received binary ""
(amb8465/3665) reading response... 3
(serial) received binary ""
(amb8465/3665) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465/amb3665)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001601F3390000C61B"
(im871a) checkIM871AFrame "A581100833001601F3390000C61B"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 1bc6 expected 1bc6
(im871a) received full frame
(device info bytes) "33001601F3390000"
(im871a/im170a) info: type=im871a mode=collector firmware=16 hci=01 uid=000039f3
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000A00B32552381000010001FF07023200010000007986"
(im871a) checkIM871AFrame "A5810616FF000A00B32552381000010001FF07023200010000007986"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 8679 expected 8679
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(device config bytes) "FF000A00B32552381000010001FF0702320001000000"
(im871a/im170a) config: other link_mode=ct_n1a id=00103852 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 00103852 im871a firmware: 16
(main) opening t1,c1
Started auto im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000A00B32552381000010001FF07023200010000007986"
(im871a) checkIM871AFrame "A5810616FF000A00B32552381000010001FF07023200010000007986"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 8679 expected 8679
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00103852
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001601F3390000C61B"
(im871a) checkIM871AFrame "A581100833001601F3390000C61B"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 1bc6 expected 1bc6
(im871a) received full frame
(im871a) device info completed
(im871a) device info: type=im871a mode=collector firmware=16 hci=01 uid=000039f3
(im871a) set config to set link mode 0a
(serial /dev/ttyUSB0) sent "A501030600020A300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received binary "A5C2031944E61EBF0351870006B2710654168A55BF80B3C60A9555F1635CFDD5"
(im871a) checkIM871AFrame "A5C2031944E61EBF0351870006B2710654168A55BF80B3C60A9555F1635CFDD5"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 92 (-76 dBm)
(im871a) got crc16 d5fd expected d5fd
(im871a) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 26
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(wmbus) parseDLL @0 26
(telegram) DLL L=19 C=44 (from meter SND_NR) M=1ee6 (GWF) A=875103bf VER=00 TYPE=06 (Warm Water (30°C-90°C) meter) (driver unknown!) DEV=im871a[00103852] RSSI=-76
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) TPL CI=b2
Received telegram from: 875103bf
...
```

weetmuts, 2022-12-05:
Hmmm, if you try just im871a:c1,t1 ?

weetmuts, 2022-12-05:
Arghghghgh, you are quite right. Please hold.

marcelstoer, 2022-12-05:
> if you try just im871a:c1,t1

```
pi@white-rose:~ $ wmbusmeters/build/wmbusmeters --donotprobe=/dev/ttyAMA0 im871a:t1,c1
Started config im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
```

Xaver445, 2022-12-05:
mh

weetmuts, 2022-12-05:
Fix pushed. Please test.

marcelstoer, 2022-12-05:
Thank you Fredrik!