wmbusmeters/wmbusmeters #725: WHE542 decoding

algorni, 2022-12-06:
Cross posting here too this:

https://github.com/merbanan/rtl_433/issues/2256

Hi,
this is about a Siemens Heating cost allocator device using W M-BUS.

I'm able to receive the package but it seems the data packet is not fully decoded as i never got "inst_volume_flow_min_1" valued

Look:

{"time":"2022-12-01 10:05:58","model":"Wireless-MBus","mode":"S","M":"LSE","id":93442859,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532592844933408780b6e0000000dff5f2f0001a0807f3208b06effffde2967100000de2b000000007501f700a9003a00080000000000000000000000d5fb0000046d130bc12c0648","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

And i receive a lot of them!
How can i help to understand if this is something doable or not?

Here is some other JSON output

{"time":"2022-12-01 10:06:03","model":"Wireless-MBus","mode":"S","M":"LSE","id":92838933,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532338983923408780b6e1901000dff5f2f0001b2807e3408b06effffde2990160000de2b150100009b012301ed00fe0075000000000000000000000066f97300046d1f0bc12cb240","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:08:19","model":"Wireless-MBus","mode":"S","M":"LSE","id":93451103,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532031145933408780b6e0000000dff5f2f0001b180ff1008b06effffde2901000000de2b000000000000000000000000000000000000000000000000ffff0000046d150bc12cf3e2","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:06:27","model":"Wireless-MBus","mode":"S","M":"LSE","id":93451113,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532131145933408780b6e5700000dff5f2f0001b1807f2308b06effffde2929040000de2b5400000074005a007200380000000000000000000000000053fe3600046d130bc12c003b","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:06:58","model":"Wireless-MBus","mode":"S","M":"LSE","id":92838934,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532348983923408780b6e6600000dff5f2f0001b280fe2a08b06effffde2963140000de2b630000005d0156012e01360143000000000000000000000049fa3f00046d1a0bc12c99e0","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:07:36","model":"Wireless-MBus","mode":"S","M":"LSE","id":93451085,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532851045933408780b6e0000000dff5f2f0001b180000308b06effffde2973000000de2b000000001c00210002000000010000000000000000000000b7ff0000046d160bc12ce29c","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:08:56","model":"Wireless-MBus","mode":"S","M":"LSE","id":92838938,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532388983923408780b6e0400000dff5f2f0001b2807e3508b06effffde2909000000de2b040000000400010000000100000000000000000000000000f7ff0400046d1a0bc12c1a70","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

{"time":"2022-12-01 10:08:58","model":"Wireless-MBus","mode":"S","M":"LSE","id":93442857,"version":52,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":73,"data":"46446532572844933408780b6e0300000dff5f2f0001a0807e3008b06effffde2912030000de2b02000000610068003f000c00030000000000000000000000c8fe0200046d1b0bc12ca7c7","mic":"CRC","CI":120,"AC":0,"ST":0,"CW":0,"inst_volume_flow_min_1":"0.000 m3/min"}

algorni, 2022-12-06:
The "inst_volume_flow_min_1" key  is always "0.000 m3/min".

algorni, 2022-12-06:
I receive such telegrams:

telegram=|_48446532931045933408780B6E0000000DFF5F2F0001D0807F1A08B06EFFFFDE2991020000DE2B000000005200560034001700080000000000000000000000DDFE0000046D290EC62C|+56
Received telegram from: 93442865
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 68 dBm
                driver: qcaloric
telegram=|_48446532652844933408780B6E1300000DFF5F2F0001BF80813508B06EFFFFDE2919040000DE2B090000008F005E006D0037000400000000000000000000005DFE0900046D2C0EC62C|+57
Received telegram from: 93442871
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 48 dBm
                driver: qcaloric
telegram=|_48446532712844933408780B6E9500000DFF5F2F0001BE807E0108B06EFFFFDE2963050000DE2B49000000CD00C10084000900010000000000000000000000CDFD3100046D2F0EC62C|+60
Received telegram from: 93451102
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 34 dBm
                driver: qcaloric
telegram=|_48446532021145933408780B6E0000000DFF5F2F0001D080FE1108B06EFFFFDE2900000000DE2B00000000000000000000000000000000000000000000000000000000046D2D0EC62C|+61
Received telegram from: 93451108
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 32 dBm
                driver: qcaloric
telegram=|_48446532081145933408780B6E0000000DFF5F2F0001D080FE2908B06EFFFFDE2977010000DE2B00000000260074000C0003000000000000000000000000004FFF0000046D280EC62C|+61
Received telegram from: 93450758
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 67 dBm
                driver: qcaloric
telegram=|_48446532580745933408780B6E2400000DFF5F2F0001C680FE1508B06EFFFFDE2911010000DE2B1500000029002C0005000C0000000000000000000000000091FF0F00046D2A0EC62C|+63
Received telegram from: 93451128
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                device: rtlwmbus[00000001]
                  rssi: 44 dBm
                driver: qcaloric
telegram=|_48446532281145933408780B6E1600000DFF5F2F0001BE80811B08B06EFFFFDE2945000000DE2B050000000D000B0005000400010000000000000000000000D3FF0500046D2C0EC62C|+65

weetmuts, 2022-12-06:
There are often a lot of heat cost allocators, one per radiator in your building. The telegrams look ok.
You can check the contents of the telegrams like this:
https://wmbusmeters.org/analyze/48446532931045933408780B6E0000000DFF5F2F0001D0807F1A08B06EFFFFDE2991020000DE2B000000005200560034001700080000000000000000000000DDFE0000046D290EC62C

But it sounds like you are looking for a heat meter? There is no volume_flow in heat cost allocators.
Run rtl_wmbus for a couple of hours, then remove all lines with "Heat Cost Allocator" and
see if there is a heating meter in there somewhere....

algorni, 2022-12-06:
Hi @weetmuts  first, thank you very much!!!  

Yes, we have 1 HCA per Radiator, and we have multiple of flats in this building, so I can receive a lot of them.
Luckily I've a mapping doc with the HCA ID and Flat unit and room to associate them!

I've cross post also some update here: https://github.com/merbanan/rtl_433/issues/2256

It sounds like I found out one of my heat cost allocator device!!
And, yep, i'm able to decode the telegram and i'm getting the current_consumption_hca parameter (which is shown as it is in the device display too!).   

Anyway from the Analyzer tool i see:

011   : 0B dif (6 digit BCD Instantaneous value)

What does it means?

closed 2023-03-18 by github-actions[bot]