wmbusmeters/wmbusmeters #731: Test fails on Raspberry Pi

scibi, 2022-12-09:
Hi!

I was trying to build Debian package on Raspberry Pi. I used version 1.10.1 and tests run after build failed:

```
   dh_auto_test
        make -j1 test VERBOSE=1
make[1]: Wejście do katalogu '/home/scibi/wmbusmeters/packaging/wmbusmeters-1.10.1'
Building 1.10.1-1-g535005a
ERROR lookup1 0xa0 expected "ACCESSOR_RED ACCESS_BITS_80 ALL_ACCESS" but got "ACCESSOR_RED ACCESS_BITS_0 ALL_ACCESS"
ERROR lookup1 0x35 expected "ACCESSOR_TYPE_5 ALL_ACCESS NO_ACCESS" but got "ACCESSOR_TYPE_BEA5B20C ALL_ACCESS NO_ACCESS"
OK: test internals
138c138
<   "battery": "BATTERY_330",
---
>   "battery": "BATTERY_0",
174c174
<   "dust_level": "DUST_0",
---
>   "dust_level": "DUST_BEC285C4",
ERROR: Test C1 meters
688c688
<   "status": "ERROR_FLAGS_2000000",
---
>   "status": "ERROR_FLAGS_0",
795c795
<   "other": "OTHER_FLAGS_14",
---
>   "other": "OTHER_FLAGS_0",
797c797
<   "status": "STATUS_FLAGS_CF0",
---
>   "status": "STATUS_FLAGS_0",
OK json: Test S1 meters
OK fields: Test S1 meters
OK: Test mbus
OK: Test ANYID
OK: Ignoring auto + ANYID
OK: Test list-envs and list-fields
OK: Test list-envs and list-fields
OK json: Test non-compliant apator162 meters
OK fields: Test non-compliant apator162 meters
OK json: Test Izars meters
OK: Test listen to all izars
OK: Test aes encrypted telegrams
OK: Test that telegram really is encrypted
OK: Test that failed decryption warning only prints once.
OK: Test shell invocation
OK: Test shell in config file
OK: Test shell environment
OK: Test that normal meterfiles are written
OK: Test that meterfiles with name-id are written
OK: Test that meterfiles with id are written
OK: Test that meterfiles with timestamps are written
OK: Test that meterfiles with timestamps are written
138c138
<   "battery": "BATTERY_330",
---
>   "battery": "BATTERY_0",
174c174
<   "dust_level": "DUST_0",
---
>   "dust_level": "DUST_BEFD27DC",
ERROR: Test normal config files with C1 meters
OK: Test logfile
OK: Test single meter conf file matches any id
OK: Test listen and print any meter heard in logfile
OK: Test listen and print any meter heard on stdout
OK: Test detection of drivers for all telegrams
OK: Test listen to wildcard * id
OK: Test listen to wildcard suffix 8856* id
OK: Test listen to two comma separted ids
OK: Test listen to three comma separted ids
OK: Test listen with negated ids
OK: Test conversions of units using calculations
OK: Test cmdline formulas --calculate_...
OK: Test meter config formulas calculate_...
OK: Test selected fields
OK: Test oneshot
OK: Test wrong keys
OK: Test config4 with added conversions
OK: Test cmdline --overridedevice and --useconfig
OK: Test cmdline --overridedevice without --useconfig   
OK: Test cmdline --overridedevice with --useconfig and surplus argument
OK: Test cmdline --overridedevice with --useconfig and surplus arguments
OK: Test cmdline --overridedevice with --useconfig and surplus argument prefixed
OK: Test cmdline --overridedevice with --useconfig and surplus argument infixed
OK: Test additional json from cmdline
OK: Test additional shell envs from cmdline
OK: Test additional json from wmbusmeters.conf and from meter file
OK: Test rtlwmbus starting background script to produce telegrams
OK: Reading binary telegram from stdin
OK: Reading binary telegram from file
OK: Reading rtlwmbus formatted telegrams from stdin
OK: Reading rtlwmbus formatted telegrams from file
OK: Reading rtl433 formatted telegrams from stdin
OK: Reading rtl433 formatted telegrams from file
OK: Test ok serial rawtty telegram on stdin
OK: Test bad serial rawtty telegram with interspersed noise, on stdin
OK: Test duplicates are ignored
OK: Test duplicates are left alone
OK: Test match on dll id
OK: Test match on tpl id
OK fields: Test unix timestamp in fields
OK: Test log timestamps
OK: Test meter driver extra settings
OK: Test hex on commandline
OK: Test hex on commandline with meter
OK: Test invalid hex on commandline
OK: Test hex on stdin
OK: Test auto removal of dll crcs in simulated telegrams
OK: Test broken telegrams
OK: Test rtlwmbus with crc errors
Testing drivers
OK json: driver_abbb23.cc
OK fields: driver_abbb23.cc
OK json: driver_amiplus.cc
OK fields: driver_amiplus.cc
OK json: driver_apator08.cc
OK fields: driver_apator08.cc
OK json: driver_apator162.cc
OK fields: driver_apator162.cc
OK json: driver_apator172.cc
OK fields: driver_apator172.cc
OK json: driver_apatoreitn.cc
OK fields: driver_apatoreitn.cc
OK json: driver_aventieshca.cc
OK fields: driver_aventieshca.cc
39c39
<   "error_flags": "ERROR_FLAGS_1100 HF MEASUREMENT",   
---
>   "error_flags": "ERROR_FLAGS_0 HF MEASUREMENT",
2c2
< Vatten;61070072;466.472000;ERROR_FLAGS_1100 HF MEASUREMENT;1111-11-11 11:11.11
---
> Vatten;61070072;466.472000;ERROR_FLAGS_0 HF MEASUREMENT;1111-11-11 11:11.11
ERROR: driver_aventieswm.cc
OK: Test analyze encrypted (no-key) ctr full telegram   
OK: Test analyze encrypted (no-key) ctr compact telegram
OK: Test analyze encrypted (with-key) ctr full telegram 
OK: Test analyze encrypted (no-key) ctr compact telegram
OK: Test analyze CBC IV (no-key)
Slower tests...
OK: reading all stdin before closing down
OK: Test config with oneshot
OK: Test config with exitafter
OK: Test config override with oneshot
OK: Test config override with exitafter
RUNNING Test alarms ...
STDERR---------------------------------
[ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2022-12-07 23:17)
(wmbus) successfully reset wmbus device
STDOUT---------------------------------
TMP/OUTPUT-----------------------------
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","status":"DRY","total_m3":6.408,"target_m3":6.408,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-12-07T22:17:24Z"}==
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","status":"DRY","total_m3":6.408,"target_m3":6.408,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-12-07T22:17:30Z"}==
TMP/ALARM------------------------------
ALARM_SHELL DeviceInactivity [ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2022-12-07 23:17)
---------------------------------------
OK: Test alarms
Some tests failed!
make[1]: *** [Makefile:313: test] Błąd 1
```

System is running fresh Bullseye installation.

```console
$ dpkg -l | grep 'g++'
ii  g++                              4:10.2.1-1                     armel        GNU C++ compiler
ii  g++-10                           10.2.1-6                       armel        GNU C++ compiler
```

```console
$ cat /proc/cpuinfo 
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.34
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 0000
Serial          : 00000000f3a8851e
```

I tried build package the same way on Debian Bullsyey running on my amd64 system and there were no issues.

weetmuts, 2022-12-12:
Indeed, the c++ static initializers ran differently when compiled with the deb release options......
fix pushed. Please test.

scibi, 2022-12-14:
I've tried to build Debian package on current master but it failed the same way:

```
   dh_auto_test
        make -j1 test VERBOSE=1
make[1]: Wejście do katalogu '/home/scibi/wmbusmeters/packaging/wmbusmeters-1.10.2'
Building 1.10.1-15-g28c96e1 with local changes
ERROR lookup1 0xa0 expected "ACCESSOR_RED ACCESS_BITS_80 ALL_ACCESS" but got "ACCESSOR_RED ACCESS_BITS_0 ALL_ACCESS"
ERROR lookup1 0x35 expected "ACCESSOR_TYPE_5 ALL_ACCESS NO_ACCESS" but got "ACCESSOR_TYPE_BEE56D9C ALL_ACCESS NO_ACCESS"
OK: test internals
138c138
<   "battery": "BATTERY_330",
---
>   "battery": "BATTERY_0", 
174c174
<   "dust_level": "DUST_0", 
---
>   "dust_level": "DUST_BEA0E144",
ERROR: Test C1 meters
688c688
<   "status": "ERROR_FLAGS_2000000",
---
>   "status": "ERROR_FLAGS_0",
795c795
<   "other": "OTHER_FLAGS_14",
---
>   "other": "OTHER_FLAGS_0",
797c797
<   "status": "STATUS_FLAGS_CF0",
---
>   "status": "STATUS_FLAGS_0",
OK json: Test S1 meters
OK fields: Test S1 meters   
OK: Test mbus
OK: Test ANYID
OK: Ignoring auto + ANYID   
OK: Test list-envs and list-fields
OK: Test list-envs and list-fields
OK json: Test non-compliant apator162 meters
OK fields: Test non-compliant apator162 meters
OK json: Test Izars meters  
OK: Test listen to all izars
OK: Test aes encrypted telegrams
OK: Test that telegram really is encrypted
OK: Test that failed decryption warning only prints once.
OK: Test shell invocation   
OK: Test shell in config file
OK: Test shell environment  
OK: Test that normal meterfiles are written
OK: Test that meterfiles with name-id are written
OK: Test that meterfiles with id are written
OK: Test that meterfiles with timestamps are written
OK: Test that meterfiles with timestamps are written
138c138
<   "battery": "BATTERY_330",
---
>   "battery": "BATTERY_0", 
174c174
<   "dust_level": "DUST_0", 
---
>   "dust_level": "DUST_BEE9C36C",
ERROR: Test normal config files with C1 meters
OK: Test logfile
OK: Test single meter conf file matches any id
OK: Test listen and print any meter heard in logfile
OK: Test listen and print any meter heard on stdout
OK: Test detection of drivers for all telegrams
OK: Test listen to wildcard * id
OK: Test listen to wildcard suffix 8856* id
OK: Test listen to two comma separted ids
[... 70 lines cut ...]
OK: Test analyze encrypted (no-key) ctr compact telegram
OK: Test analyze CBC IV (no-key)
Slower tests...
OK: reading all stdin before closing down
OK: Test config with oneshot
OK: Test config with exitafter
OK: Test config override with oneshot
OK: Test config override with exitafter
RUNNING Test alarms ...
STDERR---------------------------------
[ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2022-12-13 19:20)
(wmbus) successfully reset wmbus device
STDOUT---------------------------------
TMP/OUTPUT-----------------------------
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","status":"DRY","total_m3":6.408,"target_m3":6.408,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-12-13T18:20:04Z"}==
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","status":"DRY","total_m3":6.408,"target_m3":6.408,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-12-13T18:20:10Z"}==
TMP/ALARM------------------------------
ALARM_SHELL DeviceInactivity [ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2022-12-13 19:20)
---------------------------------------
OK: Test alarms
Some tests failed!
make[1]: *** [Makefile:318: test] Błąd 1
```

Please note that I was running it on Rasspberry Pi 1 (Debian architecture armel). When that failed I set up fresh system on RPi 4 (Debian architecture arm64) and all tests passed.

weetmuts, 2022-12-25:
Can you give me the exact commands from: git clone.. building  and then test?

closed 2023-03-27 by github-actions[bot]

scibi, 2023-08-11:
Sorry it took me so long to respond. I've just installed fresh Raspberry Pi OS on RPi 4 and tried to build Debian package. Here is a complete log:
[screenlog_wmbusmeters.txt](https://github.com/wmbusmeters/wmbusmeters/files/12325808/screenlog_wmbusmeters.txt)