wmbusmeters/wmbusmeters #732: im871a stops working after computer reboot

Nikodemski2002, 2022-12-10:
Hi everyone! I'm using wmbusmeters with an IM871A dongle to read my MC803. I've set up a static USB address to my IM871A dongle because I have two devices containing SL CP210x UART bridge devices plugged into my computer. I've also set up a cron job to reset the computer every day at 2:05 AM. After resetting the computer, wmbusmeters is running normally but it doesn't collect any data (systemctl status shows running active). If I do a manual reset of the dongle with [usbreset](https://github.com/jkulesza/usbreset), it starts working normally and collects the data. Another scenario is when the computer resets itself again at 2:05 AM, but it means 24 hours without readings. Yet another scenario is when the dongle starts working by itself randomly. To sum up, is there anything I can do in the wmbusmeters config file or anywhere else to reset the dongle if it's not collecting any data for example in a 5-minute period? Below, I attach my wmbusmeters.conf content:

`loglevel=normal
device=/dev/im871a:im871a:c1
donotprobe=/dev/im871a
logtelegrams=false
format=fields
selectfields=id,volume_flow_m3h,t1_temperature_c,t2_temperature_c,timestamp
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log
ignoreduplicates=false`

Best regards,
Nikodem

weetmuts, 2022-12-12:
The static link from /dev/im871a is not something that I have good experiences with since the usb numbers sometimes change at reboot....

If the other serial device can handle a bit of probing, then perhaps using just
`device=im871a:c1` works better? And remove the donotprobe.
The probe is quick because it knows that it is probing for an im871a.

Nikodemski2002, 2023-01-08:
Hi. Sorry for my late response. Autofiding of im871a solved the problem, thank you very much. I've tested it for several days, and since using `device=im871a:c1` there are no problems at all.

closed 2023-01-08 by Nikodemski2002