wmbusmeters/wmbusmeters #733: Suche tatsächliche Leistung in KW (laut Display werden diese angezeigt ,aber im "Protokoll" nicht vorhanden// Ist es vll ein Errechneter Wert ?

Xaver445, 2022-12-10:
Explain the problem..... 

https://wmbusmeters.org/analyze/5e442d2c0105798240047a7d0050252f2f0406c50e000004147B86000004ff074254000004ff086047000002594117025d9a14023Bed0302ff220000026cca2c4406750B00004414ad680000426cc12c2f2f2f2f2f2f2f2f2f2f2f2f2f2f2f30 

TEST // TEST

Suche tatsächliche ,vom Wärmmengenzähler multical303 ;Leistung in KW (laut Display werden diese angezeigt ,aber im "Protokoll" nicht vorhanden// Ist dies vielleicht  ein errechneter Wert..

Vielleicht so =    Delta T2 minus T1 x  Durchfluss  oder so =  "Angezeigter Wert "  ?????   <Ist ein Wert von 0 bis 20 (21,22 kW)>

Ziel ist es die aktuelle ausgegebene Leistung anzuzeigen / darzustellen  . Mh 

Prot.:
Wert in Zeile 58 ist rot nicht grün und laut WMZ- Wärmemengenzähler drücke ich 8 x die Taste => der 8te Wert  . ROT ?- Power_khw ?-fehlt .  Denke das es nicht Darstellbar ist- (Flippt zu stark) ODER ?

Xaver445

( 5E442D2C0105798240047A590050252F2F_0406D10E00000414E286000004FF078254000004FF089547000002596117025D8714023B9A0602FF220000026CCA2C4406750B00004414AD680000426CC12C2F2F2F2F2F2F2F2F2F2F2F2F2F2F2F
)

weetmuts, 2022-12-12:
I glaube das Zeile 58 ist ein status wort.  Der power_kw wird von diesem Zähler nicht gesendet.

JensHoRi, 2023-01-27:
Für die Berechnung bräuchtest du den exakten aktuellen Durchfluss. Da du den Durchfluss nur je Stunde hast (insgesamt? Letzte Stunde) würde ich einfach alle paar Minuten den kwh Wert lesen (z.B. alle 15 Minuten) und dann kannst du die Leistung der letzten 15 Minuten sichtbar machen.

closed 2023-04-29 by github-actions[bot]