wmbusmeters/wmbusmeters #734: HA-Addon MQTT Discovery without values vor Multical 303

thecem, 2022-12-10:
Hi,

since the latest HA-Addon update to 1.10.2 no values for Multical available.
All values are 0, beside this total volume areactual.

Whats changed?

A falback to 0.2.23 is working.

THX a lot!

mieschc, 2022-12-10:
Had a similiar issue after the update with my Multical302. All values were still received via the Addon (looked it up in the log) but not forwarded to my Home Assistant MQTT Sensor. 

I found out that the naming of my current power (kW) in the telegram changed. Before it was "current_power_consumption_kw":0 and after the update it is: "power_kw":4.4

Had to change the value template name for the mqtt sensor within configuration.yaml to the new name "power_kw" and everything is working again.

thecem, 2022-12-11:
Thx! In between I found the solution.
Mqtt was still available, but the driver changed.

the configured `driver=multical303 ` led to not updated entity in ha by discovery, only one value was there.
I saw in the log that wmbusmeters is using `kamheat ` as driver even it was diffrent configured (I think there is a another issue open regarding this).
After changing the configuration to kamheat, the discovery function was working and changed the entitys, so I could get finaly  the values. For now it works

**What I already adressed is: If the value name change, or the entity name changes and you use the discovery funktion, all your energy statistics are wasted.**

So maybe it is better to do the entity not via discovery, but in configuration.yaml (old hard style).

weetmuts, 2022-12-12:
Thanks for figuring this out!

The multical heat driver changes were properly documented in the CHANGES file. 
https://github.com/weetmuts/wmbusmeters/blob/master/CHANGES

Hopefully this should be the last of the significant driver changes, it was part of the big rewrite.

I am not sure I understand exactly the functionality in the discovery function, but relying on the exact driver name to be chosen by wmbusmeters could be a problem, if that is what you are doing.

My attempt at guarantee (which was broken for the multcail heat to kamheat driver update, but kept for most of the other drivers) is that the same driver should be backwards compatible, but there might be better or more normalized drivers that is picked by the wmbusmeters auto detection in the future. 
Ie wmbusmeters might switch name from "kamheat" to "kamheat2" in the future as the recommended driver,
but kamheat should still work.

thecem, 2022-12-12:
Thanks for the work! Even if the driver is changing, this should not be the problem, the main problem is changing the value names.
But finally I think this now most likely aligned.

thx a lot!

closed 2023-04-05 by github-actions[bot]