wmbusmeters/wmbusmeters #740: Add: JSON output with --longheaders

blackenstock, 2022-12-12:
I do like to test the payload from the water-/gas-meter (if the radio does send the correct payload). Add --longheaders argument which allows to verify all content (decrypted) and headers from the telegram as json output.

weetmuts, 2023-04-24:
I have thought about adding --addheaders to include the lower level headers in the json.
Something like this:

{
  "media":"bus/system component",
  "meter":"water",
  "name":"",
  "id":"20221031",
  "actuality_duration_s":20,
  "total_m3":0.062,
  "target_m3":4294967.295,
  "target_date":"2128-03-31",
  "status":"BATTERY_LOW POWER_LOW",
  "power_mode":"SAVING",
  "battery_y":0,
  "timestamp":"2023-01-10T20:26:15Z"
  "headers": {
     "dll-c":"44 from meter SND_NR
     "dll-mfct":"ffff (XXX)",
     "dll-id": 11111111 (11111111)",
     "dll-version":"01",
     "dll-type":"0e (Bus/System component)",
      "ell-ci-field":"8c (ELL: Extended Link Layer I (2 Byte))",
      "ell-cc":"04 (slow_resp)",
      "ell-acc":"23",
      "afl-ci-field":"90 (AFL: Authentication and Fragmentation Sublayer)",
      "afl-len":"0f (15)",
      "afl-fc":"002c (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)",
      "afl-mcl":"25 (AES_CMAC_128_8 MessCounter)",
      "afl-counter":"03000000 (3)",
      "afl-mac":"B95DB7E22587AC57",
      "tpl-ci-field":"7a (EN 13757-3 Application Layer (short tplh))",
      "tpl-acc-field":"37",
      "tpl-sts-field":"04 (POWER_LOW)",
      "tpl-cfg":"2007 0720 (AES_CBC_NO_IV )",
      "tpl-cfg-ext":"10 (KDFS=1)"
  }
}

blackenstock, 2023-04-26:
The JSON --addheaders example you provided is a good approach, I would prefer if you could split the value and the explanation, eg: 
"dll-c":"44 from meter SND_NR"
to
"dll-c":"44"
"dll-c-desc":"from meter SND_NR"
To simplify the filtering and remain the human readability.