wmbusmeters/wmbusmeters #743: Support data going trough http/https/websocket,...

Hadatko, 2022-12-13:
Hi, is there a way where raw data can arrive using network endpoint instead of usb dongle? I would like to add support for device which can collect data from many devices, but it is not connected via usb, instead of it is sending data to network endpoint.  So it would be great if i can setup endpoint in this project and send data there.

BIBOLV, 2022-12-18:
Hi, yes, of course, as described in documentation, you can run netcat that would forward received data over network to wmbusmeters - `nc -lku 4444 | wmbusmeters stdin` . If you need http/s, you can use haproxy or nginx also supports tcp proxy if I`m not mistaken, and then forward data to local port where netcat is listening.

Hadatko, 2022-12-18:
Will that works when wmbusmeters are running as addon in homeassistant? I wanted use data in HA.

BIBOLV, 2022-12-18:
No, currently it is not possible to achieve that with HA-addon IMO. You need to set up wmbusmeters standalone instance and then you can push data to HA.

BIBOLV, 2022-12-18:
@weetmuts maybe cmd execution could be added to stdin, same as for rtlwmbus - `stdin:CMD(<command line>)`? Then together with this [PR](https://github.com/weetmuts/wmbusmeters/pull/752) requested functionality would be achievable.

SzczepanLeon, 2022-12-19:
@Hadatko you can use "dummy" rtlwmbus, i.e. use
`device=rtlwmbus:CMD(/usr/bin/nc -lku 9011)`
but without all fields. For example send only timestamp, rssi, rrame type, crc status, telegram (etc):
`T1;1;1;2022-12-19 10:32:01.000;-35;;;0x6E4401065`
I know that this is workaround, but it is available right now, without any implementation.

itmay, 2022-12-30:
Any idea how to redirect stdout as response?

weetmuts, 2022-12-30:
Actually it works right now (at least with the latest version of wmbusmeters on github)
 But you have to use rawtty or hex to define the format, stdin is just where you want to fetch it from. :-) 
If you just supply stdin then it defaults to rawtty. I.e. stdin conflicts with CMD.

You can do read raw bytes from stdin:
```
echo -n "1844AE4C4455223368077A55000000041389E20100023B0000" | xxd -r -p | ./build/wmbusmeters --silent --format=json stdin:rawtty Moo iperl 33225544 NOKEY | jq .
```

You can read hex bytes from stdin:
```
echo -n "1844AE4C4455223368077A55000000041389E20100023B0000" | ./build/wmbusmeters --silent --format=json stdin:hex Moo iperl 33225544 NOKEY | jq .
```

You can read raw bytes from incoming tcp connects:
```
wmbusmeters --silent --format=json 'rawtty:CMD(nc -lk 4444)' Moo iperl 33225544 NOKEY | jq .
```

and then send some raw bytes like this:
```
echo -n "1844AE4C4455223368077A55000000041389E20100023B0000" | xxd -r -p | nc -q0 localhost 4444
```

Remember to change the telegrams you send while testing, or add --ignoreduplicates=false
otherwise the second identical telegram will be ignored.

The output is always on stdout unless you use meter files, log files or use a shell command.

itmay, 2022-12-30:
Thanks for explanation. I have already tested receiving frames using a TCP connection and it works, but results of analyze is printed to stdout. In my scenario I would like to send output to tcp client (which sends telegram to decode). I would like to build something like wmbusmeters.org, but I don't need UI. It would be ideal to get a REST API.

SzczepanLeon, 2022-12-30:
Check how shell option works.

weetmuts, 2022-12-30:
Yes analyze is special, it is not meant to be piped anywhere, it is user interface feature.
Why do you want analyze through REST? :-) Remeber that it is not a stable output format, it will change over time, when I feel like it.

itmay, 2022-12-30:
I have a small project where I receive wmbus frames from multiple sources (GPRS, LoRa, XML) and I would like to use wmbusmeters to preview the structure of unsupported frames (just for research purposes) I think for this moment I will create myself an http server in nodejs, which will run wmbusmeters as a process.

weetmuts, 2022-12-30:
Is this a project on github? 

I will add LoRA support as well to wmbusmeters. Is it relayed wmbus meters over LoRA you are looking at or is it native LoRA meters sending embedded wmbus?

Yeah, setting up a nodejs server should be trivial. That is how its done on wmbusmeters.org except its not node. :-)

itmay, 2022-12-30:
Right now it's my private project. Focused on hardware research. It's relayed wmbus frames over LoRa.

closed 2023-04-01 by github-actions[bot]