wmbusmeters/wmbusmeters #746: Can't read telegrams - Unknown driver

Mufika, 2022-12-14:
Hi,
I can't read my telegrams. I get the following error 

Started config rtlwmbus[00000001] listening on t1
(meter) MainWaterMeter: meter detection did not match the selected driver auto! correct driver is: unknown!
(meter) Not printing this warning again for id: 27604781 mfct: (WEH) E Wehrle, Germany (0x5ca8) type: Water meter (0x07) ver: 0x03
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) MainWaterMeter: meter detection could not find driver for id: 27604781 mfct: (WEH) E Wehrle, Germany (0x5ca8) type: Water meter (0x07) ver: 0x03
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|5244A85C8147602703077A5B0840252F2F_0413B39100004413000000004D931E2C73FE0000000000000000000000000000000000000000000000000000000000000000000000009885001A0C002F2F426CBE29|+20

BIBOLV, 2022-12-18:
Hi,
Please check if this driver gives you correct reading:
```
{
    "media":"water",
    "meter":"waterstarm",
    "name":"",
    "id":"27604781",
    "total_m3":37.299,
    "current_status":"PERMANENT_ERROR",
    "consumption_at_history_1_m3":0,
    "history_reference_date":"2021-09-30",
    "history_1_date":"2021-09-30",
    "timestamp":"2022-12-18T19:03:12Z"
}
```

weetmuts, 2023-01-02:
Fix pushed, https://wmbusmeters.org/analyze/5244a85c8147602703077a5B0840252f2f0413B39100004413000000004d931e2c73fe0000000000000000000000000000000000000000000000000000000000000000000000009885001a0c002f2f426cBe29

Please test!

Mufika, 2023-01-03:
Thanks. Will try.

closed 2023-04-05 by github-actions[bot]