wmbusmeters/wmbusmeters #747: Exactly the same issue as

madnat, 2022-12-14:
I saw this issue [182](https://github.com/weetmuts/wmbusmeters/issues/182) and see I have what seems to be exactly the same.

I am running NanoCul (868Mhz) with the recommended firmware (nanoCUL_r568_mbus_c1t1_bufsize300.hex), and with 
wmbusmeters --debug --logtelegrams --verbose /dev/ttyUSB0:cul:c1

I receive telegrams with
_telegram) DLL L=5e C=44 (from meter SND_NR) M=11a5 (DME) A=11406109 VER=7c TYPE=07 (Water meter) (driver unknown!) DEV=cul RSSI=-60
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(mfct) added default key
(TPL) num encrypted blocks 5 (80 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "CEE02196E436DB7D1399E5707D507867CBE7E1EED2F9F78952AAF402DE5D29F93A0E99DB0079896D396A574B344005E189C22850CFD2E2341464DA554168248C15F485F5F202BE32298BB263B360BF4F"
(TPL) IV A511096140117C070404040404040404
(TPL) decrypted  "20079D88472C54C679909D079EB47B461B8D0837BB2AB0001EC02609F7974F9079C58E25764C27E6FF1C5C6FFC4CE1EF62F431F77E777DB90E0220D1A58983BA47449D5B67A86AA98F42D72B8E3AC4DA"
(telegram) TPL CI=7a ACC=04 STS=00 CFG=0550 (AES_CBC_IV nb=5 cntn=0 ra=0 hc=0)
Received telegram from: 11406109
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x7c
                device: cul
                  rssi: -60 dBm
                driver: unknown!
(wmbus) 000   : 5e length (94 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : a511 dll-mfct (DME)
(wmbus) 004   : 09614011 dll-id (11406109)
(wmbus) 008   : 7c dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : 04 tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 5005 tpl-cfg 0550 (AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
(wmbus) 015   : 2007 decrypt check bytes (ERROR should be 2f2f)
(wmbus) 017 CE: 9D88472C54C679909D079EB47B461B8D0837BB2AB0001EC02609F7974F9079C58E25764C27E6FF1C5C6FFC4CE1EF62F431F77E777DB90E0220D1A58983BA47449D5B67A86AA98F42D72B8E3AC4DA failed decryption. Wrong key?
telegram=|_5E44A511096140117C077A0400500520079D88472C54C679909D079EB47B461B8D0837BB2AB0001EC02609F7974F9079C58E25764C27E6FF1C5C6FFC4CE1EF62F431F77E777DB90E0220D1A58983BA47449D5B67A86AA98F42D72B8E3AC4DA|+145
(wmbus) telegram from 11406109 ignored by all configured meters!_
 
When I execute:
 wmbusmeters --format=json /dev/ttyUSB0:cul:t1 MyWater izar 11406109 NOKEY

I receive:
_Started config cul on /dev/ttyUSB0 listening on t1
(meter) MyWater: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) Not printing this warning again for id: 11406109 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x7c
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (MyWater 11406109 izar) did not handle telegram!_

Is this a config issue?

Thanks so much for all of your hard work on this interesting project.

weetmuts, 2022-12-30:
This does not seem to be an izar meter... it looks like a normal encrypted water meter, but one that
has not yet been added to wmbusmeters.

Do you have the decryption key? We cannot decode the telegram and add the meter without it.

https://wmbusmeters.org/analyze/5e44a511096140117c077a0400500520079d88472c54c679909d079eB47B461B8d0837BB2aB0001ec02609f7974f9079c58e25764c27e6ff1c5c6ffc4ce1ef62f431f77e777dB90e0220d1a58983Ba47449d5B67a86aa98f42d72B8e3ac4da:izar

madnat, 2022-12-31:
I was under the impression that the meter serial number is the decryption key for these meters?

Here is an image of the meter
[image]
it says it is an Izar meter

BIBOLV, 2023-01-03:
Decryption key is 32 characters logs, so serial number will not fit. You need to get decryption key in order to read values of telegrams and continue with support implementation.

madnat, 2023-01-09:
Ok thanks - I misunderstood some posts from some of the Home Assistant forums.  I have no idea how to access the decryption key, so will investigate other options to read my water meter.

closed 2023-01-09 by madnat

BIBOLV, 2023-01-09:
Try to ask your utility company or whoever owns the meter - sometimes it is sucessful.