wmbusmeters/wmbusmeters #750: Sharky775 support?

KuhnMartin, 2022-12-15:
Hello,

i have a Diehl DE-10-MI004-PTB013 Type 775 installed and try to read it out with an im871a. 

It's really strange that the driver is unknown....

-----------------------------------------------------------------------------
Received telegram from: 51444158
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Heat meter (0x04) encrypted
                   ver: 0x40
                device: im871a[00103776]
                  rssi: -87 dBm
                driver: unknown!
-----------------------------------------------------------------------------
Received telegram from: 51444158
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Heat meter (0x04) encrypted
                   ver: 0x40
                device: im871a[00103776]
                  rssi: -87 dBm
                driver: unknown!
(wmbus) 000   : 5e length (94 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : a511 dll-mfct (DME)
(wmbus) 004   : 58414451 dll-id (51444158)
(wmbus) 008   : 40 dll-version
(wmbus) 009   : 04 dll-type (Heat meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : af tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 5005 tpl-cfg 0550 (AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
(wmbus) 015   : d5f3 decrypt check bytes (ERROR should be 2f2f)
(wmbus) 017 CE: D1932961A129964C681980C491812E8D81F5AE2415FF495D54DCEB08F8EFE63F94A99FE07B2D84A6B14A6767D895DF05CC88342AB7866E507D67C1BECCEA09DBCDFCA193D4069D771C0467C8D314 failed decryption. Wrong key?
(wmbus) telegram from 51444158 ignored by all configured meters!
(serial) received binary "A5C20319442423850760144018A1514900135355B8531892DF2B2737345660EB"
(im871a) checkIM871AFrame "A5C20319442423850760144018A1514900135355B8531892DF2B2737345660EB"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
----------------------------------------------------------------------------------------
wmbusmeters --logtelegrams --format=json auto:t1 heatmeter sharky 51444158 NOKEY
Started auto im871a[00103776] on /dev/ttyUSB0 listening on t1
(meter) heatmeter: meter detection did not match the selected driver sharky! correct driver is: unknown!
(meter) Not printing this warning again for id: 51444158 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Heat meter (0x04) ver: 0x40
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (heatmeter 51444158 sharky) did not handle telegram!
----------------------------------------------------------------------------------------

I thought wmbusmeters is supporting this device? Or is it simply the missing key which I don't know?  Thanks a lot!

BIBOLV, 2022-12-18:
Hi,
Please log some sample telegrams and post them here.

BIBOLV, 2023-01-03:
And key is also missing as you correctly deducted. It is necessary to decrypt telegrams in order to read them.

qwert4u, 2023-01-04:
hello, i also have two diehl meters. one sharky and one RAY heat meter. my diehl water meters (hydrus) work fine. How can I log the telegrams? what information is needed?

qwert4u, 2023-01-04:
(meter) WMZ: meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning again for id: 72698991 mfct: (HYD) Hydrometer (0x2324) type: Heat meter (0x04) ver: 0x44
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (WMZ 72698991 hydrus) did not handle telegram!

qwert4u, 2023-01-04:
my water meters are working, but I get the following message with the devices

Adding meter: WW Küche ...
  File /config/wmbusmeters/etc/mqtt_discovery/hydrus.json not found.

i set in meters :    driver=hydrus

qwert4u, 2023-01-04:
now i have logged a telegram from my sharky 775.
as driver it recognizes the sharky 774. a few values are to be seen. here the result:

https://wmbusmeters.org/analyze/4E44A5113084706140047A030040052F2F0C06500300000C13517107000B3B2424000C2B332401000A5A94040A5E4904CC020664020000CC021345800500C2026CDF2C8C1006000000000AA6180000

the log in wmbusmeters:
Started auto cul on /dev/ttyACM0 listening on t1
(meter) WW gesamt: meter detection did not match the selected driver sharky774! correct driver is: unknown!
(meter) Not printing this warning again for id: 61708430 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Heat meter (0x04) ver: 0x40
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|4E44A5113084706140047A030040052F2F_0C06500300000C13517107000B3B2424000C2B332401000A5A94040A5E4904CC020664020000CC021345800500C2026CDF2C8C1006000000000AA6180000|+0

BIBOLV, 2023-01-04:
To log telegrams you need to add `logtelegrams=false` to config.
Looks like you are running wmbusmeters as ha-addon, in that case for automatic recognition template files are being used, [here](https://github.com/weetmuts/wmbusmeters/tree/master/ha-addon/mqtt_discovery) are ones that are created by community already, but if template for your meter is not there than you can still integrate meter in HA manually as described in documentations. 
Tnx for sharky telegram. Missing data is for previous period? Current values are parsed correctly?

2007M4N, 2023-01-12:
I've also a sharky 775 Meter:
```
(meter) Not printing this warning again for id: 72527464 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Heat meter (0x04) ver: 0x40
(telegram) DLL L=5e C=44 (from meter SND_NR) M=11a5 (DME) A=72527464 VER=40 TYPE=04 (Heat meter) (driver sharky) DEV=cul RSSI=-26
```

I' ve tested it with the latest version: 1.11.0-31-gcf074a1.
If looks like it is working better, since this commit:
  [5d42e94f](https://github.com/weetmuts/wmbusmeters/commit/5688401ca62c75a25ed2af0f285021f65d42e94f)

If I check the telegrams also with the driver=sharky774, the operation time looks also correct for my meter and the value is very plausible.

[https://wmbusmeters.org/analyze/5E44A5116474527240047ABE0050052F2F0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A06020A5E08020A6202F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F:sharky774](https://wmbusmeters.org/analyze/5E44A5116474527240047ABE0050052F2F0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A06020A5E08020A6202F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F:sharky774)
```
  060   : 26 vif (Operating time hours)
  061 C!: 110201 ("operating_time_h":10211)
```

Here are a few more telegrams:

```
sharky 72527464 NOKEY
telegram=|5E44A5116474527240047ABE0050052F2F_0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A06020A5E09020A6202F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F|+22
telegram=|5E44A5116474527240047ABE0050052F2F_0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A07020A5E08020A6201F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F|+9
telegram=|5E44A5116474527240047ABE0050052F2F_0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A07020A5E09020A6202F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F|+24
telegram=|5E44A5116474527240047ABE0050052F2F_0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A06020A5E09020A6202F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F|+108
telegram=|5E44A5116474527240047ABE0050052F2F_0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A06020A5E09020A6202F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F|+56
```

```
telegram=|5E44A5116474527240047ABE0050052F2F_0C06975100008C1006000000000C13849345000C2B000000000B3B0000000A5A06020A5E08020A6202F00B26110201CC020623500000CC021329554400C2026CDF2C2F2F2F2F2F2F2F2F2F2F2F2F|+28
{"media":"heat","meter":"sharky","name":"heatmeter","id":"72527464","total_energy_consumption_kwh":5197,"total_energy_consumption_tariff1_kwh":0,"total_volume_m3":459.384,"volume_flow_m3h":0,"power_kw":0,"flow_temperature_c":20.6,"return_temperature_c":20.8,"temperature_difference_c":-0.2,"timestamp":"2023-01-12T23:09:46Z"}"

```

thanks for the great software and documentation!

weetmuts, 2023-04-18:
Pushed an update and the driver sharky now handles 775 and the new fields that you provided in your telegram!
Please test!

closed 2023-07-18 by github-actions[bot]

fpetursson, 2024-11-20:
i realise this has been stale for a while, but i thought i'd try anyway - and thanks in advance, this is a great project.

Got some telegrams from a Sharky 775, unencrypted (nokey), but wmbusmeters doesn't seem to recognise them properly, seems like a toss-up between a 775 and a 774. I believe it also transmits the differential temperature, which isn't getting decoded it seems.

`3E44A51193681376400C7A0C003005A0E4021B13884D96737AE28A5B8D54B25FFE0755A10EEC26B32AECCD170B4CADC84AC0CB724E7F2FBDB9EE2C4BE541D5`

`3E44A51193681376400C7A0B003005F84D25EFE2A8F2F40C296A8D72C9E42D3D8594854881F7C4FD2359875C7E3E2AD00F95007B9727A0236265CA102FADBA`

`3E44A51193681376400C7A0B00300557167343EEFE3177322B4C3C77F856CE428F20C5A2F0368C1061BACB6EA95DFE3AFB8BF9B1FA3B64E7C3D0AA26A59B91`

cheers.