wmbusmeters/wmbusmeters #759: encoding Diehl HYDRUS

jamsnow, 2022-12-20:
Hello,

i am new in all this topics of wmbusmeters, i read a lot of configuration aso but i am unable to get the wmbusmeters running in the right way.
Since days i am reading different posts everywhere in www but with my low HA knowledge im not able to get it running.
i am using HA for about a year and most of the cases i get the addons started and working, most of the time with copy past and some try and error work.
but in this case i need help from someone, to get the thing started.

as you can see enclosed i can reach the meter, but after i am bit lost. everything i tried was not successfull
I have the key from the meter (can create it by my self) but after i implement it the troubles occur.
official documentation didnt work,
maybe someone can give me more detailed how to 

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Syncing wmbusmeters configuration ...
Registering meters ...
Generating MQTT configuration ... 
Broker 10.0.0.26 will be used.
MQTT Discovery ...
MQTT Discovery cleanup...
Running wmbusmeters ...
Started auto im871a[00103838] on /dev/ttyUSB0 listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 72700014
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x76
                device: im871a[00103838]
                  rssi: -56 dBm
                driver: hydrus

BIBOLV, 2022-12-20:
You need to follow [docs](https://github.com/weetmuts/wmbusmeters/blob/master/ha-addon/DOCS.md) how to add meter config. Provided example is being added by going to addon > Configuration > 3 dots on top right > edit in YAML > add meter config as in example > save > restart addon.

jamsnow, 2022-12-20:
I tried it but did not work

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
parse error: Expected string key before ':' at line 1, column 4
[20:27:47] ERROR: Unknown HTTP error occured
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
parse error: Expected string key before ':' at line 1, column 4
[20:27:48] ERROR: Unknown HTTP error occured
Broker  will be used.
MQTT Discovery cleanup...
Running wmbusmeters ...
Started auto im871a[00103838] on /dev/ttyUSB0 listening on t1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1

Do i need to activate enavle_mqtt_discovery?

BIBOLV, 2022-12-20:
Something with config is not correct, can you post it here?

jamsnow, 2022-12-20:
I only deleted the login and pwd for mqtt

`data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: |-
  loglevel=normal
  device=auto:t1
  donotprobe=/dev/ttyAMA0
  logtelegrams=false
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - |-
    name=MainWater
    driver=hydrus
    id=72700014
    key=2218de76727000142218de7672700014
mqtt:
  host: 10.0.0.26
  port: 1883
  user: .......
  password: ......`

BIBOLV, 2022-12-20:
Try to remove everything from meter config below key=, leaving only first 4 lines - same as in example.

jamsnow, 2022-12-20:
I got the following massage if try to safe

Fehler beim Speichern der Add-on-Konfiguration, Missing option 'mqtt' in root in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': False, 'conf': 'loglevel=normal\ndevice=auto:t1\ndonotprobe=/dev/ttyAMA0\nlogtelegrams=false\nformat=json\nlogfile=/dev/stdout\nshell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"', 'meters': ['name=MainWater\ndriver=hydrus\nid=72700014\nkey=2218de76727000142218de7672700014']

BIBOLV, 2022-12-20:
Try with cfg like this:
```
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
mqtt: true
conf: |-
loglevel=normal
device=auto:t1
donotprobe=/dev/ttyAMA0
logtelegrams=false
format=json
logfile=/dev/stdout
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - |-
    name=MainWater
    driver=hydrus
    id=72700014
    key=2218de76727000142218de7672700014
```

jamsnow, 2022-12-21:
same error as before, unable to save

Fehler beim Speichern der Add-on-Konfiguration, Invalid dict for option 'mqtt' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': False, 'mqtt': True, 'conf': 'loglevel=normal\ndevice=auto:t1\ndonotprobe=/dev/ttyAMA0\nlogtelegrams=false\nformat=json\nlogfile=/dev/stdout\nshell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"', 'meters': ['name=MainWater\ndriver=hydrus\nid=72700014\nkey=2218de76727000142218de7672700014']

jamsnow, 2022-12-21:
after i deleting the mqtt: true i save again and get the following message:

Can't create container from addon_cada3e3b_wmbusmeters: 409 Client Error for http+docker://localhost/v1.41/containers/create?name=addon_cada3e3b_wmbusmeters: Conflict ("Conflict. The container name "/addon_cada3e3b_wmbusmeters" is already in use by container "baabde3596f981b5885b20614a9fe8fd4293910cef28913a6af017b2c8ca67df". You have to remove (or rename) that container to be able to reuse that name.")

jamsnow, 2022-12-21:
So i have checked a a different encryption key as well as a differnent version of Hydrus.
There are Hydrus Version 1.x and Hydrus 2.0 
For Hydrus 2.0 i got the driver hydrus but something is wrong with the readout and encryption.
For Hydrus 1.x i got the driver unknown! so no encryption possible

Did anyone of you know for which version of Hydrus the driver is writen?

BIBOLV, 2022-12-22:
I do not know about version but you can read trough other issues opened/closed here - most of information is available there.

Please try this config.
```
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
mqtt: {}
conf: |-
loglevel=normal
device=auto:t1
donotprobe=/dev/ttyAMA0
logtelegrams=false
format=json
logfile=/dev/stdout
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - |-
    name=MainWater
    driver=hydrus
    id=72700014
    key=2218de76727000142218de7672700014
```

jamsnow, 2023-01-13:
Hello, 
after reading a lof of the issues topics i get the meters running.
The reason for the problems where the wrong config and data path

The config which is working for me:

data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: |-
  loglevel=debug
  device=/dev/ttyUSB0:im871a[00103838]:t1
  donotprobe=/dev/ttyAMA0
  logtelegrams=false
  format=json
  logfile=/config/wmbusmeters/wmbusmeters.log
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=MainWater
    driver=hydrus
    id=XXXXXXXX
    key=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
mqtt:
  host: XX.X.X.XX
  port: 1883
  user: XXXXXXXX
  password: XXXXXXX

BIBOLV, 2023-01-19:
Can we close this issue?

jamsnow, 2023-01-19:
Yes for sure

closed 2023-01-19 by BIBOLV

hummermann87, 2024-12-08:
@jamsnow: How do you generate the key? It seems to me that I understand SOME of the structure, however, it would be interesting for me, as I am also trying to get a key for my Hydrus 171A.