wmbusmeters/wmbusmeters #762: Docker pull on Raspberry pi 4

agrauballe, 2022-12-23:
Hi,
Just tried to pull the docker image using docker-compose.

I get the error that there is no manifest for armv7, anybody else having this issue?
I'm running the 32bit buster distribution
uname -m
armv7l

Pulling wmbusmeters ... pulling from weetmuts/wmbusmeters

ERROR: for wmbusmeters  no matching manifest for linux/arm/v7 in the manifest list entries
ERROR: no matching manifest for linux/arm/v7 in the manifest list entries

docker-compose.yaml
version: "2"
services:
  wmbusmeters:
    image: weetmuts/wmbusmeters
    container_name: wmbusmeters
    restart: always
    volumes:
      - /opt/wmbusmeters:/wmbusmeters_data
      - /etc/localtime:/etc/localtime:ro

Boldfor, 2022-12-29:
Just encountered the same when I tried to re-deploy my wmbusmeters container. For some reason (am also not a pro) I had to add `platform: linux/armv7` (am running a Raspi 3B+) to make it work. Hope this helps.

@weetmuts , how about adding the required parameter-options (`platform: linux/armv7` for ARMv7 32-bit, `platform: linux/arm64` for ARMv8 64-bit, `platform: linux/armv7` for Linux x86-64) into the docker-compose.yml-example [here](https://github.com/weetmuts/wmbusmeters/tree/master/docker)? It's implicitely stated in your documentation, but maybe explicitly including it into the docker-compose.yml could help. I have to admit I'm not a pro, but from what I can see, there are others who struggled here as well.

weetmuts, 2022-12-29:
@BIBOLV Can you help out here?

agrauballe, 2023-01-01:
Inserting `platform: linux/armv7` did make it work, other docker images does not require this setting so maybe there is a way to "auto detect". However it can be added to the documentation if we need to define the platform.

BIBOLV, 2023-01-01:
I will look into this.

BIBOLV, 2023-01-19:
Fix has been pushed - please test if it works for you.

closed 2023-01-19 by agrauballe

Boldfor, 2023-01-19:
Thanks to everyone involved. :)