wmbusmeters/wmbusmeters #764: Config file gets overwritten each start of the HA addon

Gahrlaag, 2022-12-25:
So when using wmbusmeters in HA, I cannot change the config line in the addon, since it writes a single line into the wmbusmeters.conf but what's also strange is that it overwrites the config file each time I start the addon (even after changing to r-r-r on the file with chmod).
I'm trying to use a nano-cul with 1.67 firmware installed from the vendor but it doesn't support t1 mode.

weetmuts, 2022-12-25:
I am trying to do the same thing. I am a complete noob on the HA addon, but we need some automatic testing. 

@to-masz Can you help us? I am trying to do the same thing. I have managed to install a kvm version of HA and added the wmbusmeters add-on, needed to restart the HA before it showed up....

Now I want to change the local wmbusmeters.conf I want to change device to:
device=rtlwmbus:CMD(nc -lku 4444) 
to try out feeding simulated date into HA.

Gahrlaag, 2022-12-25:
Managed to get everything up and running by going into yaml mode instead of the visual editor

weetmuts, 2022-12-25:
@Gahrlaag Where is the yaml mode and/or editor in the gui?

Gahrlaag, 2022-12-25:
Click on the e dots in the right corner in the config screen :)

weetmuts, 2022-12-25:
@Gahrlaag Woho! I am learning! Thanks!

kennethnilssen, 2022-12-30:
Same issue on my side, and if I have anything else than "device=*" in the config, it throws an error in the log. 
Also, in the HA addon, would be good with a guide on how to decrypt the KEM file for my Kamstrup addon, as all the guides point to decrypting this within an environment that is not HA, if possible @weetmuts

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

BIBOLV, 2023-04-01:
@kennethnilssen - KEM file processing added to edge ha addon

closed 2023-07-02 by github-actions[bot]