wmbusmeters/wmbusmeters #766: Add support for Q caloric 5.5

rocket-plane, 2022-12-26:
Hi

it would be nice, if support for the Q caloric 5.5 could be added. Using the qcaloric driver seems not to be working. Some of the messages are shown as enrcypted, others as unencrypted.

Meternumber: 30535282
Meter Reading: 00034

I have attached some a log files which hopefully shows the necessary telegrams
[telegram.txt](https://github.com/weetmuts/wmbusmeters/files/10303563/telegram.txt)

klassenpeter, 2023-01-08:
Hi,
currently i alsolisten to my qcaloric 5.5 devices.
@rocket-plane  They come in different Types and versions (S-Type /C-Type), maybe that is of interest. 

I have the (quite new) Devices, typed with:
P2-AMR WB.5
HCA5xxxT (C-Mode (Q AMR, Q walk-by) ).
according to this [data sheet](https://qundis.com/wp-content/uploads/2022/10/DST5-00CA-GB0-CAL55-A_01082022.pdf)

what I observed so far:

They send out two different telegrams with 0x31 and 0x49 bytes of length.
Only the shorter telegrams are  decoded correctly, though the `set_date` seems odd.

if somebody is interested, i could provide more telegrams, i've collected so far. 

```
/wmbusmeters # ./wmbusmeters --version
wmbusmeters: 1.11.0-25-gf6345e4
f6345e4c461ab41f44624b0fb5dd393f74c114d8

/wmbusmeters # ./wmbusmeters --analyze=qcaloric 314493442842003036087ac20000200b6e5704004b6e000000426cffffcb086e230400c2086cdf2c326cffff046d0210e821
Auto driver  : not found!
Best driver  : qcaloric 19/19
Using driver : qcaloric 00/00
000   : 31 length (49 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 9344 dll-mfct (QDS)
004   : 28420030 dll-id (30004228)
008   : 36 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : c2 tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 0B dif (6 digit BCD Instantaneous value)
016   : 6E vif (Units for H.C.A.)
017 C!: 570400 ("current_consumption_hca":457)
020   : 4B dif (6 digit BCD Instantaneous value storagenr=1)
021   : 6E vif (Units for H.C.A.)
022 C!: 000000 ("consumption_at_set_date_hca":0) ("consumption_at_set_date_1_hca":0)
025   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
026   : 6C vif (Date type G)
027 C!: FFFF ("set_date":"2127-15-31") ("set_date_1":"2127-15-31")
029   : CB dif (6 digit BCD Instantaneous value storagenr=1)
030   : 08 dife (subunit=0 tariff=0 storagenr=17)
031   : 6E vif (Units for H.C.A.)
032 C!: 230400 ("consumption_at_set_date_17_hca":423)
035   : C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
036   : 08 dife (subunit=0 tariff=0 storagenr=17)
037   : 6C vif (Date type G)
038 C!: DF2C ("set_date_17":"2022-12-31")
040   : 32 dif (16 Bit Integer/Binary Value during error state)
041   : 6C vif (Date type G)
042 C!: FFFF ("error_date":"2127-15-31")
044   : 04 dif (32 Bit Integer/Binary Instantaneous value)
045   : 6D vif (Date and time type)
046 C!: 0210E821 ("device_date_time":"2023-01-08 16:02")

{
    "media":"heat cost allocation",
    "meter":"qcaloric",
    "name":"",
    "id":"30004228",
    "status":"OK",
    "current_consumption_hca":457,
    "set_date":"2127-15-31",
    "consumption_at_set_date_hca":0,
    "set_date_1":"2127-15-31",
    "consumption_at_set_date_1_hca":0,
    "set_date_17":"2022-12-31",
    "consumption_at_set_date_17_hca":423,
    "error_date":"2127-15-31",
    "device_date_time":"2023-01-08 16:02",
    "timestamp":"2023-01-08T16:07:46Z"
}

/wmbusmeters # ./wmbusmeters --analyze=qcaloric 49449344284200303608780dff5f350082c50000800007b06effff58040000ffff00000000df2c23040000008000800080008000800000000000003b003d006300cc002f046d1e10e821
Auto driver  : not found!
Best driver  : bfw240radio 07/66
Using driver : qcaloric 00/00
000   : 49 length (73 bytes)(OK)
001   : 44 dll-c (from meter SND_NR)
002   : 9344 dll-mfct (QDS)
004   : 28420030 dll-id (30004228)
008   : 36 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
011   : 0D dif (variable length Instantaneous value)
012   : FF vif (Manufacturer specific)
013   : 5F vife (?)
014   : 35 varlen=53
015 C?: 0082C50000800007B06EFFFF58040000FFFF00000000DF2C23040000008000800080008000800000000000003B003D006300CC002F
068   : 04 dif (32 Bit Integer/Binary Instantaneous value)
069   : 6D vif (Date and time type)
070 C!: 1E10E821 ("device_date_time":"2023-01-08 16:30")

{
    "media":"heat cost allocation",
    "meter":"qcaloric",
    "name":"",
    "id":"30004228",
    "status":"OK",
    "device_date_time":"2023-01-08 16:30",
    "timestamp":"2023-01-08T16:08:04Z"
}

```
chers
Peter

rocket-plane, 2023-01-09:
Hi,

I had a look at my meter. It's written 
P2-AMR WBC5  
HCA5xxx8
on it. According to [this](https://qundis.de/wp-content/uploads/2020/07/QUNDIS_Preisliste_2020_V3.0_01072020.pdf) that should be the same as your one (despite the missing IrDA) (C-Mode Q walk-by and Q AMR).

I have two telegrams: one with 0x49 length and one with 0x38 length. But the 0x38 is encrypted. Is there a chance to get the information out of the 0x49 telegram?

klassenpeter, 2023-01-09:
@rocket-plane, well the payload from the longer telegram which you send looks random to me, but i am no expert.
in the longer telegram, i see my current consuption at the following position:
```
your data:
49449344825253303608780dff5f350082420035dcae4a6f6df70218a88273844e3a6122037ec15b1d8077028604a0632eba0c9de20d8f98def1e2e6e0306b784ec178ad046d1f0dda2c
---my data--------------------------------------------|here|----------
49449344284200303608780dff5f350082c50000800007b06effff58040000ffff00000000df2c23040000008000800080008000800000000000003b003d006300cc002f046d1e10e821
```
in your case it looks random, so i guess is also encrypted, but once again! i am not an expert,
since my knowledge is based on only  2 hours of skimming through and not understanding some webpages

klassenpeter, 2023-01-09:
@rocket-plane  @BIBOLV i guess the issue may be closed?

currently i am looking into these longer telegrams with manufacturer specific values.
if i manage to decode more of  them. I'll let you know.

```
49449344284200303  boring
608                new device, that is not present in the code yet, right? 
780dff5f35         header
0082c50000800007   no idea
b06e               no idea
ffff               maybe error date?
58040000           current hca reading
ffff               set_date_1
00000000           consumption_at_set_date_1_hca
df2c2              date  last month 
3040               consumption_last_month_date_1_hca
000008000800080008000800000000000003b003d006300cc002f046d   no idea, my guess is: usage of last 12 to13 months, since this is stated in the datasheet
1e10e821           timestamp
```

BIBOLV, 2023-01-12:
@weetmuts - what`s your take on this? Two type of telegrams from meter, one can be decoded already, other not.

rocket-plane, 2023-01-12:
@klassenpeter Yes I guess the issue may be closed, because the meter is somewat supported, but only my telegrams seem to be encrypted. 

Is there a flag in the telegram that clearly indicates if a message is encrypted? Or is that a kind of heuristik?

klassenpeter, 2023-01-12:
@rocket-plane I looked once again into the telegrams you posted.
At least for the shorter diagrams `wmbusmeters` tells you
```
(TPL) num encrypted blocks 2 (32 bytes and remaining unencrypted 10 bytes)
```

It is this telegram you posted:
``
0x384493448252533036087a430020253f59515bd90f76e8576af36c988eea9b398ec5c205e5dbbe3f2698408947cb8e326cffff046d230dda2c
```
further below you see:
```
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : 43 tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 2025 tpl-cfg 2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
```
for the longer telegrams.. no idea..

You could try and ask the company which operates the meters, maybe you are lucky and they
hand out the key. You could argue that knowing your consumption enables you to save energy and
thus reduce your personal environmental impact. Or.. if you're located in the EU.. saving gas is currently
also a valid argument.
cheers

klassenpeter, 2023-01-12:
> @weetmuts - what`s your take on this? Two type of telegrams from meter, one can be decoded already, other not.

my motivation to decode the longer telegrams is: according to the manual/data sheet, they contain statistical data (usage of the previous 13 Months). I'll have a look into this next month, when the data changes.
I case i'm able to decode them.. i could open a new issue.

BIBOLV, 2023-01-12:
Ok, closing for, please reopen this or open new issue if you are able to decrypt telegram.

closed 2023-01-12 by BIBOLV

weetmuts, 2023-01-13:
Ok, I added the missing mfct/type/version to the qcaloric driver lookup and pushed it. Thanks for the telegrams @klassenpeter and @rocket-plane !

wmbusmeters currently does not decode and print the readable bytes at the end of the qcaloric encrypted telegrams. This is something to fix for the future, the information is not that useful for qcaloric.

The readable bytes are: 326cffff046d230dda2c which decodes to
```
32 dif (16 Bit Integer/Binary Value during error state)
6C vif (Date type G)
FFFF
04 dif (32 Bit Integer/Binary Instantaneous value)
6D vif (Date and time type)
230DDA2C
32 dif (16 Bit Integer/Binary Value during error state)
6C vif (Date type G)
FFFF ("error_date":"2127-15-31")
04 dif (32 Bit Integer/Binary Instantaneous value)
6D vif (Date and time type)
230DDA2C ("device_date_time":"2022-12-26 13:35")
```