wmbusmeters/wmbusmeters #769: Failed to install add-on  / returned a non-zero code: 7

pioootrek, 2022-12-28:
Hello Team,

I'm trying to install addon with GUI of haos.

GUI asys that it wants to install 1.10.2

After fwe seconds I get following error:

`The command '/bin/ash -o pipefail -c apk add --no-cache alpine-sdk gcc linux-headers ncurses-dev librtlsdr-dev cmake libusb-dev' returned a non-zero code: 7`
[image]

My HAOS:

`Home Assistant 2022.12.7
Supervisor 2022.11.2
Operating System 9.4
Frontend 20221213.1 - latest`

Any ideas appriciated!

BIBOLV, 2023-01-02:
Hi,
Retry multiple times, some restarts, etc - sometimes it happens that docker build fails.

pioootrek, 2023-01-04:
I tried it multiple times, with resets, etc :/

in logs I found only temporary error related to aplinelinux repo. I checked manualy from cli, and that url is reachable :/

`23-01-04 08:57:37 INFO (SyncWorker_6) [supervisor.docker.addon] Starting build for cada3e3b/amd64-addon-wmbusmeters:1.11.0
23-01-04 08:57:49 ERROR (SyncWorker_6) [supervisor.docker.addon] Can't build cada3e3b/amd64-addon-wmbusmeters:1.11.0: The command '/bin/ash -o pipefail -c apk add --no-cache alpine-sdk gcc linux-headers ncurses-dev librtlsdr-dev cmake libusb-dev' returned a non-zero code: 7
23-01-04 08:57:49 ERROR (SyncWorker_6) [supervisor.docker.addon] Build log: 
Step 1/30 : ARG BUILD_FROM
Step 2/30 : FROM $BUILD_FROM AS build
 ---> 63d7ea64acdd
Step 3/30 : ENV LANG C.UTF-8
 ---> Using cache
 ---> 3c4acf67ee0c
Step 4/30 : RUN apk add --no-cache alpine-sdk gcc linux-headers ncurses-dev librtlsdr-dev cmake libusb-dev
 ---> Running in c66028797728
fetch https://dl-cdn.alpinelinux.org/alpine/v3.16/main/x86_64/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/v3.16/community/x86_64/APKINDEX.tar.gz
WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.16/main: temporary error (try again later)

WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.16/community: temporary error (try again later)

ERROR: unable to select packages:

  alpine-sdk (no such package):
    required by: world[alpine-sdk]
  cmake (no such package):
    required by: world[cmake]
  gcc (no such package):
    required by: world[gcc]
  librtlsdr-dev (no such package):
    required by: world[librtlsdr-dev]
  libusb-dev (no such package):
    required by: world[libusb-dev]
  linux-headers (no such package):
    required by: world[linux-headers]
  ncurses-dev (no such package):
    required by: world[ncurses-dev]
Removing intermediate container c66028797728`

adibenz, 2023-01-04:
Hello

Do you have the addon "dnsmasq" installed? 
Please check the DNS settings of Home Assistant.
I had the same problem, when the dns server was set to the IP of Home Assistant/dnsmasq

I have now entered the router IP as DNS server and was able to install wmbusmeters

pioootrek, 2023-01-05:
Perfect. I was using ADGuard Home, and changing ip address of DNS from local dns to public google solved my problem. 
THX

closed 2023-01-05 by pioootrek