wmbusmeters/wmbusmeters #772: Qundis water meter, not receiving data

hackex, 2022-12-30:
Hi everyone,
I found this awesome software and wanted to try it out. I could receive some telegrams of a qundis water meter (I don't know where it is located though). I know the ID and driver (it is showing in the logs). I then configured the HA addon and are getting some logs out, but the MQTT sensor in HA still shows only the value 0m3.
Here is the logs:
`shell) env "METER_DUE_DATE_17="
(shell) env "METER_ERROR_CODE="
(shell) env "METER_ERROR_DATE="
(shell) env "METER_DEVICE_DATE_TIME=2022-12-30 14:08"
(shell) waiting for child 266 to complete.
Error: Address not available
(shell) /bin/sh: return code 1
(shell) /bin/sh exited with non-zero return code: 1
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2022-12-30 13:07:51.000;42;12;12430972;0x49449344720943121806780dff5f3500826c0000ee0007c113ffff70380100bf2c06120000de2b0625010077009b008b00860084004a00580044004f00180073007a002f046d080ede2c<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=74.
(serial) received ascii "C1;1;1;2022-12-30 13:09:41.000;28;9;12430972;0x49449344720943121806780dff5f3500826c0000800007c113ffff70380100bf2c06120000de2b0625010077009b008b00860084004a00580044004f00180073007a002f046d0a0ede2c<0A>C1;1;1;2022-12-30 13:09:41.000;48;12;12430972;0x49449344720943121806780dff5f3500826c0000800007c113ffff70380100bf2c06120000de2b0625010077009b008b00860084004a00580044004f00180073007a002f046d0a0ede2c<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2022-12-30 13:09:41.000;28;9;12430972;0x49449344720943121806780dff5f3500826c0000800007c113ffff70380100bf2c06120000de2b0625010077009b008b00860084004a00580044004f00180073007a002f046d0a0ede2c<0A>C1;1;1;2022-12-30 13:09:41.000;48;12;12430972;0x49449344720943121806780dff5f3500826c0000800007c113ffff70380100bf2c06120000de2b0625010077009b008b00860084004a00580044004f00180073007a002f046d0a0ede2c<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 74
(wmbus) parseELL @10 64
(wmbus) parseNWL @10 64
(wmbus) parseAFL @10 64
(wmbus) parseTPL @10 64
(meter) watermeter: for me? 12430972 in 12430972
(meter) watermeter: yes for me
(meter) watermeter(1) qwater  handling telegram from 12430972
(meter) watermeter 12430972 "49449344720943121806780DFF5F3500826C0000800007C113FFFF70380100BF2C06120000DE2B0625010077009B008B00860084004A00580044004F00180073007A002F046D0A0EDE2C"
(wmbus) parseDLL @0 74
(telegram) DLL L=49 C=44 (from meter SND_NR) M=4493 (QDS) A=12430972 VER=18 TYPE=06 (Warm Water (30°C-90°C) meter) (driver qwater) DEV=rtlwmbus[00000001] RSSI=28
(wmbus) parseELL @10 64
(wmbus) parseNWL @10 64
(wmbus) parseAFL @10 64
(wmbus) parseTPL @10 64
(telegram) TPL CI=78
telegram=|4944934472094312180678_0DFF5F3500826C0000800007C113FFFF70380100BF2C06120000DE2B0625010077009B008B00860084004A00580044004F00180073007A002F046D0A0EDE2C|+182
(meters) trying field info total(Volume)[0]...
(meters) trying field info due_date(Volume)[1]...
(meters) trying field info due_date(Text)[2]...
(meters) trying field info due_date_17(Volume)[3]...
(meters) trying field info due_date_17(Text)[4]...
(meters) trying field info error_code(Text)[5]...
(meters) trying field info error_date(Text)[6]...
(meters) trying field info device_date_time(Text)[7]...
(meters) using field info device_date_time(Text)[7] to extract 046D at offset 70
(qwater) 000   : 49 length (73 bytes)
(qwater) 001   : 44 dll-c (from meter SND_NR)
(qwater) 002   : 9344 dll-mfct (QDS)
(qwater) 004   : 72094312 dll-id (12430972)
(qwater) 008   : 18 dll-version
(qwater) 009   : 06 dll-type (Warm Water (30°C-90°C) meter)
(qwater) 010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(qwater) 011   : 0D dif (variable length Instantaneous value)
(qwater) 012   : FF vif (Manufacturer specific)
(qwater) 013   : 5F vife (?)
(qwater) 014   : 35 varlen=53
(qwater) 015 C?: 00826C0000800007C113FFFF70380100BF2C06120000DE2B0625010077009B008B00860084004A00580044004F00180073007A002F
(qwater) 068   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(qwater) 069   : 6D vif (Date and time type)
(qwater) 070 C!: 0A0EDE2C ("device_date_time":"2022-12-30 14:10")
(meters) render field total(Volume)[0] without dventry
(meters)             "total_m3":0
(meters) render field due_date(Volume)[1] without dventry
(meters)             "due_date_m3":0
(meters) render field due_date(Text)[2] without dventry
(meters)             "due_date":""
(meters) render field due_date_17(Volume)[3] without dventry
(meters)             "due_date_17_m3":0
(meters) render field due_date_17(Text)[4] without dventry
(meters)             "due_date_17":""
(meters) render field error_code(Text)[5] without dventry
(meters)             "error_code":""
(meters) render field error_date(Text)[6] without dventry
(meters)             "error_date":""
(meters) render field device_date_time(Text txt)[7] with dventry @70 key 046D data 0A0EDE2C
(meters)             "device_date_time":"2022-12-30 14:10"
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/"$METER_ID" -m "$METER_JSON""
(shell) env "METER_JSON={"media":"warm water","meter":"qwater","name":"watermeter","id":"12430972","total_m3":0,"due_date_m3":0,"due_date":"","due_date_17_m3":0,"due_date_17":"","error_code":"","error_date":"","device_date_time":"2022-12-30 14:10","timestamp":"2022-12-30T13:09:41Z","device":"rtlwmbus[00000001]","rssi_dbm":28}"
(shell) env "METER_ID=12430972"
(shell) env "METER_NAME=watermeter"
(shell) env "METER_MEDIA=warm water"
(shell) env "METER_TYPE=qwater"
(shell) env "METER_TIMESTAMP=2022-12-30T13:09:41Z"
(shell) env "METER_TIMESTAMP_UTC=2022-12-30T13:09:41Z"
(shell) env "METER_TIMESTAMP_UT=1672405781"
(shell) env "METER_TIMESTAMP_LT=2022-12-30 14:09.41"
(shell) env "METER_DEVICE=rtlwmbus[00000001]"
(shell) env "METER_RSSI_DBM=28"
(shell) env "METER_TOTAL_M3=0"
(shell) env "METER_DUE_DATE_M3=0"
(shell) env "METER_DUE_DATE="
(shell) env "METER_DUE_DATE_17_M3=0"
(shell) env "METER_DUE_DATE_17="
(shell) env "METER_ERROR_CODE="
(shell) env "METER_ERROR_DATE="
(shell) env "METER_DEVICE_DATE_TIME=2022-12-30 14:10"
Error: Address not available
(shell) waiting for child 377 to complete.
(shell) /bin/sh: return code 1
(shell) /bin/sh exited with non-zero return code: 1
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2022-12-30 13:09:41.000;48;12;12430972;0x49449344720943121806780dff5f3500826c0000800007c113ffff70380100bf2c06120000de2b0625010077009b008b00860084004a00580044004f00180073007a002f046d0a0ede2c<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=74.`

Can anyone tell me what is not working? Is there a key missing maybe? I have got none, but how do I see that the key is neccesarry?

Thanks

hackex, 2022-12-30:
I can see some total_m3":13.88 in the logs....is it true that it's not a encrypted data flow?

How do I get the value over MQTT to HA?

weetmuts, 2022-12-30:
If you see a value that matches your meter, then it is working. :-) 
Your meter sends different telegrams the ones above look like this:
https://wmbusmeters.org/analyze/49449344720943121806780dff5f3500829100005e0107c113ffff80380100bf2c06120000de2b0625010077009b008b00860084004a00580044004f00180073007a002f046d2412de2c

Ie. it is the proprietary version of the telegram. Listen for more telegrams and try them out in the analyzer.
When you hit your meter and the right telegram it should show values, or it should show a different
red text telling you that it is encrypted.

hackex, 2022-12-31:
> If you see a value that matches your meter, then it is working. :-) Your meter sends different telegrams the ones above look like this: [...]

Okay, seems to work now. I receive the value also in Home Assistant. Now, I just need to find the meter which is sending the data. Time for some detective work ;)

In default configuration all the devices sending data were listed in the logs...how can I see them while having a meter already configured?
I saw some Techem heat allocators, I would like to track as well.

BIBOLV, 2023-01-19:
Enable debug logging.

closed 2023-04-21 by github-actions[bot]