wmbusmeters/wmbusmeters #773: driver auto and wildcard id not working anymore for izar R4 (izar) and G4 (dme_07)

ldebomy, 2022-12-30:
version 1.7 works fine for receiving data from all R4 and G4 meters in range using this command :
wmbusmeters 'rtlwmbus::CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)' R auto \* NOKEY

but version 1.10 returns :

build/wmbusmeters --debug --verbose 'rtlwmbus::CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)' R auto \* NOKEY
(wmbusmeters) version: _
(config) using device: rtlwmbus:CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s) 
(config) number of meters: 1
(serial) expecting devices to work
(serial) expecting devices to work
(lookup) with cmd "rtlwmbus:CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)"
(main) opening rtlwmbus:CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)
Started config rtlwmbus listening on none using CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)
(rtlwmbus) using command: rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s
(rtlwmbus) opening cmd_0
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s"
(serialcmd) opened /bin/sh pid 958891 fd 3 (rtlwmbus)
(main) regular reset of rtlwmbus  using CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s) will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) received ascii "T1;1;1;2022-12-30 13:15:24.000;106;124;14600781;0x1e44a511810760147b077a2c731005408f1f96aa3c0c1063bca154d9ae1097<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-12-30 13:15:24.000;106;124;14600781;0x1e44a511810760147b077a2c731005408f1f96aa3c0c1063bca154d9ae1097<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(meter) R: for me? 14600781 in 
(meter) ignoring telegram from 14600781 since it matched a wildcard id rule but driver (auto) does not match.
(wmbus) telegram from 14600781 ignored by all configured meters!
(serial) received ascii "T1;1;1;2022-12-30 13:15:24.000;123;126;14600781;0x1e44a511810760147b077a2c731005408f1f96aa3c0c1063bca154d9ae1097<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-12-30 13:15:24.000;123;126;14600781;0x1e44a511810760147b077a2c731005408f1f96aa3c0c1063bca154d9ae1097<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=31.
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(shell) sending SIGTERM to process group 958891
(bgshell) sent sigterm, now waiting for child 958891 to exit.
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(bgshell) 958891 terminated due to signal 15
(wmbus) disconnected cmd_0 rtlwmbus
(serialcmd) closed /bin/sh pid=958891 fd=3 (rtlwmbus)
(wmbus) deleted rtlwmbus
(serial) closing 1 devices
(serial) no devices working emergency exit!

it works only if I specify driver to use in 1.10 :

build/wmbusmeters --debug --verbose 'rtlwmbus::CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)' R dme_07 \* NOKEY
(wmbusmeters) version: _
(config) using device: rtlwmbus:CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s) 
(config) number of meters: 1
(serial) expecting devices to work
(serial) expecting devices to work
(lookup) with cmd "rtlwmbus:CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)"
(main) opening rtlwmbus:CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)
Started config rtlwmbus listening on none using CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s)
(rtlwmbus) using command: rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s
(rtlwmbus) opening cmd_0
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s"
(serialcmd) opened /bin/sh pid 905713 fd 3 (rtlwmbus)
(main) regular reset of rtlwmbus  using CMD(rtl_sdr -p 0 -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s) will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) received ascii "T1;1;1;2022-12-30 13:05:08.000;122;125;14600781;0x1e44a511810760147b077a2b731005c31478d5751f0671b96eeb9e5774a587<0A>T1;1;1;2022-12-30 13:05:08.000;132;130;14600781;0x1e44a511810760147b077a2b731005c31478d5751f0671b96eeb9e5774a587<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-12-30 13:05:08.000;122;125;14600781;0x1e44a511810760147b077a2b731005c31478d5751f0671b96eeb9e5774a587<0A>T1;1;1;2022-12-30 13:05:08.000;132;130;14600781;0x1e44a511810760147b077a2b731005c31478d5751f0671b96eeb9e5774a587<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(meter) R: for me? 14600781 in 
(meter) R: yes for me
(meter) created R dme_07 14600781 encrypted
(meter) used meter template R  dme_07 to match 14600781
(meter) started meter 1 (R 14600781 dme_07)
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(meter) R: for me? 14600781 in 14600781
(meter) R: yes for me
(meter) R(1) dme_07  handling telegram from 14600781
(meter) R 14600781 "1E44A511810760147B077A2B731005C31478D5751F0671B96EEB9E5774A587"
(wmbus) parseDLL @0 31
(telegram) DLL L=1e C=44 (from meter SND_NR) M=11a5 (DME) A=14600781 VER=7b TYPE=07 (Water meter) (driver dme_07) DEV=rtlwmbus[cmd_0] RSSI=122
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(mfct) added default key
(TPL) num encrypted blocks 1 (16 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "C31478D5751F0671B96EEB9E5774A587"
(TPL) IV A511810760147B072B2B2B2B2B2B2B2B
(TPL) decrypted  "2F2F0413BC00000002FD1703002F2F2F"
(dvparser) found new format "041302FD17" with hash f216, remembering!
(telegram) TPL CI=7a ACC=2b STS=73 CFG=0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
(meters) trying field info status(Text)[0]...
(meters) using field info status(Text)[0] to extract 02FD17 at offset 26
(meters) trying field info total(Volume)[1]...
(meters) using field info total(Volume)[1] to extract 0413 at offset 19
(meter) Volume total_m3 decoded m3 default m3 value 0.188
(dme_07) 000   : 1e length (30 bytes)
(dme_07) 001   : 44 dll-c (from meter SND_NR)
(dme_07) 002   : a511 dll-mfct (DME)
(dme_07) 004   : 81076014 dll-id (14600781)
(dme_07) 008   : 7b dll-version
(dme_07) 009   : 07 dll-type (Water meter)
(dme_07) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(dme_07) 011   : 2b tpl-acc-field
(dme_07) 012   : 73 tpl-sts-field (ALARM TEMPORARY_ERROR)
(dme_07) 013   : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
(dme_07) 015   : 2f2f decrypt check bytes (OK)
(dme_07) 017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(dme_07) 018   : 13 vif (Volume l)
(dme_07) 019 C!: BC000000 ("total_m3":0.188)
(dme_07) 023   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(dme_07) 024   : FD vif (Second extension FD of VIF-codes)
(dme_07) 025   : 17 vife (Error flags (binary))
(dme_07) 026 C!: 0300 ("status":"ERROR_FLAGS_3")
(dme_07) 028   : 2F skip
(dme_07) 029   : 2F skip
(dme_07) 030   : 2F skip
(meters) render field status(Text txt)[0] with dventry @26 key 02FD17 data 0300
(meters)             "status":"ERROR_FLAGS_3"
(meters) render field total(Volume m3)[1] with dventry @19 key 0413 data BC000000
(meters)             "total_m3":0.188
R       14600781        0.188 m³        ERROR_FLAGS_3   2022-12-30 14:05.08
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-12-30 13:05:08.000;132;130;14600781;0x1e44a511810760147b077a2b731005c31478d5751f0671b96eeb9e5774a587<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=31.
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(shell) sending SIGTERM to process group 905713
(bgshell) sent sigterm, now waiting for child 905713 to exit.
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(bgshell) 905713 terminated due to signal 15
(wmbus) disconnected cmd_0 rtlwmbus
(serialcmd) closed /bin/sh pid=905713 fd=3 (rtlwmbus)
(wmbus) deleted rtlwmbus
(serial) closing 1 devices
(serial) no devices working emergency exit!

BIBOLV, 2023-01-12:
Please retest with latest version - wildcard functionality was [restored](https://github.com/weetmuts/wmbusmeters/issues/786).

BIBOLV, 2023-01-19:
Can you test and confirm that everything works as expected now and we can close this issue?

ldebomy, 2023-01-20:
Ok, it works again. Thank you !

closed 2023-01-20 by BIBOLV