wmbusmeters/wmbusmeters #774: Run docker in Synology NAS ?

softeg-klee, 2022-12-30:
Hello,

I have a synology NAS with USB plugged (RTL2838 DVB-T - Realtek Semiconductor Corp w) and i want to run the docker version, but i have this issue : 

`/wmbusmeters/wmbusmeters --debug auto:t1 
`

`
(wmbusmeters) version: 1.11.0-9-g8110c90
(config) number of meters: 0
(main) device /dev/ttyACM0 not currently used, detect contents...
(serial) check if /dev/ttyACM0 can be accessed
(serial) cannot open/find tty /dev/ttyACM0.
(main) ignoring /dev/ttyACM0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyS0 not currently used, detect contents...
(serial) check if /dev/ttyS0 can be accessed
(serial) cannot open/find tty /dev/ttyS0.
(main) ignoring /dev/ttyS0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyS1 not currently used, detect contents...
(serial) check if /dev/ttyS1 can be accessed
(serial) cannot open/find tty /dev/ttyS1.
(main) ignoring /dev/ttyS1, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyS2 not currently used, detect contents...
(serial) check if /dev/ttyS2 can be accessed
(serial) cannot open/find tty /dev/ttyS2.
(main) ignoring /dev/ttyS2, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyS3 not currently used, detect contents...
(serial) check if /dev/ttyS3 can be accessed
(serial) cannot open/find tty /dev/ttyS3.
(main) ignoring /dev/ttyS3, it does not respond as any of the supported wmbus devices.
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 18686 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(shell) return code 0
(shell) 18687 exited
(serial) EVENT thread interrupted
(main) rtl_wmbus found in path
(main) rtlsdr device x¬õþ not currently used.
(main) ignoring rtlsdr x¬õþ since it is unavailable.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 18690 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>18680 8
<<<
(shell) return code 0
(shell) 18691 exited
Notice! Other wmbusmeters (pid 8) is running and it might hog any wmbus devices.
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
(main) rtlsdr device  not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 18712 fd 7 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(bgshell) 18712 exited with return code 0
(serial) closing non working fd=7 ""
(wmbus) disconnected  rtlwmbus
(serialcmd) closed /bin/sh pid=18712 fd=7 (rtlwmbus)
Lost  closing rtlwmbus
(wmbus) closing....
(wmbus) deleted rtlwmbus
(serial) EVENT thread interrupted
(serialcmd) closed /bin/sh pid=18712 fd=-1 (rtlwmbus)
(main) rtlsdr device  not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 18740 fd 7 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(bgshell) 18740 exited with return code 0
(serial) closing non working fd=7 ""
(wmbus) disconnected  rtlwmbus
(serialcmd) closed /bin/sh pid=18740 fd=7 (rtlwmbus)
Lost  closing rtlwmbus
(wmbus) closing....
(wmbus) deleted rtlwmbus
(serialcmd) closed /bin/sh pid=18740 fd=-1 (rtlwmbus)
(serial) EVENT thread interrupted
(main) rtlsdr device  not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 18768 fd 7 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(bgshell) 18768 exited with return code 0
(serial) closing non working fd=7 ""
(wmbus) disconnected  rtlwmbus
(serialcmd) closed /bin/sh pid=18768 fd=7 (rtlwmbus)
Lost  closing rtlwmbus
(wmbus) closing....
(wmbus) deleted rtlwmbus
(serialcmd) closed /bin/sh pid=18768 fd=-1 (rtlwmbus)
(serial) EVENT thread interrupted
(main) rtlsdr device  not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d -1 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 18796 fd 7 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(bgshell) 18796 exited with return code 0
(serial) closing non working fd=7 ""
(wmbus) disconnected  rtlwmbus
(serialcmd) closed /bin/sh pid=18796 fd=7 (rtlwmbus)
`
My docker-compose version : 
`
version: "2"
services:
  wmbusmeters:
   image: weetmuts/wmbusmeters
   container_name: wmbusmeters
   restart: always
   network_mode: host
   volumes:
    - /volume1/docker/wmbusmeters/wmbusmeters_config:/opt/wmbusmeters
    - /volume1/docker/wmbusmeters/wmbusmeters_data:/wmbusmeters_data
    - /etc/localtime:/etc/localtime:ro
    - /dev/bus/usb/:/dev/bus/usb/
`
How can i solve this ? just for information i use the last version of images (a93de8561028)

BIBOLV, 2022-12-31:
Hi,
This is very similar to https://github.com/weetmuts/wmbusmeters/issues/657 with non-ASCII chars in serial. 
Do you have access to system/kernel logs? If you unplug and plug dongle back in does it show details about dongle, like serial? 
Example from linux:
```
 Jan 24 00:06:13 raspberrypi kernel: [ 2930.822220] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822229] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822234] usb 1-1.3: Product: RTL2838UHIDIR
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822238] usb 1-1.3: Manufacturer: Realtek
Jan 24 00:06:13 raspberrypi kernel: [ 2930.822242] usb 1-1.3: SerialNumber: 00000001
```

snapspizza, 2023-01-02:
What version of DSM are you using? 

Maybe this could be related to the fact that Synology removed support for USB devices in DSM 7, including support for DVB-T dongles. So there is no driver to handle the dongle when you plug it in. I haven't seen any third party kernel modules available, but maybe you can compile it yourself from sources?

BIBOLV, 2023-01-02:
Quick googling about DSM7 provided following [link](https://mariushosting.com/synology-how-to-add-usb-support-on-dsm-7/). I do not have Synology, so can`t comment how legit is it.

snapspizza, 2023-01-02:
> Quick googling about DSM7 provided following [link](https://mariushosting.com/synology-how-to-add-usb-support-on-dsm-7/). I do not have Synology, so can`t comment how legit is it.

I can confirm that this works with many zigbee dongles, but it doesn't include the dvb_usb_rtl28xxu.ko kernel module that I think is necessary for the RTL-SDR to work. It doesn't seem to make a difference on my Synology NAS anyways.

This [repository ](https://github.com/robertklep/dsm7-usb-serial-drivers)also provides the serial drivers, including a guide on how to set up a toolchain for building new modules. But I guess it can be quite a mouthful, depending on experience, to set it up.

softeg-klee, 2023-01-03:
> What version of DSM are you using?
>
> Maybe this could be related to the fact that Synology removed support for USB devices in DSM 7, including support for DVB-T dongles. So there is no driver to [...]

Thanks for reply, My version of DSM is : **DSM 7.1-42661**.

softeg-klee, 2023-01-03:
> 

Thanks for reply.

Yes i confirm i have anathor usb zigbee dongles conbee 2 and it works correctly, i used this repo to detect the dongle https://github.com/EddieDSuza/techwitheddie/blob/main/dsm7_usb_suppot/dsm7_enable_usb_support.sh

@snapspizza  I don't find the module dvb_usb_rtl28xxu.ko in the list of modules of this [repository ](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules)

How do you fix it ?

snapspizza, 2023-01-03:
> Thanks for reply, My version of DSM is : **DSM 7.1-42661**.

Okay. Then you will need additional drivers for the rtlsdr to run.

> @snapspizza I don't find the module dvb_usb_rtl28xxu.ko in the list of modules of this [repository ](https://github.com/robertklep/dsm7-usb-serial-drivers/tree/main/modules)

No, it only seems to have the serial drivers, which wouldn't be relevant here. I was referring to the section of the [README ](https://github.com/robertklep/dsm7-usb-serial-drivers#building-from-source)that explains how you could build kernel modules yourself. I have no experience with this, so can't really help with that. I moved everything to a Raspberry Pi, to reduce electricity usage, before I bought my rtlsdr stick. Alternatively, you could try to create an issue within his repo. It looks like he is willing to help people create the modules they need.

BIBOLV, 2023-01-03:
@snapspizza - tnx for sharing your experience and help. Closing issue for now.

closed 2023-01-03 by BIBOLV