wmbusmeters/wmbusmeters #775: Add support for Kamstrup 601 in kamheat driver

pilehave, 2022-12-30:
Hi!

I use the kamheat driver for my Kamstrup 601, which is similar to a 602. It reports back the correct data.

How can I help adding it as a supported device?

This is the output in the meter_readings log:

{
"media":"heat volume at inlet",
"meter":"kamheat",
"name":"Fjernvarme",
"id":"06375861",
"status":"OK",
"total_energy_consumption_kwh":200869,
"total_volume_m3":5940.59,
"volume_flow_m3h":0.003,
"t1_temperature_c":62.38,"t2_temperature_c":30.25,
"forward_energy_m3c":397043,
"return_energy_m3c":220360,
"timestamp":"2022-12-30T22:00:31Z",
"device":"rtlwmbus[00000001]",
"rssi_dbm":148
}

This is an example telegram:
telegram=|31442D2C615837061A0C8D2003600F192780B179C4D75157_9A100300E10E0600BF5C030072100900000019000000FE1B2609|+94581

weetmuts, 2022-12-30:
Thanks! Can you collect a long telegram as well? This is a compressed telegram which can only be understood if wmbusmeters has seen the previous long telegram.

pilehave, 2022-12-30:
Not sure what I'm looking for, but these are the lines before the compressed one:

(ELL) decrypted "5C030072100900000019000000FE1B26"
(ELL) block 2 block_size 1 offset 32
(ELL) decrypted "09"
(ELL) decrypted "80B179C4D751579A100300E10E0600BF5C030072100900000019000000FE1B2609"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=03 SN=600f1927 (AES_CTR session=0 time=7442678) CRC=80b1
(wmbus) parseNWL @19 31
(wmbus) parseAFL @19 31
(wmbus) parseTPL @19 31
(dvparser) found remembered format for hash d7c4
(dvparser) using format "040604FF0704FF08041402FD17043B0259025D"
(telegram) TPL CI=79

weetmuts, 2022-12-30:
Nice! It decodes everything except the status bits. 
https://wmbusmeters.org/analyze/40442D2C615837061A0C8D2004840F192719D87804069B10030004FF07E30E060004FF08C05C030004147410090002FD170000043B270000000259001C025DB70F

Do you have  any documentation on the 16 bit of error bits?

BIBOLV, 2022-12-30:
Could be those that are described [here](https://www.multical.hu/upload/files/MC601_UF14_cooling_install.pdf) in chapter 3. Also in wmbus modules [description](https://www.kamstrup.com/en-en/heat-solutions/meters-devices/discontinued-products/wirelessmbus-for-multical-61-601-801/documents) it is said that info codes are supported although in [mbus module](https://www.arigo-software.de/media/original/178/2282512519181644375/m-bus_Module_Multical602_DataSheet_ENG.pdf) there are more codes. Without proper documentation or test telegrams, we can`t be sure what`s there.
It`s a shame that just few of manufacturers actually provides full information about configuration of fields.

pilehave, 2022-12-31:
> Nice! It decodes everything except the status bits. https://wmbusmeters.org/analyze/40442D2C615837061A0C8D2004840F192719D87804069B10030004FF07E30E060004FF08C05C030004147410090002FD170000043B270000000259001C025DB70F
> 
> Do you have any documentation on the 16 bit of error bits?

Sorry, can't really help there :-/

closed 2023-04-02 by github-actions[bot]