wmbusmeters/wmbusmeters #776: Fail to decrypt with working and confirmed key, Multical21 on HA

kennethnilssen, 2022-12-31:
Hi, 

First of all, an amazing program, you cooked up here!

I have a Multical21, in Lillestrøm, Norway.

Running the solution on a HA, and fetched the signals via rtl433.

Got the KEM file to decrypt it.

Did run the python script, and got it decrypted. This is also confirmed via: 
https://wmbusmeters.org/analyze/23442D2C325717751B168D208924E433210FC4FFCCC8A63E0244D0F68DDC64129E89914A:multical21:4354CF9AB41B58D15F90CA5E1BB92E47

What is wrong on my side here? 

`Found meter MC21 (02140B0G844)
    number : 75175732
    serial : 75175732
    type   : VolumeCold
    driver : multical21
    config : 0100200023533
    key    : 4354CF9AB41B58D15F90CA5E1BB92E47
    meter file: home/pi/GIT/wmbusmeters/etc/wmbusmeters.d/75175732`

`Received telegram from: 75175732
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: rtl433[00000001]
                  rssi: 999 dBm
                driver: multical21
(wmbus) 000   : 23 length (35 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 2d2c dll-mfct (KAM)
(wmbus) 004   : 32571775 dll-id (75175732)
(wmbus) 008   : 1b dll-version
(wmbus) 009   : 16 dll-type (Cold water meter)
(wmbus) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : 96 ell-acc
(wmbus) 013   : 68e53321 sn (AES_CTR)
(wmbus) 017   : e635 payload crc (calculated ac84 ERROR)
(wmbus) 019 CE: 76F0826C3671EBD70A20F49A246868F90B failed decryption. Wrong key?
telegram=|_23442D2C325717751B168D209668E53321E63576F0826C3671EBD70A20F49A246868F90B|+104
(wmbus) telegram from 75175732 ignored by all configured meters!`

The config is:
`data_path: /etc/wmbusmeters
enable_mqtt_discovery: true
conf: |-
  loglevel=debug
  device= rtl433:t1,c1
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - name=MainWater driver=auto id=75175732 key=4354CF9AB41B58D15F90CA5E1BB92E47
mqtt:
  host: 10.0.0.90
  port: 1883
  user: Kenneth
  password: **********
`

_And, if a smarter person out there can point out why it does not post to MQTT, please help me. I have tested with both the {} and config...._

kennethnilssen, 2023-01-02:
I got a bit closer to understanding what the issue is:

From the addon log:
[23:50:53] INFO: Syncing wmbusmeters configuration ...
[23:50:53] INFO: Registering meters ...
[23:50:54] INFO: Adding meter-0001 ...
[23:50:54] INFO: Generating MQTT configuration ... 
[23:50:55] INFO: Broker 10.0.0.90 will be used.
MQTT Discovery ...
 Adding meter: MainWater driver ...
MQTT Discovery cleanup...
[23:50:58] INFO: Running wmbusmeters ...
(config) "device" "rtlwmbus:c1"
(wmbusmeters) version: 1.11.0-10-g08ca21e
(config) store meter files in: "/config/wmbusmeters/logs/meter_readings/wmbusmeters_readings.log"
(config) using device: rtlwmbus:c1 
(config) number of meters: 0

So, it does not pick up the actual meter... 

And then:

(config) loading meter file /config/wmbusmeters/etc/wmbusmeters.d/meter-0001
(config) name=MainWater driver=multical21 id=75175732 key=4354CF9AB41B58D15F90CA5E1BB92E47
Not a valid meter id nor a valid meter match expression ""

Not sure how to solve it tho....

snapspizza, 2023-01-04:
The complete lack of line shift make it very difficult to reading your logs. In the future, please format both logs and configs as code to preserve line shifts and indentations.

I think the problem is in your config in the Home Assistant add-on. For each meter, the driver, id, name, key etc. needs to be on separate lines. I have posted my working config as a reference below. Try editing your config in YAML-mode (change to this by using the hamburger menu in the top right corner of the options window of add-on configuration page) and use the same syntax and indentation as in my example.

```
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: |-
  loglevel=normal
  device=rtlwmbus
  listento=t1 
  donotprobe=/dev/ttyUSB0
  donotprobe=/dev/ttyACM0
  logtelegrams=false
  format=json 
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=electricity
    driver=abbb23
    id=00674565
  - |-
    name=heat
    driver=sharky774
    id=64313374
mqtt:
  host: core-mosquitto
  port: 1883
  user: ******
  password: *************

```

If this doesn't work, could you post your config from YAML-mode, formatted as code, or alternatively posted to pastebin.com or something similar.

kennethnilssen, 2023-01-04:
Hi @snapspizza 
Thank you for your comment, and yes, I will make a better post in the future, I did add the < > tag on the code, removed now so it looks better, but sure, pastebin is also good.

I can try with your example as code, to test. It feels like it takes everything to one line after saving, but I will give it a shoot.

kennethnilssen, 2023-01-04:
@snapspizza 

Your code example worked, thank you so much!

Can I kindly ask you as well to share some good looking lovelace cards as well if you have them? 

Now it does report into HA, and I would nominate your example to go into the HA template, hint to @weetmuts

snapspizza, 2023-01-04:
@kennethnilssen That's great! I'm glad I could help.

I don't really have much to show in relation to lovelace cards based on my meter data. My primary motivation for implementing wmbusmeters in HA was to use my meter data in the energy dashboard. Besides that I have a few [mini-graph-card](https://github.com/kalkih/mini-graph-card)s showing current and past heat and electricity usage.

kennethnilssen, 2023-01-04:
> 

Thank you, I found these cards that I will try on for the water:
https://github.com/zibous/ha-watermeter/tree/master/docs/homeassistant

closed 2023-01-04 by kennethnilssen