wmbusmeters/wmbusmeters #779: (apator162) telegram contains a register (97) with unknown size.

philips1923, 2023-01-02:
Hi, Im reporting an issue: 
`3E4401069250960205077AF10030852F2F0F28422197980000439B0387948CB65590D605E409001070DE050071019ADC0500F1B80500A0126DEF03FFFF3EF0`

`3E4401069250960205077AF20030852F2F0F29422197980000439B0383000897745A8A01CC8F0E1070DE050071019ADC0500F1B80500A0126DEF03FFFF44A2`
`
3E4401069250960205077AF30030852F2F0F2A422197980000439B0384B23E8A447FCC71BA04001070DE050071019ADC0500F1B80500A0126DEF03FFFF7CF5`

etc..very similar issue as on 1st January 2022 https://github.com/weetmuts/wmbusmeters/issues/432

All the best in 2023 :)

weetmuts, 2023-01-02:
Have you tried the latest version of wmbusmeters? I think it handles this case.
 
https://wmbusmeters.org/analyze/3E4401069250960205077AF10030852F2F0F28422197980000439B0387948CB65590D605E409001070DE050071019ADC0500F1B80500A0126DEF03FFFF3EF0

kniazio, 2023-01-03:
Please write step by step how to update the version wmbusmeters

kniazio, 2023-01-03:
Sorry to bother you but I really don't know how to update wmbusmeters

weetmuts, 2023-01-03:
you have to explain what computer you are using. what software and how you installed wmbusmers in the first place.

philips1923, 2023-01-03:
uh, it was difficult for me. My configuration is HassIO on HP T630. I stopped wmbusmeters in addons section, made backup of config folder (usually HA-machine/config/wmbusmeters/etc/), removed old version of wmbusmeters repository (I saw that URL of repository has changed), removed config folder of wmbusmeters in HA, given new repository in addons (URL is https://github.com/weetmuts/wmbusmeters ), installed wmbusmeters, started and gave old configuration of my water meter (taken from backup before removing this all stuff). Now it works

KrzysztofHajdamowicz, 2023-01-05:
> uh, it was difficult for me. My configuration is HassIO on HP T630. I stopped wmbusmeters in addons section, made backup of config folder (usually HA-machine/config/wmbusmeters/etc/), removed old version of wmbusmeters repository (I saw that URL of repository has changed), removed config folder of wmbusmeters in HA, gave new repository in addons (URL is https://github.com/weetmuts/wmbusmeters ), installed wmbusmeters, started and gave old configuration of my water meter (taken from backup before removing this all stuff). Now it works

That's because you upgraded wmbusmeters that way :)

closed 2023-04-18 by github-actions[bot]