wmbusmeters/wmbusmeters #785: (wmbus) telegram ignored by all configured meters!

olivejer, 2023-01-03:
Hi
first of all congrats it nice project.
I have got an issue with Izar sensor : IZAR RC 868 | R4

Telegram is ignored despite a working analyse.

(serial) received ascii "T1;1;1;2023-01-03 20:25:43.000;103;147;211E42AF;0x1944304caf421e219801a10116001327bfff632728feb30e3afc<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-01-03 20:25:43.000;103;147;211E42AF;0x1944304caf421e219801a10116001327bfff632728feb30e3afc<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(diehl) preprocess necessary SAP_PRIOS
(diehl) Pre-processing: setting device type to water meter for SAP PRIOS
(wmbus) parseDLL @0 26
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(diehl) preprocess necessary SAP_PRIOS
(diehl) Pre-processing: setting device type to water meter for SAP PRIOS
(wmbus) parseDLL @0 26
(telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=211e42af VER=00 TYPE=07 (Water meter) (driver izar) DEV=rtlwmbus[00000001] RSSI=103
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) TPL CI=a1
Received telegram from: 211e42af
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: rtlwmbus[00000001]
                  rssi: 103 dBm
                driver: izar
(wmbus) 000   : 19 length (25 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 304c dll-mfct (SAP)
(wmbus) 004   : af421e21 dll-id (211e42af)
(wmbus) 008   : 00 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : a1 tpl-ci-field (Mfct specific)
(wmbus) 011 C?: 0116001327BFFF632728FEB30E3AFC mfct specific
(wmbus) telegram from 211e42af ignored by all configured meters!

https://wmbusmeters.org/analyze/1944304caf421e219801a10116001327bfff632728feb30e3afc:izar

fariouche, 2023-02-09:
I have the same problem, any news?

fariouche, 2023-02-11:
seems like if you have no config meters and no meter in the command line, it will print this message for every telegram.
If I understand correctly this trace, it just means that the telegram is not for any of the configured meters.
By the way, while looking at the code, I've seen a strange line:
in src/metermanager.cc line 144, I see this:
```
for (auto &m : meters_)
{
    bool h = m->handleTelegram(about, input_frame, simulated, &ids, &exact_id_match);
    if (h) handled = true;
}
```

Is it normal that there is no break if the telegram was handled?

weetmuts, 2023-02-11:
Yes. Multiple meters can match the telegram if you want to decode using multiple drivers.

closed 2023-05-14 by github-actions[bot]