wmbusmeters/wmbusmeters #786: Wildcard rule and auto driver are incompatible now

bb-froggy, 2023-01-04:
Since [9ce4328d](https://github.com/weetmuts/wmbusmeters/commit/9ce4328d2413a7fea825c238af39ac8f3f100211#diff-7ddfe606edc35b4bfa2f81d73cbffc72979dfc82913b2b600e3e13a1f9bd78c5L1012-R959), you cannot use the auto driver if you have a wildcard ID rule. Is that on purpose? When searching for new devices while still logging the specific ones I configured, I used to have such an auto/wildcard meter configuration.

pswiatki, 2023-01-04:
Yes, exactly. I have:

```
(diehl) Pre-processing: setting device type to water meter for SAP PRIOS
(wmbus) parseDLL @0 26
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(meter) sasiedzi/others: for me? 60e4ff30 in *,!4122bcb2,!00109314
(meter) ignoring telegram from 60e4ff30 since it matched a wildcard id rule but driver (auto) does not match.
(meter) prund/P15_Main: for me? 60e4ff30 in 00109314
(meter) prund/P15_Main: not for me: not my id
(meter) wasser/P15_Main: for me? 60e4ff30 in 4122bcb2
(meter) wasser/P15_Main: not for me: not my id
(wmbus) telegram from 60e4ff30 ignored by all configured meters!
```
Not sure what it means `[...]since it matched a wildcard id rule but driver (auto) does not match`.

weetmuts, 2023-01-04:
Ah, yes, actually it seems like actively disabled this. (But I can revert this change easily if you have a good use case.)

My thinking was that if you use id wildcards you do not want to get unexpected meters triggering,
because that would upset your reporting/analysis. Imagine thinking that you were receiving many water meters, and suddenly there is a new heat meter installed in the vicinity (which also provides a total_m3) and you get weird data.

So, since I also nowadays also say that you should only use auto for detecting, but not for prod. Because if you want stability in the driver you should use the driver name. A new version of wmbusmeters might switch to a newer driver (picked by auto) that has changes in the field names. This will part of a longer plan normalizing all field names between drivers.

Thus, it made sence not to allow auto for wildcard rules....I think. :-)

Are you actively listening to multiple meters, matched using wildcards that have different drivers?

bb-froggy, 2023-01-06:
My configuration is like this:
- meter definition 1: Some specific IDs and a specific driver
- meter definition 2: Some specific IDs and a specific driver
- ...
- meter definition "Clutter": id=* and driver=auto

"Clutter" is usually disabled, but when I want to add a new set of devices, I enable it and then create a specific meter definition based on what it finds, and then disable again. For example, I currently want to add some smoke detectors, but I haven't really figured out all of their IDs and messages types. I could of course achieve the same by deleting all meter definitions, which would print more info in the logs, but I don't want to miss out on the specific meter definitions that I have.

weetmuts, 2023-01-07:
Ok, so you use auto to find new meters, do you need it run at the same time as the existing meters are still working?

Do you capture the auto found meters using a log file or do you use shell and for example mqtt_publish to collect the auto data?

weetmuts, 2023-01-10:
Ok, I agree with you @bb-froggy Using the clutter rule is clever. I have restored the functionality.
Please test!

BIBOLV, 2023-01-18:
@bb-froggy - everything works now? This issue can be closed?

bb-froggy, 2023-01-19:
Yes, this works again. Sorry for the long delays.

closed 2023-01-19 by bb-froggy