wmbusmeters/wmbusmeters #787: Unit conversion is not working - addconversions=XXX

DomAtHome, 2023-01-04:
I've updated the wmbusmeters to the newest version yesterday to check whether the sharky 774 will work correctly after all the changes. It seems that the value which is sent in GJ is recognized that it is in GJ and correctly converted to kwh in the reading file. What stopped working is the automatic conversion, which should extend the reading file with another field. I've checked my `wmbusmeter.conf` file and the `addconversions=GJ` is still there but the property `total_energy_consumption_gj` is missing in the reading file. Can it be connected with the refactoring of the driver or there is something wrong with my setup?

weetmuts, 2023-01-04:
As is documented in the CHANGES file the addconversions has been removed. It was too adhoc solution. 

Now you do:
--calculate_total_energy_consumption_gj=total_energy_consumption_kwh 

or any name and any kind of formula.
--calculate_foo_gj=total_energy_consumption_kwh+1111kwh

DomAtHome, 2023-01-04:
Ok, I see. Should I put it in main config file or in meter config file?

weetmuts, 2023-01-04:
meter file

BIBOLV, 2023-01-12:
@DomAtHome, did you manage to achieve what`s needed?

DomAtHome, 2023-01-12:
Sorry I did not check it. The kWh value is saved in the file so I converted the values later on in my process. I am sure that what @weetmuts wrote is correct ;) The issue can be closed.

closed 2023-01-12 by DomAtHome