wmbusmeters/wmbusmeters #789: New version of Itron Cyble with two headers (radio converter and radio meter)

vohum, 2023-01-05:
```
serial) received binary "3C4495"
(amb8465) checkAMB8465Frame "3C4495"
(amb8465) not enough bytes yet, partial frame 3 61.
(serial) received binary "264147140042378C205E900F002C2571"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "3C4495264147140042378C205E900F002C2571"
(amb8465) not enough bytes yet, partial frame 19 61.
(serial) received binary "09000073A5E028FE3CC21772414714"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "3C4495264147140042378C205E900F002C257109000073A5E028FE3CC21772414714"
(amb8465) not enough bytes yet, partial frame 34 61.
(serial) received binary "00952642075E081007102F8507D4A2"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "3C4495264147140042378C205E900F002C257109000073A5E028FE3CC2177241471400952642075E081007102F8507D4A2"
(amb8465) not enough bytes yet, partial frame 49 61.
(serial) received binary "DA27E65D22FD7A8F797C3112"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "3C4495264147140042378C205E900F002C257109000073A5E028FE3CC2177241471400952642075E081007102F8507D4A2DA27E65D22FD7A8F797C3112"
(amb8465) received full frame
(amb8465) rssi 18 (-65 dBm)
(wmbus) parseDLL @0 61
(wmbus) parseELL @10 51
(wmbus) parseNWL @13 48
(wmbus) parseAFL @13 48
(wmbus) parseTPL @30 31
(wmbus) input to kdf for enc "00710900004147140007070707070707"
(wmbus) no key, thus cannot execute kdf.
(meter) MyTapWater: for me? 00144741,00144741 in 00144741
(meter) MyTapWater: meter detection did not match the selected driver auto! correct driver is: unknown!
(meter) Not printing this warning again for id: 00144741 mfct: (ITU) ITRON United States Itron, United States of America (0x2695) type: Radio converter (meter side) (0x37) ver: 0x42
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) MyTapWater: yes for me
(meter) MyTapWater(1) auto  handling telegram from 00144741
(meter) MyTapWater 00144741 "3C4495264147140042378C205E900F002C257109000073A5E028FE3CC2177241471400952642075E081007102F8507D4A2DA27E65D22FD7A8F797C3112"
(wmbus) parseDLL @0 61
(telegram) DLL L=3c C=44 (from meter SND_NR) M=2695 (ITU) A=00144741 VER=42 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV=amb8465[00000000] RSSI=-65
(wmbus) parseELL @10 51
(telegram) ELL CI=8c CC=20 (slow_resp sync) ACC=5e
(wmbus) parseNWL @13 48
(wmbus) parseAFL @13 48
(telegram) AFL CI=90
(wmbus) parseTPL @30 31
(wmbus) input to kdf for enc "00710900004147140007070707070707"
(wmbus) ephemereal Kenc 0EB57DD5E9E68B96358BA55742B12FDB
(wmbus) input to kdf for mac "01710900004147140007070707070707"
(wmbus) ephemereal Kmac 269F9F48A18BE31D72876D88BBA1E3D0
(wmbus) input to mac 25710900007241471400952642075E081007102F8507D4A2DA27E65D22FD7A8F797C3112
(wmbus) calculated mac 87159CD798E8EFE180B83E6F013A0767
(wmbus) received   mac 73A5E028FE3CC217
(wmbus) mac NOT ok!
BADMAC 000   : 3c length (60 bytes)
BADMAC 001   : 44 dll-c (from meter SND_NR)
BADMAC 002   : 9526 dll-mfct (ITU)
BADMAC 004   : 41471400 dll-id (00144741)
BADMAC 008   : 42 dll-version
BADMAC 009   : 37 dll-type (Radio converter (meter side))
BADMAC 010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
BADMAC 011   : 20 ell-cc (slow_resp sync)
BADMAC 012   : 5e ell-acc
BADMAC 013   : 90 afl-ci-field (AFL: Authentication and Fragmentation Sublayer)
BADMAC 014   : 0f afl-len (15)
BADMAC 015   : 002c afl-fc (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)
BADMAC 017   : 25 afl-mcl (AES_CMAC_128_8 MessCounter)
BADMAC 018   : 71090000 afl-counter (2417)
BADMAC 022   : 73A5E028FE3CC217 afl-mac 8 bytes
BADMAC 030   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
BADMAC 031   : 41471400 tpl-id (00144741)
BADMAC 035   : 9526 tpl-mfct (ITU)
BADMAC 037   : 42 tpl-version
BADMAC 038   : 07 tpl-type (Water meter)
BADMAC 039   : 5e tpl-acc-field
BADMAC 040   : 08 tpl-sts-field (PERMANENT_ERROR)
BADMAC 041   : 1007 tpl-cfg 0710 (AES_CBC_NO_IV )
BADMAC 043   : 10 tpl-cfg-ext (KDFS=1)
(wmbus) WARNING! telegram mac check failed, did you use the correct decryption key? Permanently ignoring telegrams from id: 00144741 mfct: (ITU) ITRON United States Itron, United States of America (0x2695) type: Radio converter (meter side) (0x37) ver: 0x42
(telegram) TPL CI=72 ACC=5e STS=08 CFG=0710 (AES_CBC_NO_IV) ID=41471400 MFT=9526 VER=42 TYPE=07 (Water meter)
(wmbus) telegram from 00144741,00144741 ignored by all configured meters!
```

can't decode it, probably because there are two headers 
meter number: 00144741
key: 39BED49427DEEC0ABE44D57570CB1EBB

weetmuts, 2023-04-18:
Thank you! Do you know for sure this telegram is supposed to be decoded using this key?

weetmuts, 2023-04-18:
When I am looking at this telegram the MAC is calculated over the tpl data, ie start from the byte 0x72 at offset 30.
This should be correct, the ell-header should not interfere with this... I think. :-)

closed 2023-07-18 by github-actions[bot]