wmbusmeters/wmbusmeters #790: Ernst Heitland Watermeter

electron23, 2023-01-06:
Hi,

I use an Ernst Heitland M100i watermeter with an Falcon MJ Puls/M-Bus Modul via mbus and an mbus master. 

I configured wmbusmeters properly to get an output. However the data is not send via mqtt to home assistant. The output, if I start it manually without homeassistant, looks like:

`#./wmbusmeters --pollinterval=60s MAI/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_AKCMn114J20-if00-port0t0:mbus:2400 MainWaterMeter emerlin868:MAIN:mbus XXXXXXX NOKEY
Started config MAIN=mbus on /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_AKCMn114J20-if00-port0
(meter) MainWaterMeter: meter detection did not match the selected driver emerlin868! correct driver is: unknown!
(meter) Not printing this warning again for id: 00000000 mfct: (@@@) Unknown (0x00) type: Other (0x00) ver: 0x00
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
MainWaterMeter  XXXXXXX  0.24 m³ 0 m³    2023-01-06 15:35.19`

The output from home asstistant addon log: 

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:09:15] INFO: Syncing wmbusmeters configuration ...
[16:09:15] INFO: Registering meters ...
[16:09:15] INFO: Adding meter-0001 ...
[16:09:15] INFO: Generating MQTT configuration ... 
[16:09:15] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[16:09:16] INFO: Running wmbusmeters ...
Started config MAIN=mbus on /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_AKCMn114J20-if00-port0
(meter) MainWaterMeter: meter detection did not match the selected driver emerlin868! correct driver is: unknown!
(meter) Not printing this warning again for id: 00000000 mfct: (@@@) Unknown (0x00) type: Other (0x00) ver: 0x00
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination`

`root@pve:~# mbus-serial-request-data-multi-reply -d -b 2400 /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_AKCMn114J20-if00-port0 254
main: debug: sending init frame
[2023-01-06 14:28:12] SEND (005): 10 40 FF 3F 16
[2023-01-06 14:28:12] SEND (005): 10 7B FE 79 16
[2023-01-06 14:28:13] RECV (087): 68 51 51 68 08 00 72 29 05 03 42 92 15 B2 07 01 00 00 00 0C 13 40 02 00 00 04 6D 1C 10 E6 21 42 6C E1 21 4C 13 00 00 00 00 42 EC 7E 01 31 0C 93 3C 01 00 00 00 12 6C E6 21 14 3B 48 03 00 00 04 3B 00 00 00 00 C4 03 6D 00 00 00 00 84 03 6D 00 00 00 00 0F 00 B1 16
mbus_frame_print: Dumping M-Bus frame [type 4, 87 bytes]: 68 51 51 68 08 00 72 29 05 03 42 92 15 B2 07 01 00 00 00 0C 13 40 02 00 00 04 6D 1C 10 E6 21 42 6C E1 21 4C 13 00 00 00 00 42 EC 7E 01 31 0C 93 3C 01 00 00 00 12 6C E6 21 14 3B 48 03 00 00 04 3B 00 00 00 00 C4 03 6D 00 00 00 00 84 03 6D 00 00 00 00 0F 00 B1 16
<MBusData>

    <SlaveInformation>
        <Id>XXXXXXX</Id>
        <Manufacturer>ELR</Manufacturer>
        <Version>178</Version>
        <ProductName></ProductName>
        <Medium>Water</Medium>
        <AccessNumber>1</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <Unit>Volume (m m^3)</Unit>
        <Value>240</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2023-01-06T16:28:00</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <Unit>Time Point (date)</Unit>
        <Value>2023-01-01</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <Unit>Volume (m m^3)</Unit>
        <Value>0</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <Unit>Time Point (date)</Unit>
        <Value>2024-01-01</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <Unit>Volume (m m^3)</Unit>
        <Value>1</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Maximum value</Function>
        <Unit>Time Point (date)</Unit>
        <Value>2023-01-06</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="7">
        <Function>Maximum value</Function>
        <Unit>Volume flow (m m^3/h)</Unit>
        <Value>840</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="8">
        <Function>Instantaneous value</Function>
        <Unit>Volume flow (m m^3/h)</Unit>
        <Value>0</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="9">
        <Function>Instantaneous value</Function>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2000-00-00T00:00:00</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="10">
        <Function>Instantaneous value</Function>
        <Unit>Time Point (time &amp; date)</Unit>
        <Value>2000-00-00T00:00:00</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

    <DataRecord id="11">
        <Function>Manufacturer specific</Function>
        <Value>00</Value>
        <Timestamp>2023-01-06T14:28:13</Timestamp>
    </DataRecord>

</MBusData>`

What should I do, to get it to work? 

Thanks.

electron23, 2023-01-07:
Hi, 

sorry for the inconvenience, it's already working. I hadn't set up the MQTT sensor.

Best regards,
Nico

closed 2023-01-07 by BIBOLV

mklacan, 2024-01-12:
Hi @electron23,
Can you help me? I have very similar setup with Home assistant.  
Look at here: https://github.com/wmbusmeters/wmbusmeters/issues/1137 Honeywell mbus clip merlin 868 and m100i.

Thx.

mklacan, 2024-01-12:
Thx, for answer.
Can you send me info about m-bus in node red? 
I have node-red addon in Home Assistant  run on Synology. 
Or Node Red in docek on Synology.

I will try setup with Node red.