wmbusmeters/wmbusmeters #792: ISQL help to configure shell command in wmbusmeters.conf

hngfngzl, 2023-01-08:
Hello Mr Weemuts,

I got working wmbusmeters and I try to write the values per shell command into an MS Sql database over the network. 

The command on the command line that is testet and works is:
echo „insert into meter values ('6', 'Test6');“ | isql -c wfiot user password

So my next step was to try this on the command line with wmbusmeters direct but didn't work because I don't get the correct syntax.

Same with my line in wmbusmeters.conf. At the moment it is:
wmbusmeters --shell= echo "insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP')" /dev/ttyUSB0:im871a:c1 | isql wfiot user password

Yet I don't understand what /dev/ttyUSB0:im871a:c1 does or how the correct syntax is for it. 

I'd be happy for hints. We can also do the Text in German - easier for my - perhaps also for you.

Best Regards
Daniel Wolf

BIBOLV, 2023-01-09:
Hi, try something like this:
`wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP')\" | isql wfiot user password" /dev/ttyUSB0:im871a:c1` 
And escape any special characters in username or password. Basically you need provide cmd that would be executed in shell parameter and then point to device and all meter related parameters. Please check readme for similar example.

hngfngzl, 2023-01-09:
Hi,

I tested but no success yet. I don’t understand what the “ do and what the \ does.
I guess I am very near but so far I can tell the variables \$Meter_ID and so on are not filled when the isql command is handed over to the shell to execute.
Is there a way to just print out one of the variables in from the command in the shell?

BIBOLV, 2023-01-09:
`--shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP')\" >> /tmp/wmbus_tmp_file"` should print to file insert statement

hngfngzl, 2023-01-09:
Ok, progress. It generates no file an with printenv he is not showing any of the env variables that should be there – so at last this is a explanation why in the json/logfile some is written and the rest doesn’t make anything….

I installed wmbusmeters with snap and run it for the testing withsudo snap run wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP');" \" /dev/rtlwmbus >> /snap/wmbus_tmp

I guess there comes the problem from so I’ll see if I find the env variables somewhere.

But for today I go to bed (quit late in Germany).

BIBOLV, 2023-01-10:
Couple of things, firstly you will not be able to run snap as isql command is not included in there, so if you need it to insert values in db, you will need to run standalone version. Second, in order to print some values you need to specify meter details, like `wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP');\" >> /tmp/wmbus_tmp_file" /dev/rtlwmbus  MyColdWater multical21 12345678 NOKEY`. And thirdly, if you are not sure what variables are available for you meter run `wmbusmeters --listenvs=multical21` where multical21 is your meter driver in both cases.
Please read readme carefully once more.

hngfngzl, 2023-01-12:
Ok I installed it without snap (which is really comfortable comparing without...)

Installed are librtlsdr-dev and libncurses-dev. CMake and https://gitea.osmocom.org/sdr/rtl-sdr.git and https://github.com/xaelsouth/rtl-wmbus

lsusb finds Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T (what should be rtlwmbus). 

when I run `wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP');\" >> /tmp/wmbus_tmp_file" /dev/rtlwmbus:c1,t1,s1`
I get:
[ALARM SpecifiedDeviceNotFound] the device rtlwmbus:t1 is not working
like in issue 242
When I use your example wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP');\" >> /tmp/wmbus_tmp_file" /dev/rtlwmbus than same error. if i use an invalid device he says in addion "Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path.

If i run rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus
bash: rtl_wmbus: command not found
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 1600000 S/s.
Tuned to 868900000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
Reattached kernel driver

An hint would be really helpful.

BIBOLV, 2023-01-12:
I suggest to spit this in 3 steps:

-  Get wmbusmeters up and running - ok, did you run make install for rtl_wmbus or just make? If not, you can either run make install or copy binary from build/rtl_wmbus to system. With that you should try again if everything works. 
- Get reading from meter you are interested in - to continue you need set up your meter and get reading from it. You will need to specify meter name, meter id, meter driver and decryption key
- One you have it, you can continue to implement meter value submission to db, by putting all steps together and adding shell command.

Also I would suggest to create all of above in config files and run as service, after testing from command line and seeing that everything works.
There is example of end result running in simulation mode:
```bash
~# ./wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP');\" >> /tmp/wmbus_tmp_file" 1844ae4c4455223368077a55000000041389e20100023B0000 MyCounter iperl 33225544 NOKEY
~# cat /tmp/wmbus_tmp_file
insert into readings values ('33225544','123.529','2023-01-12T21:22:18Z');
```

hngfngzl, 2023-01-13:
the "make install" was really helpfull - sorry I am not very in to linux so the basics can bring problems...

so the service is not running yet. but for testing wmbusmeters `rtlwmbus:s1` brings nice telegrams data.

The next step was to run `wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP')" \ >> /tmp/wmbus_tmp_file" " rtlwmbus:c1,s1,t1 `
brings "At least one valid device (or hex) must be supplied!" that means I have to put in meters in the heat file? For me it would be fine if he simply writes all in the file (or later fields in the database and i'll sort it out there.)

BIBOLV, 2023-01-13:
Ok, now you are in step two - you need to get reading from one of your meters. For that, as I already wrote, you need to specify meter name, meter id, meter driver and decryption key. Set parameters for one to test in general that everything works, and later on, when you will set up everything as service, you will be able to create more meters in config directory.

hngfngzl, 2023-01-13:
Ok, step two: 
`wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP');\" >> /tmp/wmbus_tmp_file" rtlwmbus:s1 MyCounter qcaloric 29319306 NOKEY `
brings 
insert into readings values ('29319306','','2023-01-13T19:57:03Z');
in the text file - what is not perfect - but good enough to proceed. so this text would be what should be executed by isql so the next step is to change to command:
wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TOTAL_M3','\$METER_TIMESTAMP'); \" | isql wfiot user pass" rtlwmbus:s1 MyCounter qcaloric 29319306 NOKEY 
I'll test this tomorrow - looks like the meter doesn't send in night times...

BIBOLV, 2023-01-13:
No, not really, you are using heat cost allocator driver and trying to extract values that represents water meter.
Please run `wmbusmeters --logtelegrams rtlwmbus:s1` and post here some telegrams of meter that you want to read.

hngfngzl, 2023-01-14:
Like you said two telegrams:

Received telegram from: 29319301
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: rtlwmbus[00000001]
                  rssi: 13 dBm
                driver: qcaloric
telegram=|_49449344019331293508780DFF5F350082120000610107B06EFFFF05010000DF2C50180000DF2C50180000AB002701F200840021000300000000002C006B000E0169012F046D210AEE21|+28
Received telegram from: 29318908
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: rtlwmbus[00000001]
                  rssi: 11 dBm
                driver: qcaloric
telegram=|_49449344088931293508780DFF5F350082CB0000810007B06EFFFF00000000DF2C03000000DF2C030000000200000000000000000000000000000000000000000000002F046D1F0AEE21|+29

I altered the Database a bit just to see if it generally works: `wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TIMESTAMP','c'); \" | isql wfiot user pass" rtlwmbus:s1 MyCounter qcaloric 29319306 NOKEY` 
Writes to the database - not usefull yet but a proof of concept. My next steps would be:
- useful fields from the meters to the database
- getting the service of wmbusmeters back to work with it

weetmuts, 2023-01-14:
Run with --logtelegrams then you will have the hex for re-use. Check that you are receiving the correct meter, you gave qcaloric examples above, those are heat cost allocators and not water meters. I.e. there is no METER_TOTAL_M3.

When you have captured your water meter the use the decrypted output in telegram=|....|
then you can re-use the telegram like this:
```
./build/wmbusmeters --shell="echo \"insert into readings values ('\$METER_ID','\$METER_TIMESTAMP','\$METER_TOTAL_M3');\" " 2A442D2C998734761B168D2091D37CAC21576C78_02FF207100041308190000441308190000615B7F616713 TestCounter multical21 76348799 NOKEY
```

You can also add --debug to see which METER files are populated in the env.

weetmuts, 2023-01-14:
I.e. you can use hex on the commandline to debug the transfer to your database. :-) No need to wait a day for new data.

hngfngzl, 2023-01-21:
Ok - now it runs fine on comman line. Time to get it working with the deamon. 
Same idea start with really less and then put step by step to it.
My wmbusmeters.conf only contains:

**loglevel=normal
device=rtlwmbus
listento=s1**

heat contains
**name=test
id=2*
key=NOKEY
driver=qcaloric**

I run `systemctl start wmbusmeters` and look at the status afterwards that brings: (code=exited, status=1/FAILURE)

weetmuts, 2023-01-21:
sudo tail -f /var/log/syslog
and/or
systemctl status wmbusmeters.service

The id must be 8 digits.

hngfngzl, 2023-01-21:
I changed heat Id=2* but following the log this is not the problem. The log brings: 

Jan 21 14:01:23 wfrp03 wmbusmetersd[701134]: (rtlwmbus) error: when starting as daemon, wmbusmeters looked for /rtl_sdr and /usr/bin/rtl_sdr, but found neither!Jan 21 14:01:23 wfrp03 systemd[1]: wmbusmeters.service: Main process exited, code=exited, status=1/FAILURE

hngfngzl, 2023-01-22:
Ok, i copied the file to the location and the daemon runs. 
The rest of the .conf file works so far too. 

But my easy think to just copy the wmbusmeters --shell="echo \"insert into... without the wmbusmeters -- into the .conf file didn't work out. What further changes are necessary?

BIBOLV, 2023-01-22:
Post your config here so we can take a look.

hngfngzl, 2023-01-22:
loglevel=normal
device=rtlwmbus
listento=s1
logtelegrams=true
format=json
meterfiles=/var/wfiot
meterfilesaction=append
logfile=/var/wfiot/wmbusmeter
shell="echo \"insert into wfiot_meter_qcaloric_live values ('\$METER_ID','\$METER_TIMESTAMP','\$METER_STATUS','\$METER_CURRENT_CONSUMPTION_HCA','\$METER_SET_DATE','\$METER_CONSUMPTION_AT_SET_DATE_HCA','\$METER_SET_DATE_1','\$METER_CONSUMPTION_AT_SET_DATE_1_HCA','\$METER_SET_DATE_17','\$METER_CONSUMPTION_AT_SET_DATE_17_HCA','\$METER_ERROR_DATE'); \" | isql wfiot user pass" rtlwmbus:s1 MyCounter qcaloric 2* NOKEY

BIBOLV, 2023-01-23:
Try `shell="echo \"insert into wfiot_meter_qcaloric_live values ('\$METER_ID','\$METER_TIMESTAMP','\$METER_STATUS','\$METER_CURRENT_CONSUMPTION_HCA','\$METER_SET_DATE','\$METER_CONSUMPTION_AT_SET_DATE_HCA','\$METER_SET_DATE_1','\$METER_CONSUMPTION_AT_SET_DATE_1_HCA','\$METER_SET_DATE_17','\$METER_CONSUMPTION_AT_SET_DATE_17_HCA','\$METER_ERROR_DATE'); \" | isql wfiot user pass"`

hngfngzl, 2023-01-23:
I did that already while testing, but same error code with 127
Value 127 is returned by /bin/sh when the given command is not found within your PATH system variable and it is not a built-in shell command. looks more like a problem in the shell itself.

BIBOLV, 2023-01-23:
Try then 
```
shell="echo \"insert into wfiot_meter_qcaloric_live values ('\$METER_ID','\$METER_TIMESTAMP','\$METER_STATUS','\$METER_CURRENT_CONSUMPTION_HCA','\$METER_SET_DATE','\$METER_CONSUMPTION_AT_SET_DATE_HCA','\$METER_SET_DATE_1','\$METER_CONSUMPTION_AT_SET_DATE_1_HCA','\$METER_SET_DATE_17','\$METER_CONSUMPTION_AT_SET_DATE_17_HCA','\$METER_ERROR_DATE'); \" | /usr/bin/isql wfiot user pass"
```
And if it does not work - set `loglevel=debug` in config, restart and attach log here.

hngfngzl, 2023-01-23:
ok, i shortend the log a bit, because he receives tons. this Telegram looked fine with data. 

(wmbusmeters) logging started 2023-01-23 09:53:00 using 1.11.0-25-gf6345e4
(wmbusmeters) version: 1.11.0-25-gf6345e4
(config) store meter files in: "/var/wfiot"
(config) using device: rtlwmbus 
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 70282 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 70284 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
[2023-01-23_09:53:00] Started config rtlwmbus[00000001] listening on s1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 70291 fd 6 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams

(telegram) TPL CI=7a ACC=4c STS=00 CFG=2000 (synchronous)
telegram=|314493448492312935087A4C000020_0B6E4500004B6E230400426CDF2CCB086E230400C2086CDF2C326CFFFF046D3809F721|+42
(meters) trying field info status(Text)[0]...
(meters) trying field info current_consumption(HCA)[1]...
(meters) using field info current_consumption(HCA)[1] to extract 0B6E at offset 17
(meter) HeatCostAllocation current_consumption_hca decoded hca default hca value 45
(meters) trying field info set_date(Text)[2]...
(meters) using field info set_date(Text)[2] to extract 426C at offset 27
(meters) trying field info consumption_at_set_date(HCA)[3]...
(meters) using field info consumption_at_set_date(HCA)[3] to extract 4B6E at offset 22
(meter) HeatCostAllocation consumption_at_set_date_hca decoded hca default hca value 423
(meters) trying field info set_date_1(Text)[4]...
(meters) using field info set_date_1(Text)[4] to extract 426C at offset 27
(meters) trying field info consumption_at_set_date_1(HCA)[5]...
(meters) using field info consumption_at_set_date_1(HCA)[5] to extract 4B6E at offset 22
(meter) HeatCostAllocation consumption_at_set_date_1_hca decoded hca default hca value 423
(meters) trying field info set_date_8(Text)[6]...
(meters) trying field info consumption_at_set_date_8(HCA)[7]...
(meters) trying field info set_date_17(Text)[8]...
[... 82 lines cut ...]
(shell) env "METER_CONSUMPTION_AT_SET_DATE_1_HCA=423"
(shell) env "METER_SET_DATE_8=null"
(shell) env "METER_CONSUMPTION_AT_SET_DATE_8_HCA=null"
(shell) env "METER_SET_DATE_17=2022-12-31"
(shell) env "METER_CONSUMPTION_AT_SET_DATE_17_HCA=423"
(shell) env "METER_ERROR_DATE=2127-15-31"
(shell) env "METER_DEVICE_DATE_TIME=2023-01-23 09:56"
(shell) env "METER_MODEL_VERSION=null"
(shell) env "METER_FLOW_TEMPERATURE_C=null"
(shell) waiting for child 70404 to complete.
(shell) /bin/sh: return code 127
[2023-01-23_09:53:42] (shell) /bin/sh exited with non-zero return code: 127
(wmbus) parseDLL @0 50
(wmbus) parseELL @10 40
(wmbus) parseNWL @10 40
(wmbus) parseAFL @10 40
(wmbus) parseTPL @10 40
(meter) dixi: for me? 29319284 in 29319027
(meter) dixi: not for me: not my id
(wmbus) parseDLL @0 50
(wmbus) parseELL @10 40
(wmbus) parseNWL @10 40
(wmbus) parseAFL @10 40
(wmbus) parseTPL @10 40
(meter) dixi: for me? 29319284 in 29318818

hngfngzl, 2023-01-25:
I tried – and IT WORKS! 😊
If you could you’d see a very happy face!

So I didn’t yet check the permissions on the /usr/bin/isql yet (I would have to google how this is done…)

Try

it works for me, but do not understand why you get 127 response. What is /usr/bin/isql are permitions?

hngfngzl, 2023-01-25:
> Glad to hear that everything is ok and no need to check file since it works.
> Do you have any other questions or we can close this issue?

No issues at the moment with the isql thing anymore. I’ll run it and will if necessary post to this or a new issue.

(For my curiosity: Any chance to read hekatron smoke detectors? They seem to have an own solution/box for this but I guess it uses the same technology with 866 mhz telegrams, like the watermeters and heatmeters do… Are you located in the US or in Germany/Europe – in Germany Hekatron is well known company for fire protection. But this could be a new issue…)

BIBOLV, 2023-01-26:
Ok, will close this issue and you can reopen if needed.
I`m from EU but have not heard of it, I`m using Lansen one. But if you have one, you can study information about it, if it does work on wmbus and if you have a key to decrypt telegrams, you can capture them and open new issue and we will take a look if it can be integrated.

closed 2023-01-26 by BIBOLV