wmbusmeters/wmbusmeters #793: support for apator APT-WMBUS-NA-1

slawomirg, 2023-01-09:
can support for APT-WMBUS-NA-1 be added? It identifies as unknown:
MainWater: meter detection could not find driver for id: 02547285 mfct: (APA) Apator, Poland (0x601) type: Water meter (0x07) ver: 0x14

slawomirg, 2023-01-09:
`telegram=|1C440106857254021407A0_CA00362D006E915E9976D6295EF9A42287B8|+14
telegram=|1C440106857254021407A0_CD008C521554DA8B5F58B7A823A95394AFAA|+43
telegram=|1C440106857254021407A1_CE008948044867CAF0A9E77EA7EA6E1E9F03|+53
telegram=|1C440106857254021407A0_CF00162670294D55549F8EDB6B1DBE042B0C|+63
telegram=|1C440106857254021407A0_D2000BCEC15967CA376B294A489201918F00|+93
telegram=|1C440106857254021407A0_D300BE754B835DAA6E1FB11D3C5F8B21D6FD|+103
telegram=|1C440106857254021407A1_D4009BEA2915F71941F7F89D0A55B70A52CB|+112
telegram=|1C440106857254021407A1_D70073F09DDC549FBA77E5519276B69D6FC0|+142
telegram=|1C440106857254021407A0_D900D41F32FC1E836F839B94C6E9A24949A2|+162`

weetmuts, 2023-01-20:
Is it encrypted? Can you provide the meter readings alongside the first and last telegram?

slawomirg, 2023-01-22:
If I understood correctly, it's encrypted with key 0...0 by my provider, 
the current value is around:298,81m^3
and the latest telegrams are:
telegram=|1C440106857254021407A0_430090761C93AEB8A4B4595094D70BA47FC6|+2026
telegram=|1C440106857254021407A0_47000107A5203D1AB5933FC7BE5F07F466BF|+2065
telegram=|1C440106857254021407A1_4800F0B9599921E927F69BA9C03449A118D5|+2074
telegram=|1C440106857254021407A0_49006C0ADE925405C9F3A9F1205372CBE9BC|+2084
telegram=|1C440106857254021407A0_4D002F0D1364919F7E479CEC1775E2290208|+2123
telegram=|1C440106857254021407A1_4E0085599AF02FAA24D2AAE10AED67FE4E0A|+2133
telegram=|1C440106857254021407A0_50001A62E73E9371AA26F4C7DE8192C75B88|+2152
telegram=|1C440106857254021407A1_540026CBF28FB6E2CABD57C07CED71F2F450|+2191
telegram=|1C440106857254021407A0_550027FDB632AEC052F39D4ED689811BFA60|+2201
telegram=|1C440106857254021407A1_5700FACB6ADDCEBF1A650492199BC72095D1|+2220
telegram=|1C440106857254021407A0_59001F12166576CF53B56C553E139F991997|+2239
telegram=|1C440106857254021407A0_5C00734E32F87CF59D16F24D646AA0E39816|+2268
telegram=|1C440106857254021407A1_6000455EDFA9A46497E3E4CF261FBE1AD0E0|+2307
telegram=|1C440106857254021407A0_61004AA492427AAF07B9CAF133EA75324D13|+2316
telegram=|1C440106857254021407A0_6200308CAE2CEB36A11749ECF16F97DF71D3|+2326
telegram=|1C440106857254021407A1_63004EB2A767BCA17E604B35B61D1454A244|+2335
telegram=|1C440106857254021407A0_680080E84982719BB18FFF645B4404BD7182|+2383
telegram=|1C440106857254021407A0_6B007F764CF600F28AF3E86D86E2FA99D7A5|+2412
telegram=|1C440106857254021407A1_6C002639BFFF8DD402F3A1C8C8D1A4B96F10|+2422
telegram=|1C440106857254021407A0_6D008368B69EBADC50E2A80506784E04A8F7|+2431
telegram=|1C440106857254021407A1_6F0006086515F07AD8B78852E6D7BE516E67|+2450
telegram=|1C440106857254021407A0_73006E695D980C039699466C66FC516BF0C7|+2488
telegram=|1C440106857254021407A0_790085D28D89E048D8AF06883CE75B979AE4|+2546
telegram=|1C440106857254021407A0_7A00EB5C7526E7B5E275DDA9A691DCBFB5E8|+2555
telegram=|1C440106857254021407A1_7B00C56D1C30CEF85943E60DBF724362C193|+2565
telegram=|1C440106857254021407A0_7C005C9412EEA2D14616B3D7DD0F39D008E2|+2574
telegram=|1C440106857254021407A1_7E00133FE692F28DCE7FDE138407A177778B|+2593
telegram=|1C440106857254021407A0_800018475C96AB4AE9A2172FA40129E43111|+2612
telegram=|1C440106857254021407A0_8300FCD3BB45EF7064B5D03F574EAD339D28|+2640
telegram=|1C440106857254021407A1_8400B2219032A32A7B6190F5CDA191DDA82B|+2650
telegram=|1C440106857254021407A0_85009681549E034C9E2F21401A1767B48D86|+2659
telegram=|1C440106857254021407A0_8600E5781FF09FD35BAAD48C1223E7A63832|+2669
telegram=|1C440106857254021407A1_870047C9049C8D830A2A92426C52C5CD6BA1|+2678
telegram=|1C440106857254021407A0_8800DD069FF98F34000E042FC27FD99CBD7F|+2688
telegram=|1C440106857254021407A1_8D0077DAC297E638C9FB0C27B9C8C05BDEC7|+2735
telegram=|1C440106857254021407A0_8E00F6BF8C8372C94963F6A18D771304A9FA|+2745
telegram=|1C440106857254021407A1_9000BD57B6F4F0E3E9266431EDB745419871|+2764
telegram=|1C440106857254021407A0_9100D79217486959AA024BB64E2956F12B64|+2773
telegram=|1C440106857254021407A1_930030F7F5B968AC586EF0ADB5DCDD574AD1|+2793
telegram=|1C440106857254021407A0_9400B23926BBCB54E9A5585ACD6049962D19|+2802
telegram=|1C440106857254021407A0_95009664BD2E19C7AD04F70099E34D2288F5|+2812
telegram=|1C440106857254021407A1_96003726906D2ABD0E3763171BAE13E5C042|+2821
telegram=|1C440106857254021407A0_9800EEBBEF0326FB37894E094834EF136F20|+2840
telegram=|1C440106857254021407A0_9A00A91ACD2ECB6BB97DD700A2FCE4601C99|+2859
telegram=|1C440106857254021407A0_A1003B9FE83B37406F75F53143AF4340E9C5|+2927
telegram=|1C440106857254021407A0_A300D6B49DE1858181557963D015A91AF266|+2946
telegram=|1C440106857254021407A0_A40018A31153FFF36B6974D3526F136FD367|+2956
telegram=|1C440106857254021407A1_A500A1A6FBB8BE46B0E4C0FCD9850B12A68F|+2965
telegram=|1C440106857254021407A0_A60032119CE545E30B37822F9F5AF7A61A3D|+2975
telegram=|1C440106857254021407A0_A7000FD05377CEF15EEF3F7D304FFB2BABF1|+2984
telegram=|1C440106857254021407A1_A8007CDA83FD375273E6F53D637EF885575F|+2994
telegram=|1C440106857254021407A1_AE0004C83B564E8FEA87E35596F63B183487|+3052
telegram=|1C440106857254021407A0_AF00C29AE5EB4027558B61FEA0DFA01AF67C|+3062
telegram=|1C440106857254021407A0_B200964F41F0417DF9105EA301A3CD0B5FA6|+3091
telegram=|1C440106857254021407A0_B3000FA7509E5644FCEF873E3977D3DBC853|+3101
telegram=|1C440106857254021407A1_B400F008CCC0FF4843D7EC19437076C0436C|+3110
telegram=|1C440106857254021407A0_B500AA252BB41CC34176187C23D0658B0001|+3120
telegram=|1C440106857254021407A0_B6002EA0420DE82ADC2DA048F0B0FCB423DA|+3130
telegram=|1C440106857254021407A0_B8001B2DA5B97FC3392E85E756A00617846C|+3149
telegram=|1C440106857254021407A0_BC00405227D3091E9AB1358875CE73E1094B|+3188
telegram=|1C440106857254021407A0_BF0016E207BE262A41C2853EE97E4647DEE2|+3217
telegram=|1C440106857254021407A1_C0009CE85E621726B8F06841FF2ED91F5ABD|+3227
telegram=|1C440106857254021407A0_C400F802E50E5D1ED6FABFDBAB14B48D713C|+3266
telegram=|1C440106857254021407A0_C5004FCDFB27396C870A45D3852C6812225D|+3276
telegram=|1C440106857254021407A0_CA004A7DAAD03AFAC6192B13DBF39C563108|+3325
telegram=|1C440106857254021407A1_CC00405D4CA01250CFD947DAAD8BEBD33042|+3345
telegram=|1C440106857254021407A0_CD004A43A485665F1A53254FB1FF270AF3D9|+3354
[... 8 lines cut ...]
telegram=|1C440106857254021407A1_DE0099BD36ACFEAAF22E44FB110D54D7A3D9|+3522
telegram=|1C440106857254021407A0_DF0024BAA8904D30E6A9C82354A13C549D9F|+3532
telegram=|1C440106857254021407A0_E000A2299B2CD9DDBD8920216ADF04EAF066|+3542
[... cut ...]
telegram=|1C440106857254021407A1_0500A6DF83A325441B0B9FC92E2EC013AE06|+3912
telegram=|1C440106857254021407A0_0600F14DD18DA45BF1EDA4D880C953691FE4|+3922
telegram=|1C440106857254021407A0_0700C1FBD00465462EF281D5474401B4AA3B|+3932
telegram=|1C440106857254021407A0_090017CF26C983D0F653F730319AC6C3160B|+3952
telegram=|1C440106857254021407A0_0D00DE9E51E74B185F18C7AF448CA7536CEF|+3992
telegram=|1C440106857254021407A0_0F00286BC31BC49500181AE1D5060C957C0D|+4012
telegram=|1C440106857254021407A0_120057E66B4F7B8979E8ECB6E9123035A605|+4042

kitor, 2023-01-23:
This is a meter overlay which support multiple "analog" meters, so data might contain different values based on that.
I just got meter "upgrade" with the same overlay, need to take my sdr stick next time to get some readings.

kitor, 2023-02-24:
@slawomirg Does your meter ID match the one printed on its casing / qr code? 

I see a lot of meters decoded as apator162 with  but none match my ID. They all seem to decode though with a mostly low values.  

If this is a retrofitted overlay to existing meter, I'd assume the reported value may not match meter reading. If they are lazy, it might start counting from 0 (as they have physical reading when overlay is installed anyway). In my case both meters seem to be replaced so I'd expect them to match.

slawomirg, 2023-02-24:
@kitor my meter came with overlay so both suppose to start with 0, the ID matches too; still, I am not reading it via wmbusmeters as it's unsupported and I frankly have no idea how I can resolve it myself :(

kitor, 2023-02-24:
Maybe mine were not activated or something. I spent some time in my basement (it has tick concrete walls and ceiling since it was a 2nd war kind-of bunker) and I saw nothing on my ID or my neighbor ID.

Even though I see something on 868.95M which is roughly every 10 seconds (which matches module doc), but seems like a single device transmission while I have two of those next to each other. I left wmbusmeters listening on those two IDs but without much hope.

I guess I'll have to wait till the next bill/reading to see if anything changes.

closed 2023-07-27 by github-actions[bot]