wmbusmeters/wmbusmeters #794: Correct telegrams ignored because of duplicate encrypted telegram

machinekoder, 2023-01-09:
I have some Techem water meters which send data encrypted and unencrypted using the same id. Reading the unencrypted telegram works fine, until an encrypted message arrives.

How can the encrypted message be ignored altogether? The default handling of an encrypted message is to ignore that id altogether, but that also ignores my correct messages.

```
(meter) Kaltwasser: meter detection did not match the selected driver mkradio4! correct driver is: mkradio4a
(meter) Not printing this warning again for id: 82511234 mfct: (TCH) Techem Service (0x5068) type: Radio converter (meter side) (0x37) ver: 0x95
(wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 82511234 mfct: (TCH) Techem Service (0x5068) type: Radio converter (meter side) (0x37) ver: 0x95
```

machinekoder, 2023-01-09:
device file
```
name=Kaltwasser
ver=0x95
id=82511234
driver=mkradio4
```

weetmuts, 2023-01-10:
Ouch, encrypted AND not encrypted from the same id. This is not anticipated by wmbusmeters. Do you have some example telegrams?

gppanayotov, 2023-01-27:
I have seem the same warning message but unencrypted telegrams are still received. 

`(meter) Heating: meter detection did not match the selected driver compact5! correct driver is: vario451
(meter) Not printing this warning again for id: 74738876 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x04) ver: 0x27
(wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 74738876 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x04) ver: 0x27
telegram=|37446850768873742743A2_123E2D004A0088F30A0000808D588619A659647ABB0000000000000000000000000000000040816971FA85A6B0|+779
telegram=|37446850768873742743A2_123E2D004A0088F30A0000808D588619A659647ABB0000000000000000000000000000000040816971FA85A6B0|+814`

And here are some examples of both encrypted and non encrypted telegrams: 
`Received telegram from: 74738876
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0x04) encrypted
                   ver: 0x27
                device: rtlwmbus[00000001]
                  rssi: 142 dBm
                driver: vario451
telegram=|_334468507688737427047AB400202536BD6F8A1B6B4C2506AFD369D7FD760B426C6A2B4FB6283B3F0B37A83A52C4BA0F04524EFB|+110

Received telegram from: 74738876
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0x43)
                   ver: 0x27
                device: rtlwmbus[00000001]
                  rssi: 88 dBm
                driver: unknown!
telegram=|_37446850768873742743A2123E2D004A0088BA0A000000FD548619A659647ABB0000000000000000000000000000000040816971FA85A6B0|+115
`

weetmuts, 2023-01-27:
Thanks @gppanayotov 
the first telegram is indeed encrypted and the second one decodes to 
Foo	74738876	6025 kWh	762.777778 kWh	5262.222222 kWh	2023-01-27 22:26.24
when using the vario451 driver. Are these the correct values?

@machinekoder 
You need to provide two telegrams one encrypted and the other not, otherwise I cant help you.

gppanayotov, 2023-01-27:
Actually I get wrong values with vario451 driver, with  compact5 driver I can see the correct values.

weetmuts, 2023-01-28:
Ok, it seems like the same meter id, sends different telegrams with different types 0x43 old mfct specific techem heater nr (not encrypteD), and new 0x04 standard compliant heat meter nr (encrypted).

The ignore is actually based on the meter id + version + type. So since the type differs the good telegram should be fine, if you select the right driver.

I have resisted adding specification of the type and version to the meter id so far, but maybe it is time to do it.

So to listen only to the encrypted meter you would write the id:
74738876.TCH.27.04
To listen to the non-encrypted meter you would write the id:
74738876.TCH.27.43

I suppose this would be the best solution, in particular since the two telegrams need two different drivers....
someone might actually want to decode both telegrams! :-)

closed 2023-04-30 by github-actions[bot]

ifurnadjiev, 2025-05-09:
I have the exact same issue but "id + version + type" are the same for both encrypted and not encrypted telegrams:
```
Received telegram from: 30886016
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08) encrypted
                   ver: 0x36
                device: rtlwmbus[00000001]
                  rssi: 22 dBm
                driver: qcaloric
```
```
Received telegram from: 30886016
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x36
                device: rtlwmbus[00000001]
                  rssi: 22 dBm
                driver: qcaloric
```

Anything I can do in this case?