wmbusmeters/wmbusmeters #796: Apator WM-Bus-08 FLOW water + Heat Allocator

Cinhead, 2023-01-09:
Hi, I've started HA just 1 week ago.
I'm not an programmer or tech/linux user so please take this in mind.

Plug-in is great but, but i cant figure some things out.
I have 4 water meters and 4 heat allocator (all belongs to housing Association, but there are old and not encrypted so far :-)
for dongle i'm using TV Stick DVB-T + DAB + FM RTL2832U + FC0012
https://pl.aliexpress.com/item/32914826153.html?spm=a2g0o.order_list.order_list_main.55.5c171c24VtLSIm&gatewayAdapt=glo2pol

I tried monitor all devices to read telegrams in the air (48 h - file size 21MB if someone wants it :0)
note 1 - nice to have in logs date/time of telegrams that will show when they arrived (now i have logs for 48h with no timestamps)
note 2 - nice to have possibility to set time/schedule  when wmbusmaster will collect data (it consumes 20% of my RPI and cause to run fan ;-) so during night i don't want to have buzzing fan near my head ;-)

so modified config like that 
> loglevel=normal
> device=auto:t1
> donotprobe=/dev/ttyAMA0
> logtelegrams=true
> format=json
> logfile=/config/wmbusmeters/logs/wmbusmeters.log
> shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"

So after 2 days i have some responses from WM-Bus-08 FLOW water
But they are detected as Gas meters
NOTE - meters numbers are in hex not in decimal

> Received telegram from: 000a482b
>           manufacturer: (APT) Unknown (0x8614)
>                   type: Gas meter (0x03)
>                    ver: 0x03
>                 device: rtlwmbus[00000001]
>                   rssi: 150 dBm
>                 driver: apator08
> telegram=|_734414862B480A000303A053190E000C403449B5040406FF0CE2990900E4F00900D9530A00F2C80A000A3C0B002CA50B007AF20B003A610C007FCE0C004F360D008DA30D001C100E0003000000003772042000000000000024000000000000280000000000002C0038041E06152F000000000000|+29656

After decoding https://wmbusmeters.org/analyze/ flow is accurate with reality.

I set those meters as follows:

>   - |-
>     name=Lazienka_Woda_Zimna
>     driver=apator08
>     id=000A482B
>     key=00000000000000000000000000000000

and so on...
But i cant set this up - i have error:
>  Adding meter: Lazienka_Woda_Zimna ...
>   File /config/wmbusmeters/etc/mqtt_discovery/apator08.json not found.
>  Adding meter: Lazienka_Woda_Ciepla ...
>   File /config/wmbusmeters/etc/mqtt_discovery/apator08.json not found.
>  Adding meter: Kuchnia_Woda_Zimna ...
>   File /config/wmbusmeters/etc/mqtt_discovery/apator08.json not found.
>  Adding meter: Kuchnia_Woda_Ciepla ...
>   File /config/wmbusmeters/etc/mqtt_discovery/apator08.json not found.
> MQTT Discovery cleanup...
> [23:45:33] INFO: Running wmbusmeters ...

If i change to apator162 error will not sow but reading are not present.
but steel have error:

> Not a valid meter id nor a valid meter match expression "00064DAC"
> Not a valid meter id nor a valid meter match expression "000A482B"
* those are number of meters Apator 08 in HEX that are cached in LOG files.
* case sensitive?

Second thing
I have 4 Heat allocator and during this time i recieved only 3 telegrams 
https://www.apator.com.pl/en/our-solutions/water-and-heat/heating-cost-allocators/e-itn-30-6#downloads
In manual it is claimed that its send 
30.51 ~ 5 ms with transmission frequency every 30 seconds in the first month and every 240 seconds in the other 11 months of the annual period

> Received telegram from: 35335932
>           manufacturer: (APA) Unknown (0x8601)
>                   type: Heat Cost Allocator (0x08)
>                    ver: 0x04
>                 device: rtlwmbus[00000001]
>                   rssi: 39 dBm
>                 driver: apatoreitn
> telegram=|_19440186325933350408A02100009D0300000000242EF0163C17|+64474

So i cant figure it out how can i have those readings. I tried move antenna to defriend spots. But its doesn't works.
Any suggestion will be appreciated in both issues.

Cinhead, 2023-01-11:
Update 3 of 4 Heat allocator showed up.
Still cant figure with apator08

BIBOLV, 2023-01-12:
Hi, lot of text so let me start from top - firstly welcome to the fun world of HA, tech/linux, etc :wink:
Logging all telegrams are not necessary unless you are solving something by that, so in your case I think you can disable to save space.
There is parameter `meterfilestimestamp` that would create files with timestamps as frequent as you want.
It will not be possible to achieve any set times or schedules when running as HA addon. If running a standalone version then there are some workarounds for that - you can search in issues here, it was discussed and some community members shared their experience. Or you can replace RTL-SDR with purpose built wmbus dongle like imst that would significantly reduce load on raspi.
Sometimes it happens when meter manufacturer messes up meter identification, so it could happen when one meter type shows up as different meter type.
Now about the error about apator08.json - it is HA template file for each meter that is not been created yet as HA addon is in kind of experimental phase. But it does not prevent you from setting up meter manually in HA, this error is just saying that it will not be able to set it up automatically. Reading should be in HA MQTT.
Are those heat allocators significantly further away than water meter? Can you move closer to them to see if reception would be better? 
Looks like there is no driver for those, so it would be needed from your side to provide at least 10 telegrams from meters together with some reference readings and as much details about them as possible. Apator is known to not use OMS standard but implement some vendor specific fields with unknown structure, but hopefully @weetmuts will have some time sooner or later to take a look if it is possible to decode it and implement driver.

Santa77, 2023-01-13:
Some more data if they are needed...

(APA) Unknown (0x8601)
  "decId": "36185805",
  "dataRaw": "26440186055818360508A021002D2E00000600A600060005000000000000000000000000000000A00FBAB0",
  "decId": "36186720",
  "dataRaw": "26440186206718360508A021002D2E000000001400000004000000000000000000000000000000A00FBAB0",
  "decId": "36185848",
  "dataRaw": "26440186485818360508A021002D2EFD280000C7040000AB00BD0044001B000000000000000100A00F240D",
  "decId": "36185871",
  "dataRaw": "26440186715818360508A021002D2E00000000AA00000004000100000001000000000000000000A00F5860",

Cinhead, 2023-01-13:
HI thank you for the answer.
I've stopped logging. After 3 days i found that all heat allocators work and sending data.
[image]
Its in polish, but you can see that I've extracted only this and previous years (don't need any other data, but they are present).
[image]

But i cant see apator 08 data.
I've copyed and renamed apator162.json to apator08.json cous those have only total water meter in m^3.
it didn't worked. at least there is no error shown but steel i have no mqtt data to those meters.
expl telegram:

> Received telegram from: 00035771
>           manufacturer: (APT) Unknown (0x8614)
>                   type: Gas meter (0x03)
>                    ver: 0x03
>                 device: rtlwmbus[00000001]
>                   rssi: 146 dBm
>                 driver: apator08
> telegram=|_73441486715703000303A0954E01000840342C2CE50306FF0C96F100008BFD00002608010062130100E11B01004421010097250100312A010034330100123A0100E4400100AF4D0100030000000037EE052000000000000024000000000000280000000000002C0000000000002F000000000000|+64492

After decoding its like dhat:
> Auto driver  : apator08
> Best driver  : apator08 04/109
> Using driver : apator08 00/00
> 000   : 73 length (115 bytes)
> 001   : 44 dll-c (from meter SND_NR)
> 002   : 1486 dll-mfct (APT)
> 004   : ac4d0600 dll-id (00064dac)
> 008   : 03 dll-version
> 009   : 03 dll-type (Gas meter)
> 010   : a0 tpl-ci-field (Mfct specific)
> 011 C?: 9F890C000840343B2CE50306FF0C7A99080008F308007967090016E50900654D0A00ABA30A006CD70A00B41C0B0039720B00C6C50B00CC1B0C00DD7F0C0003000000003733062000000000000024000000000000280000000000002C0000000000002F000000000000 mfct specific
> 025 C!: *** 10
> 9f890c00 total consumption (273.887667 m3)
Total consumption is accurate.

in MQTT i see only allocators:
[image]
I cant see where and how i can manually add my flow meter
either MQTT or wmbusmaster.

BIBOLV, 2023-01-15:
Connect to mqtt and subscribe to all ha channel - `homeassistant/#` and check where are meter info is being sent to. If nothing then subscribe to all mqtt - `#`. When you know where details are being sent to, add meter manually, some details are [here](https://weetmuts.github.io/wmbusmeterswiki/HASS.html) or you can check old issues.

Cinhead, 2023-01-16:
Hi,
I received in MQTT telegrams from all meters.

> Wiadomość 139 otrzymana w wmbusmeters/Kuchnia_Woda_Ciepla o godzinie 21:50:
>  {
>     "media": "water",
>     "meter": "apator08",
>     "name": "Kuchnia_Woda_Ciepla",
>     "id": "00035771",
>     "total_m3": 28.797667,
>     "timestamp": "2023-01-16T19:03:18Z",
>     "device": "rtlwmbus[00000001]",
>     "rssi_dbm": 143
> }
> QoS: 0 - Retain: true

I had entries to configuration.yaml

> mqtt:
>   sensor:
>     - state_topic: "wmbusmeters/Lazienka_Woda_Zimna"
>       json_attributes_topic: "wmbusmeters/Lazienka_Woda_Zimna"
>       unit_of_measurement: m³
>       value_template: "{{ value_json.total_m3 }}"
>       name: Lazienka_Woda_Zimna
>       icon: "mdi:gauge"
>       state_class: total_increasing
>       device_class: water
> 
But my mistake was that I've left "name" the same to all meters...
now works fine (only to many digits after coma, but its not issiue)
[image]

Thank you for support.

BIBOLV, 2023-01-17:
Glad to hear that everything works. Closing issue.

closed 2023-01-17 by BIBOLV