wmbusmeters/wmbusmeters #797: Energy MJ

pmiroslav, 2023-01-10:
Explain the problem..... 

https://wmbusmeters.org/analyze/5e44a5115376916140047a0B0050052f2f0c0e829311008c100e000000000c14014938000c2B751400000B3B2902000a5a52070a5e95060a6256000a279015cc020e92831100cc021478113800c2026cdf2c2f2f2f2f2f2f2f2f2f2f2f2f2f 

I have got Diehl Sharky 775 heat meter. I need to decode MJ.
I trayed sharky774 but the value was kWh not MJ.  
From this telegram correct value is 119382MJ.

How do I have to change driver?

weetmuts, 2023-01-10:
wmbusmeters normalizes all values. >All< energy values are translated into kwh and the conversion in your telegram is correct. 119382MJ = 33161.666667 KWH

You can add a calculated MJ value like this:
--calculate_total_energy_consumption_mj=total_energy_consumption_kwh
Drop the -- if you add the expression to the meter file in the config dif.

weetmuts, 2023-01-10:
That said, some drivers (including sharky, but not sharky774) looked for EnergyWh which means they
missed when the telegram was sent with MJ. I have now pushed a fix where all the remaining drivers (inluding sharky)
has been changed to look for AnyEnergyVIF, which will match both Wh and MJ.

Please test!

snapspizza, 2023-01-11:
@pmiroslav could you elaborate on how you are using wmbusmeters? Are you running it from command line, in docker, in HA add-on or something else?

snapspizza, 2023-01-11:
Then you should add the line to the meter file like this:

```
name=heat
driver=sharky
id=64313374
calculate_total_energy_consumption_mj=total_energy_consumption_kwh
```

pmiroslav, 2023-01-11:
ok it works thanx

BIBOLV, 2023-01-12:
Question answered - closing.

closed 2023-01-12 by BIBOLV