wmbusmeters/wmbusmeters #801: Radiocrafts rc1180 not recognized

pisquan8, 2023-01-13:
Hi,
I'm trying to setup wmbusmeters with my Radiocrafts rc1180 but I'm having troubles getting the radio to be recognized.
So far I have tried both the Home Assistant add-on and the stable Snap (that at this moment is version 1.10.1) on my Pop! OS machine with the same results.

In both cases the radio is not recognized as rc1180 even if in the wmbusmeters.conf file I specify:
`device=/dev/ttyUSB0:rc1180:t1`
and /dev/ttyUSB0 is the correct device (I can connect with screen and configure the radio).

If I enable debug log level I see that wmbusmeters tries to detect my rc1180 but thinks it fails (but it seems to me that both replies are OK). Here is the interesting portion of my log:
```
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary "3E"
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary "0105030100016400053C00D391DA80807C0002010000000000482400000547010301010000000004FF0800000000410005080001050000012B0000060B000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00000000000000000000000000000000005243313138302D4D5043312C322E30302C302E31384500FFFFFFFFFFFFFFFF00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3E"
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
Expected rc1180 on /dev/ttyUSB0 but did not find it! Ignoring tty!
```

Maybe the problem here is that my rc1180 replies to the second request (_the one that asks for the configuration frame, I believe?_) differently to how wmbusmeters expects. If I look in wmbus_rc1180.cc, around row 87, the second half of the configuration frame is expected to be filled with FF, mine is half 00 and half FF.
Any ideas?

weetmuts, 2023-01-14:
Yep, actually the code had a comment, commenting on this assumption! :-)
Fix pushed! Please test!

pisquan8, 2023-01-16:
Hi,
the fix is good, now my rc1180 is working.
I think I can close it now.

closed 2023-01-16 by pisquan8