wmbusmeters/wmbusmeters #808: Add support for camstrup

sveneb, 2023-01-16:


BIBOLV, 2023-01-17:
Please record a few telegrams and post them here.

sveneb, 2023-01-17:
Here are a few records:
Received telegram from: 82835301
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Heat volume at inlet meter (0x0c) encrypted
                   ver: 0x40
                device: amb8465[00000000]
                  rssi: -85 dBm
                driver: unknown!
telegram=|_3C442D2C01538382400C8D202D60ED2120651E8C98F01CD4CEF4D8660E6BF200C04C487D62C82F0928FDB528EC652818D61191AD3345063102B03F0AE9|+23

(...)
telegram=|_3C442D2C01538382400C8D202E80ED212032A75A0184D5BB8692B9B2CDCCC937CAD39D27E25942A52D819E5FAD3995F7DB58C5C61E91E012E36A406AF0|+120

(...)
telegram=|_53442D2C01538382400C8D202FA0ED2120823CCEF93020A9668B38774786A744FF8CEDB33F13586D4A36321EC96E98B5C1F29CD6882BA58182EC40FFCD32FC4E7F6F4187ABD0165DABEE258629A6C64B891083EE|+217

(...)
telegram=|_3C442D2C01538382400C8D2030B0ED2120205F0AC35D07240DAF7CF350DF4E2A6C57217489DF1D2615F33D6368071F70AE310414F8153043F7AD4775F0|+313

Let me know if this is what you need, or you need more information
Thanks in advance
Sveneb

BIBOLV, 2023-01-18:
Those [telegrams](https://wmbusmeters.org/analyze/3c442d2c01538382400c8d2030B0ed2120205f0ac35d07240daf7cf350df4e2a6c57217489df1d2615f33d6368071f70ae310414f8153043f7ad4775f0) looks to be encrypted. Do you have decryption key? Can you post decrypted telegrams?

SebastiaanBreedveld, 2023-01-20:
I proposed this update last week.

Diff:

*** 42,47 ****
--- 42,48 ----
          di.addDetection(MANUFACTURER_KAM, 0x0d,  0x30); // 302
          di.addDetection(MANUFACTURER_KAM, 0x0c,  0x30); // 302
          di.addDetection(MANUFACTURER_KAM, 0x04,  0x40); // 303
+ 	di.addDetection(MANUFACTURER_KAM, 0x0c,  0x40); // 303
          di.addDetection(MANUFACTURER_KAM, 0x0a,  0x34); // 403
          di.addDetection(MANUFACTURER_KAM, 0x0b,  0x34); // 403
          di.addDetection(MANUFACTURER_KAM, 0x0c,  0x34); // 403

A telegram:
(TPL) decrypted  "2F2F0405352B00000413A906070004FF07F255020004FF08553502000259470B025D0A0A023B9F0502FF220000026CF42144052F000000441302AD0000426CE1212F2F2F2F2F2F2F2F2F2F2F2F2F2F2F"

Processed output:
(shell) env "METER_JSON={"media":"heat volume at inlet","meter":"kamheat","name":"Kamstrup","id":"82848073","status":"OK","total_energy_consumption_kwh":1106.1,"total_volume_m3":460.457,"volume_flow_m3h":1.439,"t1_temperature_c":28.87,"t2_temperature_c":25.7,"forward_energy_m3c":153074,"return_energy_m3c":144725,"meter_date":"2023-01-20","target_energy_kwh":4.7,"target_volume_m3":44.29,"target_date":"2023-01-01","timestamp":"2023-01-20T20:14:15Z","device":"rtlwmbus[00000001]","rssi_dbm":134}"

BIBOLV, 2023-01-20:
Thanks, could you post few more telegrams - this one does not parse for some reason.

SebastiaanBreedveld, 2023-01-20:
Sure!

(TPL) decrypted  "2F2F0405342B000004137806070004FF07E555020004FF084835020002598B08025D700A023B9B0502FF220000026CF42144052F000000441302AD0000426CE1212F2F2F2F2F2F2F2F2F2F2F2F2F2F2F"

(TPL) decrypted  "2F2F0405342B000004137806070004FF07E555020004FF084835020002598B08025D700A023B9B0502FF220000026CF42144052F000000441302AD0000426CE1212F2F2F2F2F2F2F2F2F2F2F2F2F2F2F"

(TPL) decrypted  "2F2F0405631800000413A499040004FF07587C010004FF08246B010002594B05025D5205023B000002FF220000026CEF2144052F000000441302AD0000426CE1212F2F2F2F2F2F2F2F2F2F2F2F2F2F2F"

BIBOLV, 2023-01-20:
Can you post `telegram=|` lines?

SebastiaanBreedveld, 2023-01-20:
Here is a full packet:
```
(serial) received ascii "T1;1;1;2023-01-20 20:57:07.000;134;133;82848073;0x5e442d2c73808482400c7af40050257b771b91c39e5dc8b68abfb33782dc738b0d3c0483f36622ea856c3fdc4bd4590d782d727d91f6240bf89cf81b62a76afce1512a39ef6e45ba8eca17c7a8cead706ec502964d61a7c65b23e39ee96d02<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-01-20 20:57:07.000;134;133;82848073;0x5e442d2c73808482400c7af40050257b771b91c39e5dc8b68abfb33782dc738b0d3c0483f36622ea856c3fdc4bd4590d782d727d91f6240bf89cf81b62a76afce1512a39ef6e45ba8eca17c7a8cead706ec502964d61a7c65b23e39ee96d02
<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 95
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(meter) Kamstrup: for me? 82848073 in 82848073
(meter) Kamstrup: yes for me
(meter) Kamstrup(1) kamheat  handling telegram from 82848073
(meter) Kamstrup 82848073 "5E442D2C73808482400C7AF40050257B771B91C39E5DC8B68ABFB33782DC738B0D3C0483F36622EA856C3FDC4BD4590D782D727D91F6240BF89CF81B62A76AFCE1512A39EF6E45BA8ECA17C7A8CEAD706EC502964D61A7C65B23E39EE96D02"
(wmbus) parseDLL @0 95
(telegram) DLL L=5e C=44 (from meter SND_NR) M=2c2d (KAM) A=82848073 VER=40 TYPE=0c (Heat volume at inlet meter) (driver kamheat) DEV=rtlwmbus[00000001] RSSI=134
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(TPL) num encrypted blocks 5 (80 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "7B771B91C39E5DC8B68ABFB33782DC738B0D3C0483F36622EA856C3FDC4BD4590D782D727D91F6240BF89CF81B62A76AFCE1512A39EF6E45BA8ECA17C7A8CEAD706EC502964D61A7C65B23E39EE96D02"
(TPL) IV 2D2C73808482400CF4F4F4F4F4F4F4F4
(TPL) decrypted  "2F2F04056B2B00000413B70A070004FF078157020004FF08B536020002594C0E025D7F0E023BAF0502FF220000026CF42144052F000000441302AD0000426CE1212F2F2F2F2F2F2F2F2F2F2F2F2F2F2F"
(telegram) TPL CI=7a ACC=f4 STS=00 CFG=2550 (synchronous AES_CBC_IV nb=5 cntn=0 ra=0 hc=0)
(meters) trying field info on_time(Time)[0]...
(meters) trying field info status(Text)[1]...
(meters) trying field info total_energy_consumption(Energy)[2]...
(meters) using field info total_energy_consumption(Energy)[2] to extract 0405 at offset 19
(meter) AnyEnergyVIF total_energy_consumption_kwh decoded kwh default kwh value 1111.5
(meters) trying field info total_volume(Volume)[3]...
(meters) using field info total_volume(Volume)[3] to extract 0413 at offset 25
(meter) Volume total_volume_m3 decoded m3 default m3 value 461.495
(meters) trying field info volume_flow(Flow)[4]...
(meters) using field info volume_flow(Flow)[4] to extract 023B at offset 53
(meter) VolumeFlow volume_flow_m3h decoded m3h default m3h value 1.455
(meters) trying field info power(Power)[5]...
(meters) trying field info max_power(Power)[6]...
(meters) trying field info t1_temperature(Temperature)[7]...
(meters) using field info t1_temperature(Temperature)[7] to extract 0259 at offset 45
(meter) FlowTemperature t1_temperature_c decoded c default c value 36.6
(meters) trying field info t2_temperature(Temperature)[8]...
(meters) using field info t2_temperature(Temperature)[8] to extract 025D at offset 49
(meter) ReturnTemperature t2_temperature_c decoded c default c value 37.11
(meters) trying field info max_flow(Flow)[9]...
(meters) trying field info forward_energy(Energy)[10]...
(meters) using field info forward_energy(Energy)[10] to extract 04FF07 at offset 32
(meter) Any forward_energy_m3c decoded m3c default m3c value 153473
(meters) trying field info return_energy(Energy)[11]...
(meters) using field info return_energy(Energy)[11] to extract 04FF08 at offset 39
(meter) Any return_energy_m3c decoded m3c default m3c value 145077
(meters) trying field info meter_date(Text)[12]...
(meters) using field info meter_date(Text)[12] to extract 026C at offset 62
(meters) trying field info target_energy(Energy)[13]...
(meters) using field info target_energy(Energy)[13] to extract 4405 at offset 66
(meter) AnyEnergyVIF target_energy_kwh decoded kwh default kwh value 4.7
(meters) trying field info target_volume(Volume)[14]...
(meters) using field info target_volume(Volume)[14] to extract 4413 at offset 72
(meter) Volume target_volume_m3 decoded m3 default m3 value 44.29
(meters) trying field info target_date(Text)[15]...
(meters) using field info target_date(Text)[15] to extract 426C at offset 78
(kamheat) 000   : 5e length (94 bytes)
(kamheat) 001   : 44 dll-c (from meter SND_NR)
(kamheat) 002   : 2d2c dll-mfct (KAM)
(kamheat) 004   : 73808482 dll-id (82848073)
(kamheat) 008   : 40 dll-version
(kamheat) 009   : 0c dll-type (Heat volume at inlet meter)
(kamheat) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(kamheat) 011   : f4 tpl-acc-field
[... 91 lines cut ...]
(shell) env "METER_TIMESTAMP_UT=1674248227"
(shell) env "METER_TIMESTAMP_LT=2023-01-20 21:57.07"
(shell) env "METER_DEVICE=rtlwmbus[00000001]"
(shell) env "METER_RSSI_DBM=134"
(shell) env "METER_ON_TIME_Hour=null"
(shell) env "METER_STATUS=OK"
(shell) env "METER_TOTAL_ENERGY_CONSUMPTION_KWH=1111.5"
(shell) env "METER_TOTAL_VOLUME_M3=461.495"
(shell) env "METER_VOLUME_FLOW_M3H=1.455"
(shell) env "METER_POWER_KW=null"
(shell) env "METER_MAX_POWER_KW=null"
(shell) env "METER_T1_TEMPERATURE_C=36.6"
(shell) env "METER_T2_TEMPERATURE_C=37.11"
(shell) env "METER_MAX_FLOW_M3H=null"
(shell) env "METER_FORWARD_ENERGY_M3C=153473"
(shell) env "METER_RETURN_ENERGY_M3C=145077"
(shell) env "METER_METER_DATE=2023-01-20"
(shell) env "METER_TARGET_ENERGY_KWH=4.7"
(shell) env "METER_TARGET_VOLUME_M3=44.29"
(shell) env "METER_TARGET_DATE=2023-01-01"
(shell) waiting for child 802438 to complete.
(shell) /bin/sh: return code 0

```
Was compiled with a checkout on 13 January, with the only change being the addition of (0x0c,  0x40).

BIBOLV, 2023-01-20:
Sorry for troubling you but could you set `logtelegrams=true`, restart service and post few `telegram=|`. Those interim strings I can`t process.

SebastiaanBreedveld, 2023-01-20:
Ah sorry, I thought the telegrams were anyway logged in debug mode. Here is some more:

telegram=|5E442D2C73808482400C7A620050252F2F_04056C2B000004138A0B070004FF07C657020004FF08FD3602000259B209025D6A09023B000002FF220000026CF42144052F000000441302AD0000426CE1212F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+28

telegram=|5E442D2C73808482400C7A6E0050252F2F_04056C2B000004138A0B070004FF07C657020004FF08FD36020002594B09025DFA08023B000002FF220000026CF42144052F000000441302AD0000426CE1212F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+216

BIBOLV, 2023-01-20:
Perfect, tnx!

BIBOLV, 2023-01-21:
Could you confirm that everything is working so we could close this issue?

SebastiaanBreedveld, 2023-01-23:
Did a fresh clone, and it all works!

closed 2023-01-23 by BIBOLV