wmbusmeters/wmbusmeters #809: (meter) newly created meter (Oel 68030415 kamheat) did not handle telegram!

domii666, 2023-01-17:
hello,

i have a problem with wmbusmeters. do you have an idea?

Config:
`data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: |-
  loglevel=normal
  device=/dev/ttyUSB0:im871a:c1
  donotprobe=/dev/ttyAMA0
  logtelegrams=false
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - |-
    name=Oel
    driver=kamheat
    id=68030415
    key=E01C3F333841F1A7BA147D7897B0C95E
mqtt: {}
`

logfile

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[13:45:49] INFO: Syncing wmbusmeters configuration ...
[13:45:49] INFO: Registering meters ...
[13:45:50] INFO: Adding meter-0001 ...
[13:45:50] INFO: Generating MQTT configuration ... 
[13:45:51] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[13:45:52] INFO: Running wmbusmeters ...
Started config im871a[00103849] on /dev/ttyUSB0 listening on c1
(meter) newly created meter (Oel 68030415 kamheat) did not handle telegram!`

without the meter in the config i will get this in the protocol, i have 2 kamstrup multical302, 6803010(heatpump) and 380315(oil), i have both encryption keys. 

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[13:51:30] INFO: Syncing wmbusmeters configuration ...
[13:51:30] INFO: Registering meters ...
[13:51:30] INFO: Generating MQTT configuration ... 
[13:51:31] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[13:51:33] INFO: Running wmbusmeters ...
Started config im871a[00103849] on /dev/ttyUSB0 listening on c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 68030415
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Heat meter (0x04) encrypted
                   ver: 0x30
                device: im871a[00103849]
                  rssi: -76 dBm
                driver: kamheat
Received telegram from: 68030410
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Heat meter (0x04) encrypted
                   ver: 0x30
                device: im871a[00103849]
                  rssi: -87 dBm
                driver: kamheat`

thank you 

dominik

BIBOLV, 2023-01-17:
Hi, change logtelegrams to true, collect some and post them here.

domii666, 2023-01-17:
Hi,

here the config with both kamstrup

data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: |-
  loglevel=normal
  device=/dev/ttyUSB0:im871a:c1,t1
  donotprobe=/dev/ttyAMA0
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - |-
    name=WP
    driver=kamheat
    id=68030410
    key=F84CC96904F52665881E79E1DC752FD1
  - |-
    name=Oel
    driver=kamheat
    id=68030415
    key=E01C3F333841F1A7BA147D7897B0C95E
mqtt:
  host: 192.168.1.16
  port: 1883
  user: mosquitto
  password: topsecret

and the log

telegram=|29442D2C1504036830048D2086A12AA920DF4E79C7291D78_117500304F00624403FDA8017E6401DF2C00|+2747
telegram=|29442D2C1504036830048D2087A12AA920DF4E79C7291D78_117500304F00624403FDA8017E6401DF2C00|+2763
telegram=|29442D2C1504036830048D2088B02AA920DF4E79C7291D78_117500304F00624403FDA8017E6401DF2C00|+2778
telegram=|36442D2C1004036830048D20F8D12AA9203FDB78_0305707C054305303E054313C9D80C43FF0731103243FF08D9A82D426CDF2C01FF2100|+2793
telegram=|29442D2C1504036830048D2089B02AA920DF4E79C7291D78_117500304F00624403FDA8017E6401DF2C00|+2794
telegram=|29442D2C1004036830048D20F9E02AA920034979C729CBB3_707C05303E05C9D80C311032D9A82DDF2C00|+2809
telegram=|29442D2C1504036830048D208AB12AA920589379C729D424_127500304F00624403FDA8017E6401DF2C00|+2810
telegram=|29442D2C1504036830048D208BB12AA920589379C729D424_127500304F00624403FDA8017E6401DF2C00|+2825
telegram=|29442D2C1004036830048D20FAE02AA920034979C729CBB3_707C05303E05C9D80C311032D9A82DDF2C00|+2826
telegram=|36442D2C1504036830048D208CC02AA920204C78_03051275004305304F00431362440343FF07FDA80143FF087E6401426CDF2C01FF2100|+2841
telegram=|29442D2C1004036830048D20FBE12AA9207E0279C7298C87_717C05303E05C9D80C311032D9A82DDF2C00|+2842
telegram=|29442D2C1504036830048D208DC12AA920589379C729D424_127500304F00624403FDA8017E6401DF2C00|+2856
telegram=|29442D2C1004036830048D20FCE12AA9207E0279C7298C87_717C05303E05C9D80C311032D9A82DDF2C00|+2859
telegram=|29442D2C1504036830048D208EC22AA92025D879C7299310_137500304F00624403FDA8017E6401DF2C00|+2872
telegram=|29442D2C1504036830048D208FC22AA92025D879C7299310_137500304F00624403FDA8017E6401DF2C00|+2887
telegram=|29442D2C1004036830048D20FEF02AA9207E0279C7298C87_717C05303E05C9D80C311032D9A82DDF2C00|+2892
telegram=|29442D2C1504036830048D2090D02AA92025D879C7299310_137500304F00624403FDA8017E6401DF2C00|+2903
telegram=|29442D2C1504036830048D2091D02AA92025D879C7299310_137500304F00624403FDA8017E6401DF2C00|+2919
telegram=|29442D2C1504036830048D2092D12AA920E6C379C729469D_147500304F00624403FDA8017E6401DF2C00|+2934
telegram=|29442D2C1504036830048D2093D12AA920E6C379C729469D_147500304F00624403FDA8017E6401DF2C00|+2950
telegram=|36442D2C1504036830048D2094E02AA920B2F578_03051475004305304F00431362440343FF07FDA80143FF087E6401426CDF2C01FF2100|+2966
telegram=|29442D2C1504036830048D2095E12AA920E6C379C729469D_147500304F00624403FDA8017E6401DF2C00|+2981
telegram=|29442D2C1504036830048D2096E22AA920E6C379C729469D_147500304F00624403FDA8017E6401DF2C00|+2997
telegram=|29442D2C1504036830048D2097E22AA920E6C379C729469D_147500304F00624403FDA8017E6401DF2C00|+3013
telegram=|29442D2C1504036830048D2098F02AA9209B8879C72901A9_157500304F00624403FDA8017E6401DF2C00|+3028
telegram=|29442D2C1504036830048D2099F02AA9209B8879C72901A9_157500304F00624403FDA8017E6401DF2C00|+3044
telegram=|29442D2C1504036830048D209AF12AA9209B8879C72901A9_157500304F00624403FDA8017E6401DF2C00|+3060
telegram=|29442D2C1504036830048D209BF12AA9209B8879C72901A9_157500304F00624403FDA8017E6401DF2C00|+3075
telegram=|36442D2C1504036830048D209C002BA9203C9D78_03051675004305304F00431362440343FF07FDA80143FF087E6401426CDF2C01FF2100|+3091
telegram=|29442D2C1504036830048D209D012BA9201C5579C729C8F5_167500304F00624403FDA8017E6401DF2C00|+3107
telegram=|29442D2C1504036830048D209E022BA9201C5579C729C8F5_167500304F00624403FDA8017E6401DF2C00|+3123
telegram=|29442D2C1504036830048D209F022BA9201C5579C729C8F5_167500304F00624403FDA8017E6401DF2C00|+3138
telegram=|29442D2C1504036830048D20A0102BA9201C5579C729C8F5_167500304F00624403FDA8017E6401DF2C00|+3154
telegram=|29442D2C1504036830048D20A1102BA9201C5579C729C8F5_167500304F00624403FDA8017E6401DF2C00|+3170
telegram=|29442D2C1004036830048D200F402BA920478F79C729D762_747C05303E05C9D80C311032D9A82DDF2C00|+3171
telegram=|29442D2C1504036830048D20A2112BA920611E79C7298FC1_177500304F00624403FDA8017E6401DF2C00|+3186
telegram=|29442D2C1504036830048D20A3112BA920611E79C7298FC1_177500304F00624403FDA8017E6401DF2C00|+3201
[... 41 lines cut ...]
telegram=|29442D2C1504036830048D20C8B22BA920DF8279C729CE8F_1B7500304F00624403FDA8017E6401DF2C00|+3790
telegram=|29442D2C1004036830048D2036E02BA9207ECE79C7295F70_7B7C05303E05C9D80C311032D9A82DDF2C00|+3805
telegram=|29442D2C1504036830048D20C9B22BA920DF8279C729CE8F_1B7500304F00624403FDA8017E6401DF2C00|+3806
telegram=|29442D2C1004036830048D2037E12BA9207ECE79C7295F70_7B7C05303E05C9D80C311032D9A82DDF2C00|+3821
telegram=|29442D2C1504036830048D20CAC02BA920DF8279C729CE8F_1B7500304F00624403FDA8017E6401DF2C00|+3822
telegram=|36442D2C1004036830048D2038F02BA920AB1878_03057B7C054305303E054313C9D80C43FF0731103243FF08D9A82D426CDF2C01FF2100|+3837
telegram=|29442D2C1504036830048D20CBC02BA920DF8279C729CE8F_1B7500304F00624403FDA8017E6401DF2C00|+3838
telegram=|36442D2C1504036830048D20CCC12BA9203AE778_03051B75004305304F00431362440343FF07FDA80143FF087E6401426CDF2C01FF2100|+3854
telegram=|29442D2C1504036830048D20CDD02BA920DF8279C729CE8F_1B7500304F00624403FDA8017E6401DF2C00|+3871
[... cut ...]
telegram=|36442D2C1504036830048D20D4F02BA920EF6A78_03051C75004305304F00431362440343FF07FDA80143FF087E6401426CDF2C01FF2100|+3983
telegram=|29442D2C1004036830048D2042122CA920BDD579C7298AFD_7C7C05303E05C9D80C311032D9A82DDF2C00|+3997
telegram=|29442D2C1504036830048D20D5F12BA9201C9979C7291B02_1C7500304F00624403FDA8017E6401DF2C00|+4000
telegram=|29442D2C1004036830048D2043202CA920C09E79C729CDC9_7D7C05303E05C9D80C311032D9A82DDF2C00|+4013
telegram=|29442D2C1504036830048D20D6F22BA9201C9979C7291B02_1C7500304F00624403FDA8017E6401DF2C00|+4016

thank you
dominik

BIBOLV, 2023-01-18:
There are two types of telegrams - [short](https://wmbusmeters.org/analyze/29442D2C1504036830048D20D6F22BA9201C9979C7291B021C7500304F00624403FDA8017E6401DF2C00) and [long](https://wmbusmeters.org/analyze/36442D2C1504036830048D20D4F02BA920EF6A7803051C75004305304F00431362440343FF07FDA80143FF087E6401426CDF2C01FF2100).
In long telegrams all fields except two manufacturer specific ones are extracted:
{code}
{
    "media":"heat",
    "meter":"kamheat",
    "name":"",
    "id":"68030415",
    "status":"OK",
    "total_energy_consumption_kwh":2998,
    "target_energy_kwh":2027.2,
    "target_volume_m3":214.114,
    "target_date":"2022-12-31",
    "timestamp":"2023-01-18T07:13:59Z"
}
{code}

domii666, 2023-01-18:
ok and why this is null?

(shell) env "METER_TOTAL_VOLUME_M3=null"
(shell) env "METER_VOLUME_FLOW_M3H=null"
(shell) env "METER_POWER_KW=null"
(shell) env "METER_MAX_POWER_KW=null"
(shell) env "METER_T1_TEMPERATURE_C=null"
(shell) env "METER_T2_TEMPERATURE_C=null"
(shell) env "METER_MAX_FLOW_M3H=null"
(shell) env "METER_FORWARD_ENERGY_M3C=null"
(shell) env "METER_RETURN_ENERGY_M3C=null"
(shell) env "METER_METER_DATE=null"

BIBOLV, 2023-01-18:
That is because driver can extract more information from different meters, but it should not matter in your case as in MQTT you should see just info that is being detected, eg. `{"media":"heat","meter":"kamheat","name":"Oel","id":"68030415","status":"OK","total_energy_consumption_kwh":3002.9,"target_energy_kwh":2027.2,"target_volume_m3":214.114,"target_date":"2022-12-31","timestamp":"2023-01-17T18:20:07Z","device":"im871a[00103849]","rssi_dbm":-79}` or it is not so?

BIBOLV, 2023-01-18:
No, sorry, in your case ha-addon template is being used, where all available fields are described and ones that are not present are being displayed as null, so you can either remove those null readings from ha dashboard card, or implement meter manually in ha based on mqtt message.

domii666, 2023-01-18:
thank you for the information!

BIBOLV, 2023-01-18:
Do you have any more questions or we can close this issue?

domii666, 2023-01-18:
we can close this, thank you for the help

closed 2023-01-18 by domii666