wmbusmeters/wmbusmeters #810: Zenner warm and cold water meter not detected correctly

probutus, 2023-01-17:
I got warm and cold water zenner meters which are not detected properly by minomess driver.

I have the specification from Zenner and it matches the description in the driver_minomess.c file

Can someone add the detection of the meter to the minomess driver?

Log with telegram below:

Cold Water meter:
telegram=|6644496A4425155518377202742215496A0116360050052F2F_0C1355000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+14
MyTapWater 15227402 2023-01-12 22:01.56
telegram=|6644496A4425155518377202742215496A0116370050052F2F_0C1355000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+34
MyTapWater 15227402 2023-01-12 22:02.16
telegram=|6644496A4425155518377202742215496A0116380050052F2F_0C1355000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+54
MyTapWater 15227402 2023-01-12 22:02.36
telegram=|6644496A4425155518377202742215496A0116390050052F2F_0C1355000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+74
MyTapWater 15227402 2023-01-12 22:02.56

Warm Water Meter:
telegram=|6644496A8753155518377261808824496A0106300050052F2F_0C1357000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+5
MyTapWater 24888061 2023-01-12 21:59.56

telegram=|6644496A8753155518377261808824496A0106310050052F2F_0C1357000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+25
MyTapWater 24888061 2023-01-12 22:00.16
telegram=|6644496A8753155518377261808824496A0106320050052F2F_0C1357000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+45
MyTapWater 24888061 2023-01-12 22:00.36
telegram=|6644496A8753155518377261808824496A0106330050052F2F_0C1357000000026CEC2182046CE1218C0413000000808D0493132C33FE00008000008000008000008000008000008000008000008000008000008000008000008000008000008002FD1700002F2F|+65
MyTapWater 24888061 2023-01-12 22:00.56

BIBOLV, 2023-01-19:
@weetmuts, should this be added to minomess driver?

weetmuts, 2023-01-19:
@BIBOLV Yep, do you want to do it?

BIBOLV, 2023-01-19:
Thank you for report, detection added. Closing issue.

closed 2023-01-19 by BIBOLV