wmbusmeters/wmbusmeters #811: Watertech WTT

Fiorello8, 2023-01-18:
Hello !

Thanks for your fantastic software!

Unfortunately I have a problem with the detection of  the frames from one type of smart meter

when I try to decode I receive the following error 

(meter) contatore: meter detection could not find driver for id: 02105681 mfct: (WTT) Watertech, Italy (0x5e94) type: Water meter (0x07) ver: 0x59
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 02105681 mfct: (WTT) Watertech, Italy (0x5e94) type: Water meter (0x07) ver: 0x59
(meter) newly created meter (contatore 02105681 auto) did not handle telegram!

If I use the website https://wmbusmeters.org/ (see the photo)  I can receive all the information except one (no important for me)
I would like to read this frame also via script.. Is it possible ?

Can you help me ?

[image]

BIBOLV, 2023-01-18:
Hi,
Can you post few telegrams here?
Also, yuo will need to elaborate what you mean by reading frame with as script.

Fiorello8, 2023-01-18:
Hi

for the telegrams see below
the password is always e7ac09ce9e902d3aaf23eb592aa01973

I would like to use wmbusmeters via command line but with this type of
smart meter I have not a good result...

I am using

*wmbusmeters: 1.9.0-43-g0125e520125e52da723b804375e8b5075763e272897489c*

If I use the web application
*https://wmbusmeters.org/analyze/3144945e3839100259078c20007a01002025a354a553e521bd56c37ceeac2297a8aa99adcf9f064310008ec7224459641b26:auto:e7ac09ce9e902d3aaf23eb592aa01973
<https://wmbusmeters.org/analyze/3144945e3839100259078c20007a01002025a354a553e521bd56c37ceeac2297a8aa99adcf9f064310008ec7224459641b26:auto:e7ac09ce9e902d3aaf23eb592aa01973>*
I have the correct result (i am not interested in a last value in red)

3144945e6456120259078c20687a69002025116f495887e0df8b153a9dc079344f63677baf8f1f501bda3943e92ce7ba7a6c
3144945e0958120259078c20997a9a00202548aa599b90a35d4783bdf20a4a0d87f67a6e44dfd592bd611c6b27433f9bbc6a
3144945e4846100259078c20477a48002025000a4beb17804e55cf71f159bdf45c92b2e0c9126e6337a7ef0508b91e8e834e
3144945e4646100259078c206f7a70002025bef3d1f32deff738b350d07ac5bcbac5e5ab80cf87bcea5c17f09d8408dd934b
3144945e6244100259078c20dd7ade002025c81c92f79d84cb4f313bb85ea202150ff582d878545f45e5e2e060cda7823fae
3144945e1457120259078c20497a4a002025937663f9c3b3406d646f201be05828b3a4b06d30c1f5d26bd4e7bd95cb0d1bf1
3144945e7226120259078c20bc7abd002025b3aff2ab40a04a811c137526bdc26422bc56505ca746d95d325764e7de27d544
3144945e1128100259078c20ea7aeb00202515902389bf64dd9a82761cf6e0766be13ed53955e43d09b0b83c4cc4beeac1d6
3144945e5959100259078c20f47af5002025f78a262be0e94ed6d8135652fda4aed11e206c58363065eb5793b57a8b5e734d
3144945e2755120259078c206d7a6e002025a26d3bf7f947ddb16f131953730c3bdaec887cb8ec16399b5181f28266bd5cbc
3144945e5259100259078c20197a1a002025945973d743b5ac926cc7a5f179271e649b9b14f440c751defac303cf59d7bbbe
3144945e0456120259078c20267a27002025933a9dca0c21ee62f1ae34c33e981d68da67b876d18a3f087595358481acafab
3144945e2656100259078c20fc7afd002025d76e1b2b0746cdc7dbb5babde5b6a19a67a8f76e4dd58d10295163018ae73098
3144945e2728100259078c204a7a4b00202501438d4bd94487f393eee349e936f1fa8b2f88e807ad324336d0a5a18eb9170a
3144945e4058120259078c20277a28002025c6371303deff381b2f7dc6ae2b8eca72c80eebccff8accf0686620e86d331bec
3144945e3949100259078c20097a0a0020255e943f60bcaa84fcbe7444948b7caf7bafaf916770c126ccdfe9f13f9e70c620
3144945e6244120259078c201e7a1f0020257d6d2ec67ac415d8e25a173e2e934228a5362791a708f495cb00a5454c68a438
3144945e6045120259078c20c47ac50020259a4b0db9041558c755b3a32a010db448aa5b860cac7eedc75ea1ed78d81cf1ef
3144945e8148100259078c201a7a1b002025774e165dab59e86459b026a8fefd7bd1b642672c1899ef2c35574cccdf2d8a6d
3144945e6045100259078c20687a6900202554523444854c4a8f191463dd5c14d2708ec6c03ead3be9c52d71d4d1db5c80ff
3144945e3156120259078c20d27ad30020257765d2524eb74a02a6c71b049f24a211f5b99fb410ebdee9d7dd1fbec81ccc0b
3144945e1344100259078c20ba7abb002025c7027634afd2e7d340c0f43668af4eba04411a1797dbf835324f9236868a845c
3144945e0844100259078c20967a97002025c87300138568b7a2edaa3611f2b4e1213bf187b4d11eb5f3ea686882c7c00549
3144945e1856100259078c20817a82002025c5f788130dd0d272e996a61534cda41acbd7094c2d4a51677563efeeca4e9775
3144945e5045100259078c20287a290020253778b32c9b7be53a505fc9ed979ee4c23139e9ca8703549255d502cd64ba6f9c
3144945e2544120259078c20807a810020257fb23faac75597fd0594b10e09dfa665439488ce1fcc3a4e521d7ada69926d37
3144945e5526120259078c20487a49002025cd25faac1f398bfb87212072f648d418a2a69d7eebdca57b0440e069ba75bcda
3144945e8227100259078c206d7a6e00202575aeedd6615acb05038f03317ccf8cbc4b5fa4c85860fa3d94f26e75148579b8
3144945e0442120259078c20e97aea002025a27ca6b85b0573f5ef2c5ba5d27d804638d5d6b86a526e737747d25e7b78d74d
3144945e4256100259078c20607a610020254993a767d5eeac01aae35d05942754d0043a888692ef7eb85035044c766a7851
3144945e1145120259078c20ed7aee002025b9cc9bcf0d99e446c0bdd42e0f83e1b80c56d144ae752e38747fbe9e7cf63802
3144945e5858120259078c20e57ae60020257ab8ca47086de4fcbf5607ea6cb795c857ccab637bbf26b82cb2c30e0afe2906
3144945e9842120259078c20167a170020259862aba7116116d945d4a3e3fbfb295f7248d4c08cea309260ee79b13742645b
3144945e0657100259078c20f37af4002025f67a677ff6445051082e8468d08278d48f84c2b309a8522ef13a22c13d1c1878
3144945e1043120259078c20427a430020257d116f00102824ed3b07af84c54b7e74bfbc3546e88205c681393647faea0cc8
3144945e3944120259078c20b27ab300202573d7370a22a170399a431137d02067079e527a3a0d1ea0c7dc3b9da666228d76
3144945e0043120259078c20857a860020250aa24d208c079fa4d64a7787db3146cf8acb00797c11351654a2b6fba8a88b25
3144945e6146100259078c20027a030020252a837f2e3e85d541bb3581bcfc105eccfac9c82c3b7d7f5e5b63f5a7c673b156
3144945e1258120259078c20d77ad8002025e9185ca13bc3f08310c96521406a466ad70c639b15c3658ae500fc571f84d611
3144945e0956100259078c20877a880020257f7375880b775d3698ef1696425f2fc401fa8142b0d84aa17a0f79c6418842e1
3144945e9026120259078c20207a2100202545ab24e5440824062e152f4b2c3ad759849a287fd1fed41bae24aa8d96378b71
3144945e9158100259078c20d07ad10020254ea1c413f928d8f4a5c4313c2d7a2d8cc3016bcac3055b7fee5872bc050a7058
3144945e1644120259078c20487a490020251af9392045728d75fb0bb84078a40e4c8a28fdcb60b7a7ef0f8c6d6a8f178430
3144945e9745100259078c20c17ac2002025c8902192bb2a83f60b3aab6d09d08e494973fec6f101ad9dcbfe5cdcf90ae2a8
3144945e2444100259078c200c7a0d002025ab7ecc8d88efc60d7e6824f12c50e4a5b78e407553e4ee4aab7adef8cc4f629d
3144945e6555120259078c20127a1300202507952b4a9d37149925dd49abeda0fa9539614afd09041e43b24e224d10262e2c
3144945e3558120259078c20527a5300202529a30416338cf5c4881b56e13837a529a2acfddbf5db601d768c45c8b4427f1c
[... cut ...]
3144945e1060100259078c20537a54002025886cebf3f6dcb71922c34f3c3c08ff36b4ee06e3546b2c21081167087e7ab61c
3144945e7858120259078c20b27ab300202557dc9f540741e370cca2164c2f055ef0475bbc2fec14a10b096c1d6be96e6c94
3144945e3839100259078c20007a01002025a354a553e521bd56c37ceeac2297a8aa99adcf9f064310008ec7224459641b26
3144945e2159100259078c20257a26002025315b606ff4c7fae758c3eb84d3763349b52d39f623d69533d32e1879af8a0d3f
3144945e8156100259078c20c77ac800202557b9ba435c97e3985497dab5365e2cf173bb5f8afd79c17733388932194b602b

BIBOLV, 2023-01-18:
Ok, please update to latest version and try then to see if everything works for you as expected.

weetmuts, 2023-01-18:
I just added a watertech driver. It seems like watertech is a different company from those that produce waterstarm meters therefore the mft specific status bits are probably different, so we need a different driver.

Please test!

Fiorello8, 2023-01-18:
The web application works fine !!!
How can I update the application in order to use the tool via command line ?

weetmuts, 2023-01-18:
How did you install wmbusmeters in the first place?

Fiorello8, 2023-01-18:
I have to update the installation of wmbusmeters on our linux (Debian)
machine ?

I tried to:

- download the code from github (wmbusmeters-master)
- copy the folder with the code on my Debain Virtual Machine
- follow the instruction noted at:
https://snapcraft.io/install/wmbusmeters/debian
- execute the command line "sudo make install" ... but I obtain the
following error

Building _
./scripts/generate_authors.sh build/authors.h
make: ./scripts/generate_authors.sh: Permission denied
make: *** [Makefile:241: build/authors.h] Error 127

if I execute wmbusmeters --version

I have as result

wmbusmeters: 1.9.0-43-g0125e52
0125e52da723b804375e8b5075763e272897489c

What I have to do to update the software ?

thanks in advance

Francesco

BIBOLV, 2023-01-18:
I do not recommend to use snap version for scripts, better to use standalone installation. But if you would explain what you want to achieve, we could try to help with advice how to achieve it also withing snap version.

Fiorello8, 2023-01-18:
Hi

I would like to use the command line with the new driver

wmbusmeters --format=json
1e44945e5243120259078c20a57aa6002025deb6f89c5049cd6da2b6f94e67932c917d50988ca6552abc748666fc4a6cae43
contatore1 waterstarm 02105722 E7AC09CE9E902D3AAF23EB592AA01900

but i am not able to upload the version.. this is my problem

I am using Debian... standalone version would be perfect for my purpose ...
Can you tell me how to download ?

Thanks

BIBOLV, 2023-01-18:
All the documentation is available [here](https://github.com/weetmuts/wmbusmeters/blob/master/README.md). You would need to install all necessary tools, clone repo and compile/install.

closed 2023-04-20 by github-actions[bot]