wmbusmeters/wmbusmeters #813: Doesnt receive telegrams.

Benleni17, 2023-01-19:
### Type of request

Question

### HAOS version

2023.1.5

### wmbusmeters version

1.11.0

### Your message goes here

Hello Guys,

iam new and in need some help.

I have a Kemstrup Multical 21 watermeter an i want to collect some datas.

I have installed the Wmbusmeters Addon and start it with the IMST 871a (im871a) Dongle.

The Protokoll shows:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:13:05] INFO: Syncing wmbusmeters configuration ...
[16:13:05] INFO: Registering meters ...
[16:13:05] INFO: Generating MQTT configuration ... 
[16:13:06] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[16:13:07] INFO: Running wmbusmeters ...
Started auto im871a[00103512] on /dev/ttyUSB0 listening on t1
No meters configured. Printing id:s of all telegrams heard!

Iam surprised, becouse the Dongle doesnt receive telegrams.

What can i do? 

Greets Benedikt

weetmuts, 2023-01-19:
The Multical21 defaults to C1 mode. Change T1 to C1.

Benleni17, 2023-01-19:
hello,

thanks for the fast answer.

I had change the defaults under /config/wmbusmeters/etc/wmbusmeters.conf to c1, save and restart. 

loglevel=normal
**device=auto:c1**
donotprobe=/dev/ttyAMA0
logtelegrams=false
format=json
logfile=/dev/stdout
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"

But after the restart, the protokoll shows:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:41:54] INFO: Syncing wmbusmeters configuration ...
[16:41:54] INFO: Registering meters ...
[16:41:54] INFO: Generating MQTT configuration ... 
[16:41:55] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[16:41:56] INFO: Running wmbusmeters ...
Started auto im871a[00103512] on /dev/ttyUSB0 listening on t1

and at the wmbusmeters.conf and the value changes back to t1?

What can i do?

BIBOLV, 2023-01-19:
You need to change config trough config interface/tab in HA, if you will change it directly in file it will be overwritten upon restart.

Benleni17, 2023-01-19:
Ok thanks. I changed the value in de HA Card.

Now the dongle isnt detected. Is the dongle compatibel with c1?

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[19:41:30] INFO: Syncing wmbusmeters configuration ...
[19:41:30] INFO: Registering meters ...
[19:41:30] INFO: Generating MQTT configuration ... 
[19:41:31] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[19:41:33] INFO: Running wmbusmeters ...
No such log level: "normaldevice=auto:c1donotprobe=/dev/ttyAMA0logtelegrams=falseformat=jsonlogfile=/dev/stdoutshell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON""
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!

Supervisor:
23-01-19 19:40:49 INFO (MainThread) [supervisor.addons] Add-on 'cada3e3b_wmbusmeters' successfully installed
23-01-19 19:41:19 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/007 - None
23-01-19 19:41:19 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/gpiochip2 - None
23-01-19 19:41:19 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyUSB0 - /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_027593C3-if00-port0
23-01-19 19:41:29 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on cada3e3b/aarch64-addon-wmbusmeters with version 1.11.0

BIBOLV, 2023-01-19:
Please double check config, looks like you lost newline characters - `\n`, so now config is interpreted as one line and it is not recognized.

Benleni17, 2023-01-20:
Hello and thanks for your help,

now it works. And the dongle recive telegrams:

My waterdealer had givven the KEM file with my personaly password. How its goinig on? Iam new and i  doesnt works with python. It is possible to make the code for the watermeter with a easier methode?

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[15:57:00] INFO: Syncing wmbusmeters configuration ...
[15:57:00] INFO: Registering meters ...
[15:57:00] INFO: Adding meter-0001 ...
[15:57:00] INFO: Generating MQTT configuration ... 
[15:57:01] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[15:57:03] INFO: Running wmbusmeters ...
Not a valid meter key "00000000000000000000000000000000]"
Started auto im871a[00103512] on /dev/ttyUSB0 listening on c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 76406477
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: im871a[00103512]
                  rssi: -69 dBm
                driver: multical21
telegram=|_23442D2C776440761B168D2011617022239C992EB089461DD69F76B8E4D4FA78FB98ED1A|+2
Received telegram from: 76406477
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: im871a[00103512]
                  rssi: -111 dBm
                driver: multical21
telegram=|_23442D2C776440761B168D20126270222329A57F8DAB25C5287243CC7906308A87078661|+19

weetmuts, 2023-01-20:
First rename your received file to zip. Then unzip it. Then inside the archive
you find a .kem file with a name consisting of a lot of hex chars.

Install java (version 11 or later) then:

java utils/XMLExtract.java YourFilePassord YourKemFile

Mow it will print some xml and after DEK you have your key.

Benleni17, 2023-01-20:
I have 2 data types kem and kem2. What kind of datatyp ist correct?

weetmuts, 2023-01-20:
try them. i do not know. :-)

Benleni17, 2023-01-20:
Sry Man. I have installed JAVA 17 an now? How does it work? do i Need another programme to works the XML ?

BIBOLV, 2023-01-20:
Go to wmbusmeters git clone directory in cmd and run command that was provided previously.

Benleni17, 2023-01-20:
Sry it doesent works. I have copy hte wmbusmeters folder from the config to my Desktop. Then I go with cmd to the wbusmeters and try to trigger the comand. Here the log from cmd:

C:\Users\energ\Desktop>cd wmbusmeters

C:\Users\energ\Desktop\wmbusmeters>java utils/XMLExtract.java Kohlrieser kem
Fehler: Hauptklasse utils.XMLExtract.java konnte nicht gefunden oder geladen werden
Ursache: java.lang.ClassNotFoundException: utils.XMLExtract.java

This is the contents from the KEM File:

<?xml version="1.0" encoding="utf-8"?><EncryptedData Type="http://www.w3.org/2001/04/xmlenc#aes128-cbc" xmlns="http://www.w3.org/2001/04/xmlenc#"><EncryptionMethod Algorithm="http://www.w3.org/2001/04/xmlenc#aes128-cbc" /><CipherData><CipherValue>q3tzl4HfA3uhBX71MmJfEgT9icP70lwE9Uy8jW1B6NXnsum2Z1yPtN3318xEy3GfDx8ylNGKjAlkKDIxA9FlXU1IZH2blMX7NpfAA5YxKyuqDT9A7YY+bbvURzhouD4RlCY52WhykRoMPdSXXCD4Jx3+fGUU9ns1KLN0yosY6Fe4Pt2rt1Da3BeJVwVvYsRCko/ao8amfbflMlqoz5+qfNMCB9KeGdKHcGVWp+PPJJ0Cw5s8itbycwtrQDAUXYomx4DJHvoPzsGwORlLqsRH361yNH344mTQFSU3bbzJnwLg9wcPRCqgCTMDFWNMHqbUse3rcx4sPWQQo1nAUMXQqsB+II22Ht5rCfj7a6QvWh0HDZZXvkG0EmoRfCOMZFsjFEL/h1vK6SmEhA7VJxFxQSlzPpjY6/mwQazm1O3lQboGiln3866+Kn7gDVvwQva2BzBc2nWHw0eyryG1p7RynxQbAC1/hhPx64FHk1p6K04cCcxXxUUsrXZ/1zkrCN3v</CipherValue></CipherData></EncryptedData>

BIBOLV, 2023-01-20:
You probably installed jre, but you need jdk. You can check that by rinning javac and if command is not recognized then you do not have jdk. Check [readme](https://github.com/weetmuts/wmbusmeters#additional-tools) how to compile class just in case you need it.

weetmuts, 2023-01-21:
There is a new script pushed that you also can try:
./utils/kem-extract.sh YourPassword YourKemFile

For this one, you do not have to unzip anything first.

Benleni17, 2023-01-21:
Hello Guys,

thanks for your help. JAVA jdk and cmd works. I have the key. Now the protocol from wmbusmeters shows:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[10:05:20] INFO: Syncing wmbusmeters configuration ...
[10:05:20] INFO: Registering meters ...
[10:05:20] INFO: Adding meter-0001 ...
[10:05:20] INFO: Generating MQTT configuration ... 
[10:05:21] INFO: Broker core-mosquitto will be used.
MQTT Discovery cleanup...
[10:05:23] INFO: Running wmbusmeters ...
Started auto im871a[00103512] on /dev/ttyUSB1 listening on c1
telegram=|23442D2C776440761B168D200363B4222301CF79EDA88A2D_0000E44001004C2B01000408|+9
telegram=|23442D2C776440761B168D200470B4222301CF79EDA88A2D_0000E44001004C2B01000408|+25
telegram=|23442D2C776440761B168D200773B4222301CF79EDA88A2D_0000E44001004C2B01000408|+75
telegram=|23442D2C776440761B168D200880B4222301CF79EDA88A2D_0000E44001004C2B01000408|+91

I have write in the configuration.yaml:

mqtt:
  sensor:
    - state_topic: "wmbusmeters/MainWater"
      json_attributes_topic: "wmbusmeters/MainWater"
      unit_of_measurement: m³
      value_template: "{{ value_json.total_m3 }}"
      name: Water usage
      icon: "mdi:gauge"
      state_class: total_increasing
      device_class: water

and i see a new sensor, but he says unknown.

I havent worked with mqqt mosquitto broker. Do I also have to adjust the settings here so that the sensor works?

BIBOLV, 2023-01-21:
Did you add meter to ha-addon config?

Benleni17, 2023-01-21:
Yes, i has add the meter in configuration.yaml. The MQTT core-mosquitto device all datas, but it hasnt give the datats to the sensors.

Here ist the Protokoll from MQTT:

MainWater Debug-Informationen

Entitäten
MainWater status burst (binary_sensor.mainwater_status_burst)
MQTT discovery data:
Topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_burst/config
Payload
device:
  identifiers:
    - wmbusmeters_76406477
  manufacturer: Kamstrup
  model: multical21
  name: MainWater
  sw_version: '76406477'
enabled_by_default: true
device_class: problem
name: MainWater status burst
state_topic: wmbusmeters/MainWater
unique_id: wmbusmeters_76406477_status_burst
value_template: '{{ ''BURST'' in value_json.current_status }}'
payload_on: 'True'
payload_off: 'False'
platform: mqtt
Subscribed topics:
wmbusmeters/MainWater
10 zuletzt empfangene Nachricht(en)
Received 15:29:47
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.184,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:29:47Z","device":"im871a[00103512]","rssi_dbm":-76}\
Received 15:30:03
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.185,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:30:03Z","device":"im871a[00103512]","rssi_dbm":-57}\
Received 15:30:19
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.185,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:30:19Z","device":"im871a[00103512]","rssi_dbm":0}\
Received 15:30:35
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.186,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:30:35Z","device":"im871a[00103512]","rssi_dbm":-89}\
Received 15:30:51
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.186,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:30:51Z","device":"im871a[00103512]","rssi_dbm":-88}\
Received 15:31:07
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.186,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:31:07Z","device":"im871a[00103512]","rssi_dbm":-62}\
Received 15:31:23
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.186,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:31:23Z","device":"im871a[00103512]","rssi_dbm":-18}\
Received 15:31:39
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.187,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:31:39Z","device":"im871a[00103512]","rssi_dbm":-115}\
Received 15:31:55
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.187,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:31:55Z","device":"im871a[00103512]","rssi_dbm":-90}\
Received 15:32:11
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.187,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:32:11Z","device":"im871a[00103512]","rssi_dbm":-81}\
Transmitted messages:
MainWater status dry (binary_sensor.mainwater_status_dry)
MQTT discovery data:
Topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_dry/config
Payload
device:
  identifiers:
    - wmbusmeters_76406477
  manufacturer: Kamstrup
  model: multical21
[... 243 lines cut ...]
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.186,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:30:35Z","device":"im871a[00103512]","rssi_dbm":-89}\
Received 15:30:51
QoS: 0
[... cut ...] 3-01-21T14:31:55Z","device":"im871a[00103512]","rssi_dbm":-90}\
Received 15:32:11
QoS: 0
Payload: {"media":"cold water","meter":"multical21","name":"MainWater","id":"76406477","status":"OK","total_m3":82.187,"target_m3":76.62,"flow_temperature_c":4,"external_temperature_c":8,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-01-21T14:32:11Z","device":"im871a[00103512]","rssi_dbm":-81}\
Transmitted messages:
Auslöser
Keine Auslöser

But the

BIBOLV, 2023-01-21:
For multical21 there is a template, so you should be able to use meter without adding anything to HA config manually. It should show up in HA automatically.

Benleni17, 2023-01-24:
Good morning, 
after 24 hours, wmbusmeter stops recording.  After the reset, the addon does not receive any data from the water meter.  How can this be changed?

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[07:18:41] INFO: Syncing wmbusmeters configuration ...
[07:18:41] INFO: Registering meters ...
[07:18:41] INFO: Adding meter-0001 ...
[07:18:41] INFO: Generating MQTT configuration ... 
[07:18:42] INFO: Broker core-mosquitto will be used.
MQTT Discovery ...
 Adding meter: MainWater ...
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_external_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_flow_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_max_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_rssi_dbm/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_burst/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_dry/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_leak/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_reversed/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_target_m3/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_total_m3/config
MQTT Discovery cleanup...
[07:18:48] INFO: Running wmbusmeters ...
Started auto im871a[00103512] on /dev/ttyUSB0 listening on c1
(wmbus) resetting /dev/ttyUSB0:im871a[00103512]
(wmbus) reset completed /dev/ttyUSB0:im871a[00103512]

BIBOLV, 2023-01-24:
Did you try to set reset more frequent, like 12 or 5 hours, does it still stops receiving after 24? Is in morning, exact after 24h or randomly?
How do you restore - restart of addon or whole HA?
Enable debug logs and attach here the time when it stops receiving telegrams.

Benleni17, 2023-01-25:
Hello Guys,

i have now deleted MQTT and WMBUSMETERS together with the folder because I did not see the sensor values ​​in MQTT. The configuration in WMBUSMETERS looks like this:

Das sind die ursprünglichen Werte vor der deinstallation:

data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: >-
loglevel=normal\ndevice=auto:c1\ndonotprobe=/dev/ttyAMA0\nlogtelegrams=false\nformat=json\nlogfile=/dev/stdout\nshell=/wmbusmeters/mosquitto_pub.sh"wmbusmeters/$METER_NAME"
  "$METER_JSON"
meters:
  - |-
    name=MainWater
    driver=multical21
    id=76406477
    key=AD2F041B407A23917A8D810217AA304B
mqtt: {}

I do not receive any telegrams with these settings!

I receive these telegrams without a key and he shows me another id. The old ID was 76406477 and according to received telegrams without a key 57390302

[16:23:29] INFO: Running wmbusmeters ...
Started auto im871a[00103512] on /dev/ttyUSB0 listening on c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 57390302
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b

Now I have the problem that WMBUSMETERS says the code is wrong with the new ID:

[16:23:29] INFO: Running wmbusmeters ...
Started auto im871a[00103512] on /dev/ttyUSB0 listening on c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 57390302
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: im871a[00103512]
                  rssi: -50 dBm
                driver: multical21

BIBOLV, 2023-01-26:
Ok, closing issue for now.

closed 2023-01-26 by BIBOLV