wmbusmeters/wmbusmeters #816: Apator AT-WMBUS-16-2 always decodes 0

fgvalles, 2023-01-20:
### Type of request

Existing device improvement

### Meter description

Apator AT-WMBUS-16-2 (driver apator162)
From 2023/01/01 wmbusmeters always decodes 0 in all my water counters.

### Logged telegrams from meter in decrypted format

```shell
telegram=|6E4401063309620305077A9B0060852F2F_0F523B4C9608020043210283000FA7B08DD5018CE927109B5200007B015B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFF21EC|+0 -> decoded value = 21.147
telegram=|6E4401063309620305077A680060852F2F_0F543C6C9608020043220283000FA7B08DD5018CE92710375300007B015B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFF36AA|+0 -> decoded value = 21.303 telegram=|6E4401063309620305077A340060852F2F_0F533D8C9608020043230283000FA7B08DD5018CE92710EF5300007B015B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFF7F3B|+0 -> decoded value = 21.487 telegram=|6E4401063309620305077A000060852F2F_0F543EAC9608020043240283000FA7B08DD5018CE927105D5400007B015B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFF1A07|+0 -> decoded value = 21.597 telegram=|6E4401063309620305077ACC0060852F2F_0F533FCC9608020043250283000FA7B08DD5018CE92710015500007B015B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFF788B|+0 -> decoded value = 21.761 telegram=|6E4401063309620305077AF40060852F2F_0F7A21E19708020043260283000FA7B08DD5018CE92710795600007B01375600005B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFFB631|+0 -> decoded value = 0 telegram=|6E4401063309620305077A150060852F2F_0F8022219708020043270283000FA7B08DD5018CE92710D45600007B01375600005B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFFA5C3|+0 -> decoded value = 0 telegram=|6E4401063309620305077AAE0060852F2F_0F8024619708020043290283000FA7B08DD5018CE92710E05700007B01375600005B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFF4619|+0 -> decoded value = 0 telegram=|6E4401063309620305077A7A0060852F2F_0F7B258197080200432A0283000FA7B08DD5018CE92710215800007B01375600005B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFFA18A|+0 -> decoded value = 0 telegram=|6E4401063309620305077A470060852F2F_0F8026A197080200432B0283000FA7B08DD5018CE92710AF5800007B01375600005B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFFCEEB|+0 -> decoded value = 0 telegram=|6E4401063309620305077A130060852F2F_0F8027C197080200432C0283000FA7B08DD5018CE92710725900007B01375600005B45000067380000262D0000512100008819000096110000840500004A0000004A0000004A0000004A000000A0F08F3404FFFFFFFFFFFFFFFFFFFF695F|+0 -> decoded value = 0
```

weetmuts, 2023-01-20:
Upgrade to the latest version of wmbusmeters where this is fixed.

fgvalles, 2023-01-20:
Solved. Thank you very much.

closed 2023-01-20 by fgvalles