wmbusmeters/wmbusmeters #820: Telegram Decryption for Engelmann Heat cost allocators HCA E2 failed

GitKunde, 2023-01-21:
Hi,

I'm using heat cost allocators "HCA E2" from Engelmann Sensors (Manufacturer ID = 14c5). I can receive telegrams (see example below) but I'm struggling to decrypt the data. I'm pretty sure to use the right key (standard key). So I tried to analyze the telegram at you nice web page (https://wmbusmeters.org) and got same result (failed decryption. Wrong key?).
Are there any hints to find my mistake? Or is there anything special to regard with these type of sensors?

Thanks for your great work.

Example: cf44c5145041219029087a4800c025b1ca6bee1bba1e4b791ae0a22869dc66b523192799bc8a8a5a866a87a76450bd978f9a84db9aea4ac67c82f708784e16804eb783a38bc481653f002eed922a21c84ed8e2dc228af45b49822abb382ba798db0a9ea0eb40b5382b682a8e40466a5224137555860d35b56bfa726b17fcf6cb1f2f00ff174974f007bf440a27b6811cd20428c2f039248a8e2abbfec3c3908f580e36519ef1b66ac63f59ac8c5fbe2966ac18ad8b953ff51679db9f78e06c2e0b1d7931c8b6e0c219808adafe4532e816

AES-128 Standard-Key:
00112233445566778899aabbccddeeff

IV:
c5145041219029084848484848484848

JensHoRi, 2023-01-27:
I have ordered Engelmann 2 years ago, no one of them I was able to get running with different software. 

But I got a device specific key for everyone of them.

I ordered somewhere else Lorenz then. Got individual keys, everything worked out of the box after weemut (THANKS) added some additional changes for further values :-)

weetmuts, 2023-01-27:
How did you receive these telegrams? It is a bit odd that when wmbusmeters tries to decrypt the telegram
the number of encrypted blocks  add up to 192 which leaves 2 unencrypted bytes at the end.
This is sort of allowed, but usually there is more bytes at the end, with dif+vif+data codes....

Could the telegram have some dll-crc bytes not yet removed?

GitKunde, 2023-01-30:
Thanks for your comments!
In the meantime I found out the issue. I was sure with the AES-Key because it worked with Engelmann-readout software. But this software is using automatically the Engelmann-master-Key if the configured Key (in my case: 0x00112233...) is not working. But this master key seems not to be public and can not use with "self-made"-software. Or is there anybody who can help me with that. Otherwise it would be necessary to reprogram a new private AES-Key in each devise. Might be that only professional services are able to do this.
I received the telegrams just by using an AMB8465-M - M-Bus USB-Stick (+USBtoSerial driver) and read it with python. I just cutted the firs 2 Bytes (FF 03).

weetmuts, 2023-01-30:
Ah, then you are not speaking the proper amb8465-m protocol and you get the crc bytes at the end as well.
Why do you not let wmbusmeters talk to the amb8465-m instead?

Skysurfer-14, 2023-03-20:
I'm planning to buy the same heat cost allocators "HCA E2" from Engelmann. Is there still a problem to receive the protokol? The HCA e2 device will be delivered with the standard Engelmann AES master key. I'm shure that this key will be different from the key above in the first post. Is the Englemann AES master key known here?

Skysurfer-14, 2023-03-20:
Is it possible for me, to change the AES key of the device?

GitKunde, 2023-03-20:
My understanding is, that it is possible. But you need an opto-head (infrared -reader and -writer) and the device monitor software from Engelmann. For my purpose it is to expensive. If you order new heat cost allocators, you should ask the provider/seller, if he can reprogram the AES-Key for you. Might be the do the service for you.
Let me know, if you find a good way. I’m still interested in it.

Skysurfer-14, 2023-03-21:
@GitKunde , can you contact me bayar91778@oniecan.com ?

weetmuts, 2023-03-30:
Very nice! Make a copy of the driver_eursii.cc and to driver_hcae2.cc and change all eurisii references inside to hcae2. Update the addDetection to match your telegram.

Now build wmbusmeters with only this driver (speeds things up)
make DRIVER=hcae2

then do
./build/wmbusmeters --analyze CE44C5149126412331087A6B0BC0252F2F_046D1829FE23036E000000426C0000436E00000082016CBF2C83016E000000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000082096E0000C2096E0000820A6E0000C20A6E0000820B6E0000C20B6E0000820C6E0000C20C6E0000820D6E0000C20D6E0000820E6E0000C20E6E0000820F6E0000C20F6E0000317F08346D302EFB232F2F2F2F2F

And start tweaking the internal rules of the driver until you catch all fields. I.e. they become green.
Add this test telegram at the end of the source, so when you are done, you can do:
make
make test

A useful shortcut:
addOptionalCommonFields("meter_datetime");

Please have a go! Let me know if you have any questions. You can do an early pull request and I will help out!

weetmuts, 2023-03-30:
Great work! You can see that analyze does not find an auto-driver, ie a hardcoded lookup from 
the mfct type version  numbers which for your telegram are: EFE 08 31 so
remove the addDection lines and add the single line:
di.addDetection(MANUFACTURER_EFE,  0x08,  0x31);

Try again and auto will point to hcae2

Skysurfer-14, 2023-03-30:
Yes, it points now to hcae2. What´s next?

Skysurfer-14, 2023-05-02:
What can i do to make the Engelmann Heat Cost Allocator HCA e2 appears in my wmbusmeters Home Assistant Addon?

weetmuts, 2023-07-02:
Issue resolved in https://github.com/wmbusmeters/wmbusmeters/pull/956

weetmuts, 2023-07-02:
hcae2 will be part of release 1.14 which is release later today.

closed 2023-07-02 by weetmuts