wmbusmeters/wmbusmeters #825: qwater 5.5 support

FrankHdf, 2023-01-25:
### Type of request

New device support

### Meter description

Qwater 5.5
WME5-010T-6111 00V00
The driver is detected as unknown, m³ are shown as n ull

wmbusmeters --logtelegrams  auto:c1 watermeter qwater 51462865 NOKEY

Started auto cul on /dev/ttyUSB0 listening on c1
(meter) watermeter: meter detection did not match the selected driver qwater! correct driver is: unknown!
(meter) Not printing this warning again for id: 51462865 mfct: (QDS) Qundis, Germany (0x4493) type: Water meter (0x07) ver: 0x1a
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|49449344652846511A0778_0DFF5F35008266003506B2EA67201A617271A75EF279CE11F56A190D64725DD4C03B1849DF572B7F1C4704233FB70A5032D3EBD90D395FBB93046D2E0EF921|+948
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 14:47.16
telegram=|49449344652846511A0778_0DFF5F350082670035870553DB8A95B4616036C03C25634ABB18FC197BDDB9AD927DB3FFD90965D359270051DC963290472151EAC2961DBC1E046D390EF921|+1610
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 14:58.18
telegram=|49449344652846511A0778_0DFF5F3500826B0035CE204782CD7209486D5433E5268011C1CD938A8B799C5E06A84C0C1ACDD61F5DFBAB95F127667C23DF5F1DDCFCE64358046D1B0FF921|+3377
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 15:27.45
telegram=|49449344652846511A0778_0DFF5F3500826C003515D1B5A2331CF02222D797242B31C2FE8BCEF253F4360F7ED04449FEACD0712873A54E2B1E55DC8E48F42E3E2B5E10A1046D1E0FF921|+3590
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 15:31.18
telegram=|49449344652846511A0778_0DFF5F3500826D0035E0DA3A3B9C526EC2EACACE8FEA1F988F4CFBAC00F437AF8AAE34ECF70A349487CFACBF121D359122FFB17326E0CA04F1046D250FF921|+4033
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 15:38.41
telegram=|49449344652846511A0778_0DFF5F3500826F0035D2E5B447D21B53389FC88F840E487EBBCA255E4A0014500AB937F55DAD1E3518D728E716C82A366C4E7152FA910650FD046D3A0FF921|+5246
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 15:58.54
telegram=|49449344652846511A0778_0DFF5F3500827000357C2C83146B5254B36CA79389B82715FC3A1D4F052B2168BD9A1BDBA15F338FF705D7382937CB886B1993CECF7B5ACC67046D0110F921|+5463
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 16:02.31
telegram=|49449344652846511A0778_0DFF5F3500827300357E8A95949031056566ED88E4B8E35CE13D92DACC629AAAA686CC83421C04EA005A56615995DF365A7E53B5B3122C5471046D1B10F921|+7003
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 16:28.11
telegram=|49449344652846511A0778_0DFF5F350082760035867F3188129CEBE2260FF2DF6663894AE66AD6828290111938BE54F04298D90ED613A667445CCE0C342CB93ACCA94626046D2D10F921|+8095
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 16:46.23
telegram=|49449344652846511A0778_0DFF5F350082770035D64FB24F98F3383989D558842D0300056EA45908C71C6DCFC91A96E4BE44B1CB49C184FA5B54761D33D2969B273D54DF046D3410F921|+8531
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 16:53.39
telegram=|49449344652846511A0778_0DFF5F350082770035704B9E8395F7F9D6F827FF41412BACA7DFE417C3EE6E0EF3A6D8CC519DE7808D8F035E9D7785E73BC4D86298168EC181046D3810F921|+8759
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 16:57.27
telegram=|49449344652846511A0778_0DFF5F350082790035AF7DD014272A47D055EA24B5C4FE4F61F6DF5841BC11D2667BA1996F9C6A0AAA104806FECFFFFA1FE4EB4596C142C87D046D0511F921|+9295
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:06.23
telegram=|49449344652846511A0778_0DFF5F3500827A003514BAC5F55C8ACD2FF3980B3C9C011E85D38CACA849E8CA00ED19111F0BC7B97159F9CCC83E9D5CEE3FA64BA21849A905046D0E11F921|+9845
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:15.33
telegram=|49449344652846511A0778_0DFF5F3500827A003522D747E436F7379ACA79DE060EBCC64838310B121B81EFCE3B13EFC6B93D331EAB2ACC0661FAE44139A4D2F9AB2C2A42046D1011F921|+9958
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:17.26
telegram=|49449344652846511A0778_0DFF5F3500827B0035E640CAE67FD91901DCD65E8F8BC76E33EAA9C2F87785B0C633794ACFCFA18A96FD9E2D6EB58AC21C67F72C24268D6936046D1711F921|+10393
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:24.41
telegram=|49449344652846511A0778_0DFF5F3500827C00352FCDAACAC40C98BF0B5EE9BBE1832F66890AA2EDAD1EEA267485CEA8FCD5A2F5883AC0264C693641DB094208DA3B020B046D1F11F921|+10831
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:31.59
telegram=|49449344652846511A0778_0DFF5F3500827D0035F432712146D2452735CC389143043FC9C288808A2D320059B7485D1AA778288CD4498DE446CFBFBFABFDDB15B6E2C68E046D2811F921|+11380
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:41.08
telegram=|49449344652846511A0778_0DFF5F3500827E00358F0AB910CD1F37482B9FE9CF554234E01F2BDDC14195AB88076121F3DE066C78D06B059FAA63D79F347995C9BE62C182046D2A11F921|+11480
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:42.48
telegram=|49449344652846511A0778_0DFF5F3500827E003598928CD067E09AED81BDDB5D56CF4F62A3BE1F1661164C93C385C6B764F47675557568D7EB639649B380F8912946A5DB046D2B11F921|+11591
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:44.39
telegram=|49449344652846511A0778_0DFF5F3500827E003522A0C10275315AD0CEF2F8E4E77B26E5B228A678430621A3F5B415D47700DBE700805F84B46AFDB872324C40396D6EEA046D2D11F921|+11704
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:46.32
telegram=|49449344652846511A0778_0DFF5F3500827F00353DC27D674CC50A99F429CE814AEF5F1C29DF74B77CB2ECE7F7B52886CBE31E72B8ACC285D74449D8D0A3F6A714C6CD3C046D3511F921|+12142
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:53.50
telegram=|34449344652846511A077A80000820_04ED39330CE92101FD0C11046D3711F92102FD3CC2010DFF5F0C000885FF000006130705FFFC|+12269
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:55.57
telegram=|49449344652846511A0778_0DFF5F35008280003511802F4D9361553633AC50B758655B9505C972908621B7F58DECC8B99BBE8FDBD6DA92A3E5F8D2F1ABDBE6A6BD019FF2046D3A11F921|+12466
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 17:59.14
telegram=|49449344652846511A0778_0DFF5F35008281003576BCF5E6E1ED2A2066BF29869D9DA037952201650500BCCCF4AB835ABBA21C9ADEA7FB05D4AD4B97291BF282970126B1046D0312F921|+12790
watermeter	51462865	null m³	null m³	null	OK	2023-01-25 18:04.38

### Logged telegrams from meter in decrypted format

```shell
telegram=|49449344652846511A0778_0DFF5F35008266003506B2EA67201A617271A75EF279CE11F56A190D64725DD4C03B1849DF572B7F1C4704233FB70A5032D3EBD90D395FBB93046D2E0EF921|+948
telegram=|49449344652846511A0778_0DFF5F350082670035870553DB8A95B4616036C03C25634ABB18FC197BDDB9AD927DB3FFD90965D359270051DC963290472151EAC2961DBC1E046D390EF921|+1610
telegram=|49449344652846511A0778_0DFF5F3500826B0035CE204782CD7209486D5433E5268011C1CD938A8B799C5E06A84C0C1ACDD61F5DFBAB95F127667C23DF5F1DDCFCE64358046D1B0FF921|+3377
telegram=|49449344652846511A0778_0DFF5F3500826C003515D1B5A2331CF02222D797242B31C2FE8BCEF253F4360F7ED04449FEACD0712873A54E2B1E55DC8E48F42E3E2B5E10A1046D1E0FF921|+3590
telegram=|49449344652846511A0778_0DFF5F3500826D0035E0DA3A3B9C526EC2EACACE8FEA1F988F4CFBAC00F437AF8AAE34ECF70A349487CFACBF121D359122FFB17326E0CA04F1046D250FF921|+4033
telegram=|49449344652846511A0778_0DFF5F3500826F0035D2E5B447D21B53389FC88F840E487EBBCA255E4A0014500AB937F55DAD1E3518D728E716C82A366C4E7152FA910650FD046D3A0FF921|+5246
telegram=|49449344652846511A0778_0DFF5F3500827000357C2C83146B5254B36CA79389B82715FC3A1D4F052B2168BD9A1BDBA15F338FF705D7382937CB886B1993CECF7B5ACC67046D0110F921|+5463
telegram=|49449344652846511A0778_0DFF5F3500827300357E8A95949031056566ED88E4B8E35CE13D92DACC629AAAA686CC83421C04EA005A56615995DF365A7E53B5B3122C5471046D1B10F921|+7003
telegram=|49449344652846511A0778_0DFF5F350082760035867F3188129CEBE2260FF2DF6663894AE66AD6828290111938BE54F04298D90ED613A667445CCE0C342CB93ACCA94626046D2D10F921|+8095
telegram=|49449344652846511A0778_0DFF5F350082770035D64FB24F98F3383989D558842D0300056EA45908C71C6DCFC91A96E4BE44B1CB49C184FA5B54761D33D2969B273D54DF046D3410F921|+8531
telegram=|49449344652846511A0778_0DFF5F350082770035704B9E8395F7F9D6F827FF41412BACA7DFE417C3EE6E0EF3A6D8CC519DE7808D8F035E9D7785E73BC4D86298168EC181046D3810F921|+8759
telegram=|49449344652846511A0778_0DFF5F350082790035AF7DD014272A47D055EA24B5C4FE4F61F6DF5841BC11D2667BA1996F9C6A0AAA104806FECFFFFA1FE4EB4596C142C87D046D0511F921|+9295
telegram=|49449344652846511A0778_0DFF5F3500827A003514BAC5F55C8ACD2FF3980B3C9C011E85D38CACA849E8CA00ED19111F0BC7B97159F9CCC83E9D5CEE3FA64BA21849A905046D0E11F921|+9845
[... cut ...]
telegram=|49449344652846511A0778_0DFF5F3500827F00353DC27D674CC50A99F429CE814AEF5F1C29DF74B77CB2ECE7F7B52886CBE31E72B8ACC285D74449D8D0A3F6A714C6CD3C046D3511F921|+12142
telegram=|34449344652846511A077A80000820_04ED39330CE92101FD0C11046D3711F92102FD3CC2010DFF5F0C000885FF000006130705FFFC|+12269
telegram=|49449344652846511A0778_0DFF5F35008280003511802F4D9361553633AC50B758655B9505C972908621B7F58DECC8B99BBE8FDBD6DA92A3E5F8D2F1ABDBE6A6BD019FF2046D3A11F921|+12466
telegram=|49449344652846511A0778_0DFF5F35008281003576BCF5E6E1ED2A2066BF29869D9DA037952201650500BCCCF4AB835ABBA21C9ADEA7FB05D4AD4B97291BF282970126B1046D0312F921|+12790
```

weetmuts, 2023-01-27:
Sadly most of the telegram seems to be mfct specific. Take a look here:
https://wmbusmeters.org/analyze/49449344652846511A07780DFF5F35008266003506B2EA67201A617271A75EF279CE11F56A190D64725DD4C03B1849DF572B7F1C4704233FB70A5032D3EBD90D395FBB93046D2E0EF921

Then there is a shorter telegram with some meta-data about the meter.
https://wmbusmeters.org/analyze/34449344652846511A077A8000082004ED39330CE92101FD0C11046D3711F92102FD3CC2010DFF5F0C000885FF000006130705FFFC

Can you provide the list of telegrams in decrypted format again, but with your manual readings for the start end water consumption? You have to consume some water. Then we can see in the mfct specific data where the counter is.... hopefully. :-)

weetmuts, 2023-01-27:
Alas a lot of bytes change between each telegram, that would indicate that the telegrams are encrypted.....

morremeyer, 2023-02-02:
I think I have a related issue:

I do have 4 Qundis 5.5 water meters at home (2 warm, 2 cold). All of them are configured with the `qwater` driver.

I get data for all four meters, but for the two warm water meters, I get:

```
(meter) Warm Water Kitchen: meter detection did not match the selected driver qwater! correct driver is: unknown!
(meter) Not printing this warning again for id: 13313978 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x17
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```

```
(meter) Warm Water Shower: meter detection did not match the selected driver qwater! correct driver is: unknown!
(meter) Not printing this warning again for id: 13313976 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x17
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```

Example telegram from the `Warm Water Kitchen` meter: https://wmbusmeters.org/analyze/4944653278393113170678_0C13772802000DFF5F2F0001FB80803608C113FFFFDF2C93210200FF214028020048005C004000280058003B001F0014002A002E0033004100046D0C0BE222

Example from the `Cold Water Shower` meter (which does not print any warnings!): https://wmbusmeters.org/analyze/49446532571326131607780c13233908000dff5f2f0001fB807f2508c113ffffdf2c82180800ff218637080003012201ee00cc00cf004301dB009000BB007c00ac00Be00046d080Be222:qwater

Looking at the telegrams in their analyzed form, the only thing changing when switching between drivers is that the json field with the meter timestamp is either `device_date_time` when using `auto` or `meter_date_time` when using `qwater`.

Everything else looks the same.

I suppose the fields marked as `C?` are manufacturer specific fields of which we do not know the use/value?

FrankHdf, 2023-02-10:
Sorry for the late answer, I was on vacation.
I have a new list of telegrams now with the manual recordings of the water consumption.
Water consumption start: 1.249
Water consumption end: 1.270

Telegrams:
telegram=|49449344652846511A0778_0DFF5F3500822C0035FC3ACC70676E70C81FC78C318C3A13569D16F5F9E75FF778F09BA6BB1C8F8D5FADCDA61ADD1E1D3A4DB672218B84D4CD046D2307EA22|+145
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 07:36.33
telegram=|49449344652846511A0778_0DFF5F3500822C0035016AF3440AD6B7EE4A4EDCB41CAB0037A8AA8A7A91C50F1E6C4E3A9CFCAA861642EA96B4C72A33F8FA99BCC4C581FC56046D2507EA22|+258
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 07:38.26
telegram=|49449344652846511A0778_0DFF5F35008233003581BEB8A88BD865806271BF1700958E38B41F07114DC274CCDD5D31427FB24B0BE2FC3FB9351B7804DD35335DFA2C7CA8046D1E08EA22|+3433
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 08:31.21
telegram=|49449344652846511A0778_0DFF5F350082340035687ECDA1D91D4D8F316CBCDCC7256AB0DE3CAD8BAAC3B3E4F4DF40CE8587FE8648D2C8CE90820964BE07C14EFC52A5DD046D2608EA22|+3885
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 08:38.53
telegram=|49449344652846511A0778_0DFF5F35008234003568323FDD03A1AD828982425F535FEB0F59C057450DAF53AB1B42816E4EC679BC26A0BA678F9ABF27466D901629B6B589046D2708EA22|+3994
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 08:40.42
telegram=|49449344652846511A0778_0DFF5F350082350035C10A08E444744869C8A5AC50F00A417E05307FB5999FFBA0587AA4425129FB26149D1500511C880E98DDA1194EEFF805046D2908EA22|+4111
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 08:42.39
telegram=|49449344652846511A0778_0DFF5F3500823500351D4AEA892A7FAF6E954F3C1F2B9F25BB9B52ACCC03135C88BA162809F1CA787C0E5539C1EF888536E03BFAFD29115250046D2D08EA22|+4335
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 08:46.23
telegram=|49449344652846511A0778_0DFF5F3500823600351F3282A382338BAA00F70A2A7E944E13B2387665504E75468A5C38273131B2A368667CE8CDD4C276C2AFA20000D68F0B046D3108EA22|+4566
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 08:50.14
telegram=|49449344652846511A0778_0DFF5F35008236003586AF064081EF724C29232253B6E8C91ECFBEA17AD13BEACE9E8488C96DBFF51247B39810AFD445F2A281B01BCD7849E9046D3508EA22|+4790
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 08:53.58
telegram=|49449344652846511A0778_0DFF5F35008237003520AA8EDEC45EC5D9D7DFD3F3A7EBD509A5489D4A1D44BB4FDB1A2D17321F5DB439585C8E722A28FE3D0A83D4E585061F046D0209EA22|+5354
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:03.22
telegram=|49449344652846511A0778_0DFF5F3500823800350CF0841F71AE6561ADADC45E6ABBB4E9CD232B441F2ABCDBB46A35B50AB21A97B93325A77159AED2267F1C06F0607963046D0809EA22|+5694
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:09.02
telegram=|49449344652846511A0778_0DFF5F35008238003556FABDE7D56178FB4AEAB9234299C4131DFA705BEB185B74D05748EA554B340A59852F8212D9048E3720AA5F86FA06B2046D0A09EA22|+5804
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:10.52
telegram=|49449344652846511A0778_0DFF5F350082390035527E2E4AC251B88DEAC425FA57BF9358135B90923CBBDDCF8DF7C6CF22A9E3E3B4C0A58802499DCAA28F6B91238A6C88046D0F09EA22|+6144
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:16.32
telegram=|49449344652846511A0778_0DFF5F3500823A0035E2F41F3580B26070D856C018363E486E4F4F5A2AA3138ECE350259BB5F1D3D31707CBCF97271F30AB38549A59C67B009046D1309EA22|+6373
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:20.21
telegram=|49449344652846511A0778_0DFF5F3500823B0035D3A6002BEE05FACE01C173A455F853F944E51BED9D2DED0218B90FCB000633F963A8CD07CEDF5FDF42B0FE4ADE85F016046D1B09EA22|+6822
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:27.50
telegram=|49449344652846511A0778_0DFF5F3500823B00356F3B56348DC35B7658FE7AD68F3D8B389992F704C6CF2B8A69C0AE4FC55B679376BE9C2567C01211130139CED1A250E7046D1C09EA22|+6936
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:29.44
telegram=|49449344652846511A0778_0DFF5F3500823B003579F847F6178019B21326D2F2FAFCFD72857BAE508AC7DD6CF4C11B0D1CAD20BE0DBC3D784FEA8FAE51C58F4DFC3BE833046D1E09EA22|+7047
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:31.35
telegram=|49449344652846511A0778_0DFF5F3500823C00356E4F2D7AE754836D37FF7C363956D3344DD799F27B42895CE5C62FE695FF97697D0C390017B25A5BD917FC1515BD85C6046D2209EA22|+7272
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:35.20
telegram=|49449344652846511A0778_0DFF5F3500823C003536DA85324F7E47D75C20B9B4E72B6BF511FE46A211E31E15968B0E17CF22ECA8B5E20A810095A3C644990D837C197AA2046D2609EA22|+7496
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:39.04
telegram=|49449344652846511A0778_0DFF5F3500823C0035F206C6DF3B791B6F8A25A83A213217BCB36A7702E59BABAC0125278731A52CEB46A694DF396493DF5EEA8415D59B81E9046D2809EA22|+7610
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:40.58
telegram=|49449344652846511A0778_0DFF5F3500823E0035BB266D3B5EB3CAD3A8CDBD3B1258EA37EAA20878D349BEF4B0C30B3733984263063CE57373E7ACADD3302AA7301FFB2C046D3509EA22|+8397
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 09:54.05
[... cut ...]
telegram=|49449344652846511A0778_0DFF5F35008260003508AD1F1F99D63320E8E235D79EA70EF682FA5DA17B5C095EAC623094E8E098A3FEF891E2438268560A4D1C2B1F4E934E046D080EEA22|+23695
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 14:09.03
telegram=|49449344652846511A0778_0DFF5F350082610035AE6A130B8A8CF07C0C6F9EA35C8C5274671347D73DA9810CD664F2F9616388CE7B4835BD06D7E2253741F2667DC5D8C4046D0A0EEA22|+23801
watermeter      51462865        null m³ null m³ null    OK      2023-02-10 14:10.49

Hope this helps...
Thanks a lot for your efforts!

weetmuts, 2023-05-13:
Thanks! There is indeed an increasing value just at the start of the manufacturer data.
I have pushed a test driver wme5 that for your telegrams prints this:

wmbusmeters --selectfields=total_m3,total_hex,meter_datetime simulation_qundis.txt Wat wme5 51462865 NOKEY

8530944 m³	00822C00	2023-02-10 07:35
8530944 m³	00822C00	2023-02-10 07:37
8532736 m³	00823300	2023-02-10 08:30
8532992 m³	00823400	2023-02-10 08:38
8532992 m³	00823400	2023-02-10 08:39
8533248 m³	00823500	2023-02-10 08:41
8533248 m³	00823500	2023-02-10 08:45
8533504 m³	00823600	2023-02-10 08:49
8533504 m³	00823600	2023-02-10 08:53
8533760 m³	00823700	2023-02-10 09:02
8534016 m³	00823800	2023-02-10 09:08
8534016 m³	00823800	2023-02-10 09:10
8534272 m³	00823900	2023-02-10 09:15
8534528 m³	00823A00	2023-02-10 09:19
8534784 m³	00823B00	2023-02-10 09:27
8534784 m³	00823B00	2023-02-10 09:28
8534784 m³	00823B00	2023-02-10 09:30
8535040 m³	00823C00	2023-02-10 09:34
8535040 m³	00823C00	2023-02-10 09:38
8535040 m³	00823C00	2023-02-10 09:40
8535552 m³	00823E00	2023-02-10 09:53
8535552 m³	00823E00	2023-02-10 09:55
8535808 m³	00823F00	2023-02-10 09:57
8535808 m³	00823F00	2023-02-10 09:58
8535808 m³	00823F00	2023-02-10 10:02
558591 m³	000885FF	2023-02-10 10:02
8536064 m³	00824000	2023-02-10 10:08
8536320 m³	00824100	2023-02-10 10:13
8536576 m³	00824200	2023-02-10 10:21
8536832 m³	00824300	2023-02-10 10:28
8536832 m³	00824300	2023-02-10 10:30
8537088 m³	00824400	2023-02-10 10:34
8537088 m³	00824400	2023-02-10 10:38
8537344 m³	00824500	2023-02-10 10:45
8537344 m³	00824500	2023-02-10 10:47
8537600 m³	00824600	2023-02-10 10:49
8537856 m³	00824700	2023-02-10 11:00
8537856 m³	00824700	2023-02-10 11:02
8538112 m³	00824800	2023-02-10 11:06
8538112 m³	00824800	2023-02-10 11:08
8538112 m³	00824800	2023-02-10 11:10
8538368 m³	00824900	2023-02-10 11:13
8538368 m³	00824900	2023-02-10 11:15
8538368 m³	00824900	2023-02-10 11:17
8538624 m³	00824A00	2023-02-10 11:19
8538624 m³	00824A00	2023-02-10 11:23
8538624 m³	00824A00	2023-02-10 11:25
8538880 m³	00824B00	2023-02-10 11:26
8538880 m³	00824B00	2023-02-10 11:30
8539136 m³	00824C00	2023-02-10 11:34
8539136 m³	00824C00	2023-02-10 11:36
8539136 m³	00824C00	2023-02-10 11:38
8539136 m³	00824C00	2023-02-10 11:39
8539392 m³	00824D00	2023-02-10 11:41
8539392 m³	00824D00	2023-02-10 11:43
8539648 m³	00824E00	2023-02-10 11:49
8539648 m³	00824E00	2023-02-10 11:51
8539648 m³	00824E00	2023-02-10 11:54
8539904 m³	00824F00	2023-02-10 11:56
8539904 m³	00824F00	2023-02-10 11:58
8539904 m³	00824F00	2023-02-10 12:02
8540160 m³	00825000	2023-02-10 12:07
8540416 m³	00825100	2023-02-10 12:13
8540416 m³	00825100	2023-02-10 12:15
8540416 m³	00825100	2023-02-10 12:17
8540672 m³	00825200	2023-02-10 12:24
8540928 m³	00825300	2023-02-10 12:26
8540928 m³	00825300	2023-02-10 12:28
8540928 m³	00825300	2023-02-10 12:30
8540928 m³	00825300	2023-02-10 12:32
8541184 m³	00825400	2023-02-10 12:33
8541184 m³	00825400	2023-02-10 12:37
8541184 m³	00825400	2023-02-10 12:39
8541440 m³	00825500	2023-02-10 12:45
8541440 m³	00825500	2023-02-10 12:46
8541696 m³	00825600	2023-02-10 12:50
8541696 m³	00825600	2023-02-10 12:52
8541952 m³	00825700	2023-02-10 12:56
8541952 m³	00825700	2023-02-10 12:57
8541952 m³	00825700	2023-02-10 12:59
8541952 m³	00825700	2023-02-10 13:01
8542208 m³	00825800	2023-02-10 13:03
8542208 m³	00825800	2023-02-10 13:05
8542208 m³	00825800	2023-02-10 13:07
8542208 m³	00825800	2023-02-10 13:09
8542464 m³	00825900	2023-02-10 13:10
8542464 m³	00825900	2023-02-10 13:12
8542464 m³	00825900	2023-02-10 13:16
8542720 m³	00825A00	2023-02-10 13:18
8542720 m³	00825A00	2023-02-10 13:22
8542720 m³	00825A00	2023-02-10 13:23
8542976 m³	00825B00	2023-02-10 13:25
8542976 m³	00825B00	2023-02-10 13:27
8543488 m³	00825D00	2023-02-10 13:44
8543744 m³	00825E00	2023-02-10 13:49
8544256 m³	00826000	2023-02-10 14:06
8544256 m³	00826000	2023-02-10 14:08
8544512 m³	00826100	2023-02-10 14:10

weetmuts, 2023-05-13:
Can you please do some tests to see if the m3 value somehow correlates to your consumption?
The hex is the hex bytes that I use for this calculation.
The meter_datetime is from the telegram itself.

weetmuts, 2023-05-13:
There is a glitch telegram in the middle which contains some other information.

FrankHdf, 2023-05-13:
Thanks a lot, I will test this driver 😊

closed 2023-08-13 by github-actions[bot]