wmbusmeters/wmbusmeters #826: calculate_current_month_total ?

zibous, 2023-01-27:
### Type of request

Question

### OS version

ubuntu 22.10

### wmbusmeters version

docker latest

### Your message goes here

What am I doing wrong?? 

The additional fields are output in the payload, but the calculations are not?

```
name=watermeter
type=izar
id=*********
key=
pollinterval=60
field_address=Gemeinde
field_city=Hoechst
field_service=docker rb3b
calculate_current_month_total=total_m3-last_month_total_m3
caclulate_current_month_liter=current_month_total*1000
```
**payload:**

```
{"media":"water",
"meter":"izar",
"name":"watermeter",
"id":"*******",
"total_m3":448.781,
"last_month_total_m3":438.675,
"last_month_measure_date":"2023-01-01",
"remaining_battery_life_y":10.5,
"current_alarms":"no_alarm",
"previous_alarms":"no_alarm",
"transmit_period_s":8,
"timestamp":"2023-01-27T12:47:01Z",
"device":"cul",
"rssi_dbm":-55,
"address":"Gemeinde",
"city":"Hoechst",
"service":"docker rb3b"
}
```

weetmuts, 2023-01-27:
The suffix for litre is just _l

wmbusmeters --listunits

weetmuts, 2023-01-27:
Try
calculate_current_month_total_m3=total_m3-last_month_total_m3

zibous, 2023-01-27:
@weetmuts 
Thanks, changed and docker container restartet, but no changes:

```config
ame=watermeter
type=izar
id=********
key=
pollinterval=60
field_address=Gemeinde
field_city=Hoechst
field_service=docker rb3b
calculate_current_month_total_m3=total_m3-last_month_total_m3

```

```json
{
  "media": "water",
  "meter": "izar",
  "name": "watermeter",
  "id": "18444343",
  "total_m3": 448.797,
  "last_month_total_m3": 438.675,
  "last_month_measure_date": "2023-01-01",
  "remaining_battery_life_y": 10.5,
  "current_alarms": "no_alarm",
  "previous_alarms": "no_alarm",
  "transmit_period_s": 8,
  "current_month_total_m3": 10.122,
  "timestamp": "2023-01-27T14:37:51Z",
  "device": "cul",
  "rssi_dbm": -56,
  "address": "Gemeinde",
  "city": "Hoechst",
  "service": "docker rb3b"
}
```

weetmuts, 2023-01-27:
Please post a decrypted telegram here using --logtelegrams

weetmuts, 2023-01-27:
I can see the calculated value in the json....

"current_month_total_m3": 10.122,

zibous, 2023-01-27:
```
1944a511780743434418a241150013ce5c5c324c2dBd48eef6cB

Auto driver  : izar
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 43434418 dll-id (18444343)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 41150013CE5C5C324C2DBD48EEF6CB mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"*****",
    "total_m3":448.798,
    "last_month_total_m3":438.675,
    "last_month_measure_date":"2023-01-01",
    "remaining_battery_life_y":10.5,
    "current_alarms":"no_alarm",
    "previous_alarms":"no_alarm",
    "transmit_period_s":8,
    "timestamp":"2023-01-27T14:55:40Z"
}

Using: wmbusmeters: 1.11.0-60-g89c42ab
89c42ab27dc8e7ac566e8b3a7810243a3bebd84b
```

zibous, 2023-01-27:
>"current_month_total_m3": 10.122

Yes,  think I need glasses... Thank you

closed 2023-01-27 by zibous