wmbusmeters/wmbusmeters #827: Itron Gasmeter with Cyble wMBus-radio

KFTH1, 2023-01-28:
### Type of request

Enhancement

### HAOS version

Home Assistant 2023.1.7

### wmbusmeters version

Current version: 1.11.0

### Your message goes here

Hello! Thank you for the possibility to drop my problem here :)

I managed to get wmbusmeters on Home Assitant up and running.

I also can receive telegrams from my gas-meter, but I don´t know, what driver I have to choose! It says "driver unknown"...
I have tried different driver-settings and also auto - but it does not work. 
Is there any driver I can use for this specific radio?
I searched already many hours, but could not find a hint that worked for me 
(And I don´t know how to make a driver-file on my own.... ;-( )

Here a part of the logfile. Can you help me with this?

Started auto amb8465[00000000] on /dev/ttyUSB0 listening on t1
(meter) MainGas: meter detection did not match the selected driver apator162! correct driver is: unknown!
(meter) Not printing this warning again for id: 12000322 mfct: (ITW) ITRON Water, France (0x2697) type: Gas meter (0x03) ver: 0x1e
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(apator162) telegram contains a register (0c) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 46449726220300121E03724089932597261E038B4830052F2F066D0E2B0BFC21000C943A734142000C943C000000000B3C0200004C943A25634100426CE12103FD971C0100002F

[image]

BIBOLV, 2023-01-28:
Hi, have you tried unismart?
https://wmbusmeters.org/analyze/46449726220300121E03724089932597261E038B4830052F2F066D0E2B0BFC21000C943A734142000C943C000000000B3C0200004C943A25634100426CE12103FD971C0100002F

KFTH1, 2023-01-28:
Hello! Thank you for your fast reply! 

No I have not known, that a driver with this name existed...  
I will try to find it! 
(or can you tell me, where to get the unismat-driver-file so I can copy it to the home assistant?)

thanks!
br
andi

KFTH1, 2023-01-28:
Hello again!
unismart seems not to be installed as standard in my home assistant. When I use "unismart" in the meter-config-file, it says "File /config/wmbusmeters/etc/mqtt_discovery/unismart.json not found."

BIBOLV, 2023-01-28:
json file is for meter template in HA that would allow automatic discovery, but it does not mean that your meter telegrams are not parsed. Check logs and check mqtt - data should be there.

KFTH1, 2023-01-29:
Oh! ok... Thank you! 
You are right. somehow it works. I get the values in HomeAssitant now. I think I was distracted by the error-messasges in the logfile...  
thank you for you help.

closed 2023-01-29 by BIBOLV

KFTH1, 2024-03-14:
Hello Matthew! I would be happy if I can help you with that. but I am 
not sure

which file or informations you exactly need. It has been a long time 
ago, when I solved the gas-metering-problem...

not sure if I remember everything, where to find in HA :)

I have checked the mqtt_discovery area in my HA Configs, but there is no 
unismart.json.
everything works without adding unismart.json. I guess thats a 
standard-feature where you  do not need an own

json file.

thats my meter-0001-config:

driver=unismart
id=XXXXXXXXXX
key=YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
name=MainGas

And these are the json files, but I don´t address them in the config:

does that help you?

br

andi

> Hey @KFTH1 [1] can you share the unismart.json file you've found?
> 
> [3].

Links:
------
[1] https://github.com/KFTH1
[2] 
https://github.com/wmbusmeters/wmbusmeters/issues/827#issuecomment-1992003789
[3] 
https://github.com/
--=_85e9b60caac59ee92d25cc42c4cd6789
Content-Type: multipart/related;
 boundary="=_b83014fb7d2e59530a7279ee0b39c042"

--=_b83014fb7d2e59530a7279ee0b39c042
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hello Matthew! I would be happy if I can help you with that. but I am no=
t sure</p>
<p>which file or informations you exactly need. It has been a long time ago=
, when I solved the gas-metering-problem...</p>
<p>not sure if I remember everything, where to find in HA :)</p>
<p><br /></p>
<p>I have checked the mqtt_discovery area in my HA Configs, but there is no=
 unismart.json.<br />everything works without adding unismart.json. I guess=
 thats a standard-feature where you  do not need an own</p>
<p>json file.</p>
<p><br /></p>
<p>thats my meter-0001-config:</p>
<p>driver=3Dunismart<br />id=3DXXXXXXXXXX<br />key=3DYYYYYYYYYYYYYYYYYYYYYY=
YYYYYYYYYYYYY<br />name=3DMainGas</p>
<p><br /></p>
<p><br /></p>
<p>And these are the json files, but I don&acute;t address them in the conf=
ig:</p>
[... 424 lines cut ...]
a70n8cLVY0nQKsxwPQjTg/XX3mt89NfTT6H8OIX7yV9PH5lzAMip56AeBw20KdYL1/DYad5xXFh+
9X+nX/7i/0yfwb9ffvjNEbA8JAurkKiIjDUP4tdzryDe1y+nwrPIvSB2PO+BUy8xMyByf6zLXxPL
OPUcgEBYZvod+uhUY3FTsCA4n/3idfqPX811TPAs8V1gaT8/5+HR95QoZNfZFiTX6SBwe+2X06/E
U2TvJjUxyzGp79RzOPi1MvA8f89p5vDxcWGpTMNwSvb9v//h9E//PveAOQBL+3IuTcHitCtYfwtA
PjUL9wVPY6+daRvApcKN9516DqwmYZnpd+gj7fdjt+PCctbDLcEhnYnn5ff81CkKOXqW/OtVGhx7
bTpuBkWAgGncr3Dad+o56CvCMtPv0EdmPI44Pi4ss5+OP53+9eADIgw5HOW+iDx/j9AOzQFwYjRi
T4OTX6tl2G02lTOAnHoutYGwpL6w/X7+8XFhmf1PyZ9Pn3WGRa17nILJlzR934nvHcn7OLhg6qaf
gdtgyd9lUrlucGN9MsU74lzoE8JyPhRzkI0Ly0Eg1uuUuc7i8W33OWEhlI2f47cNihhKwkJYCEuj
BghLY0dxCkbPQlgICz1LowYIS2NH0bPQszAVUmManEvnqGJ94+UbIyyEhXnDGjVAWBo7ipZ+PEt/
6TH5/7/5W+SQrqc1AAAAAElFTkSuQmCC
--=_b83014fb7d2e59530a7279ee0b39c042
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_b83014fb7d2e59530a7279ee0b39c042--

--=_85e9b60caac59ee92d25cc42c4cd6789--