wmbusmeters/wmbusmeters #829: m3  Value with apator172 driver

Kanj1985, 2023-01-29:
First of all thank you for this big work.

This is my first use for this library, and I think the json values are not good for "apator172" driver as you can see for the below scan. 
cou

based on this values; I am not able to identify my device using device "ID" or "m3" value.

I used the below command to limit the scan only on "apator172" driver  and for devices with "0010" as 4 first digits.
>>>>>> wmbusmeters --logtelegrams --format=json rtlwmbus Device apator172 0010* nokey

the results: 

Started config rtlwmbus[00000001] listening on none
telegram=|1C441486077610000411A0_243D16000440030000000005FF0C6B1D1600|+3
{"media":"water","meter":"apator172","name":"Device","id":"00107607","total_m3":485814.7,"timestamp":"2023-01-29T13:26:54Z","device":"rtlwmbus[00000001]","rssi_dbm":58}
telegram=|1C441486F85C10000411A0_BFBB06000040030000000005FF0C4DB10600|+5
{"media":"water","meter":"apator172","name":"Device","id":"00105cf8","total_m3":147093,"timestamp":"2023-01-29T13:26:56Z","device":"rtlwmbus[00000001]","rssi_dbm":20}
telegram=|1C441486896F10000411A0_5A1018000000030000000005FF0CA9E01700|+7
{"media":"water","meter":"apator172","name":"Device","id":"00106f89","total_m3":525683.3,"timestamp":"2023-01-29T13:26:58Z","device":"rtlwmbus[00000001]","rssi_dbm":24}
telegram=|1C441486936E10000411A0_EE0115000000030000000005FF0C00E61400|+11
{"media":"water","meter":"apator172","name":"Device","id":"00106e93","total_m3":458916.7,"timestamp":"2023-01-29T13:27:02Z","device":"rtlwmbus[00000001]","rssi_dbm":19}
telegram=|1C441486736010000411A0_FE1A1B000440030000000005FF0C84AB1A00|+15
{"media":"water","meter":"apator172","name":"Device","id":"00106073","total_m3":592127.3,"timestamp":"2023-01-29T13:27:06Z","device":"rtlwmbus[00000001]","rssi_dbm":17}
telegram=|1C441486486D10000411A0_7BC710000000030000000005FF0C8DAE1000|+15
{"media":"water","meter":"apator172","name":"Device","id":"00106d48","total_m3":366547.7,"timestamp":"2023-01-29T13:27:06Z","device":"rtlwmbus[00000001]","rssi_dbm":22}
telegram=|1C4414866C6910000411A0_73E218000440030000000005FF0C1E981800|+17
{"media":"water","meter":"apator172","name":"Device","id":"0010696c","total_m3":543611.7,"timestamp":"2023-01-29T13:27:08Z","device":"rtlwmbus[00000001]","rssi_dbm":14}
telegram=|1C441486587010000411A0_537E07000840030000000005FF0C41650700|+19
{"media":"water","meter":"apator172","name":"Device","id":"00107058","total_m3":163697,"timestamp":"2023-01-29T13:27:10Z","device":"rtlwmbus[00000001]","rssi_dbm":18}
telegram=|1C441486875E10000411A0_51B408000040030000000005FF0CAB780800|+26
{"media":"water","meter":"apator172","name":"Device","id":"00105e87","total_m3":190149.7,"timestamp":"2023-01-29T13:27:17Z","device":"rtlwmbus[00000001]","rssi_dbm":25}
telegram=|1C4414860C6510000411A0_451016000000030000000005FF0C1AD21500|+37
{"media":"water","meter":"apator172","name":"Device","id":"0010650c","total_m3":481985.7,"timestamp":"2023-01-29T13:27:28Z","device":"rtlwmbus[00000001]","rssi_dbm":24}
telegram=|1C441486077610000411A0_243D16000440030000000005FF0C6B1D1600|+57
{"media":"water","meter":"apator172","name":"Device","id":"00107607","total_m3":485814.7,"timestamp":"2023-01-29T13:27:48Z","device":"rtlwmbus[00000001]","rssi_dbm":63}
telegram=|1C4414865F5E10000411A0_2AD915000040030000000005FF0C998B1500|+70
{"media":"water","meter":"apator172","name":"Device","id":"00105e5f","total_m3":477283.3,"timestamp":"2023-01-29T13:28:01Z","device":"rtlwmbus[00000001]","rssi_dbm":11}
telegram=|1C441486896F10000411A0_5A1018000000030000000005FF0CA9E01700|+71
{"media":"water","meter":"apator172","name":"Device","id":"00106f89","total_m3":525683.3,"timestamp":"2023-01-29T13:28:02Z","device":"rtlwmbus[00000001]","rssi_dbm":32}
telegram=|1C4414860C6510000411A0_441016000000030000000005FF0C1AD21500|+79
{"media":"water","meter":"apator172","name":"Device","id":"0010650c","total_m3":481985.3,"timestamp":"2023-01-29T13:28:10Z","device":"rtlwmbus[00000001]","rssi_dbm":25}
telegram=|1C441486F85C10000411A0_BFBB06000040030000000005FF0C4DB10600|+85
{"media":"water","meter":"apator172","name":"Device","id":"00105cf8","total_m3":147093,"timestamp":"2023-01-29T13:28:16Z","device":"rtlwmbus[00000001]","rssi_dbm":27}
telegram=|1C441486486D10000411A0_7BC710000000030000000005FF0C8DAE1000|+87
{"media":"water","meter":"apator172","name":"Device","id":"00106d48","total_m3":366547.7,"timestamp":"2023-01-29T13:28:18Z","device":"rtlwmbus[00000001]","rssi_dbm":27}
telegram=|1C441486936E10000411A0_EF0115000000030000000005FF0C00E61400|+87
{"media":"water","meter":"apator172","name":"Device","id":"00106e93","total_m3":458917,"timestamp":"2023-01-29T13:28:18Z","device":"rtlwmbus[00000001]","rssi_dbm":23}
telegram=|1C441486C17310000411A0_473E19000040030000000005FF0C5BFD1800|+99
{"media":"water","meter":"apator172","name":"Device","id":"001073c1","total_m3":551447.7,"timestamp":"2023-01-29T13:28:30Z","device":"rtlwmbus[00000001]","rssi_dbm":35}
telegram=|1C441486077610000411A0_243D16000440030000000005FF0C6B1D1600|+103
{"media":"water","meter":"apator172","name":"Device","id":"00107607","total_m3":485814.7,"timestamp":"2023-01-29T13:28:34Z","device":"rtlwmbus[00000001]","rssi_dbm":60}
telegram=|1C441486936E10000411A0_EE0115000000030000000005FF0C00E61400|+104
{"media":"water","meter":"apator172","name":"Device","id":"00106e93","total_m3":458916.7,"timestamp":"2023-01-29T13:28:35Z","device":"rtlwmbus[00000001]","rssi_dbm":25}
telegram=|1C441486896F10000411A0_5A1018000000030000000005FF0CA9E01700|+106
{"media":"water","meter":"apator172","name":"Device","id":"00106f89","total_m3":525683.3,"timestamp":"2023-01-29T13:28:37Z","device":"rtlwmbus[00000001]","rssi_dbm":31}
telegram=|1C4414862F7410000411A0_5A710A000000030000000005FF0C8A4F0A00|+120
{"media":"water","meter":"apator172","name":"Device","id":"0010742f","total_m3":228126,"timestamp":"2023-01-29T13:28:51Z","device":"rtlwmbus[00000001]","rssi_dbm":15}
telegram=|1C4414866C6910000411A0_73E218000440030000000005FF0C1E981800|+131
{"media":"water","meter":"apator172","name":"Device","id":"0010696c","total_m3":543611.7,"timestamp":"2023-01-29T13:29:02Z","device":"rtlwmbus[00000001]","rssi_dbm":30}

[image]

Thank you for your help.
regards

BIBOLV, 2023-01-29:
Did you check all sides of ista module for some more details? Is there meter ID printed somewhere - it would be easier to identify your meter with it.

Kanj1985, 2023-01-30:
No more information accessible ( at least for my eye).
I just asked ISTA supplier more infotmation like token or something else can help me to identify my water meter.

I will share this information when i have it.
Thank you for your help.

Kanj1985, 2023-02-01:
hello, 
Sorry for this delay. for now no response from ISTA.
I just found the below reference on the warm water meter (also ISTA meters)

[image]

could this one help us ? 

Thank you in advance.

BIBOLV, 2023-02-01:
Hi,
Device ID is 8 symbols long so non of those numbers match. What is in QR code - same numbers?
Also what was your rationale to capture 0100* ID`s previously? Maybe it is worth to capture all telegrams and look at results. Also you could use apator08 driver that gives 3 decimal position lower readings from few your examples that I checked.
And lastly - not related, but more a rant from my side - I really hope that those meters you caught are not imst ones as apator is PL company that makes meters with proprietary message formats and if those would make further in field also under different manufacturers it would be pity.

Schoumi, 2023-02-08:
Hi @Kanj1985,

On my ista meter the datamatrix correspond to the part and serial printed on it.
In my case part: 19399 and serial is 9 digits (ex: 310020071).
Part Number identify the product from ista catalog.

On your photo if you scan datamatrix on them you may have 165002110838999 for the left meter of something like that no? Part: 16500 serial: 110838999

Sorry it's a french version (i didn't find english one with same explanation): https://www.ista.com/fr/eau/compteur-eau-individuel/ you have an image showing ista meter and where to find serial.

closed 2023-05-10 by github-actions[bot]

Phyks, 2023-05-19:
Hi,

I have the same water meter from Ista (part 19399 ). Did any of you successfilly fetched data from it through wmbusmeters? Thanks !