wmbusmeters/wmbusmeters #834: Ability to run mqtt_disovery script outside HA Addon

KrzysztofHajdamowicz, 2023-02-02:
### Type of request

Enhancement

### HAOS version

9.x

### wmbusmeters version

latest

### Your message goes here

Thank You for your great software!

Due to distance between my meters and HAOS server, I decided to run dockerized wmbusmeters on stand-alone Raspberry Pi located near meters, this sends meter readings via MQTT to HA.  
I see there is autodiscovery script running in HA addon, is there an option to run it inside `weetmuts/wmbusmeters` docker image?

Benefit of such possibility is making meters reading automatically available for Home Asistant even without running it as HA addon.

BIBOLV, 2023-02-02:
HA autodiscovery relays on [templates](https://github.com/weetmuts/wmbusmeters/tree/master/ha-addon/mqtt_discovery) and currently small portion of existing drivers have been converted. First priority would be to implement it fully in ha-addon and when it is done we can also implement those templates and capability to use HA autodiscovery in docker and other non-ha versions of wmbusmeters. 
IMO this issue is duplicate of https://github.com/weetmuts/wmbusmeters/issues/610 so let`s close this and move discussions and tracking there.

KrzysztofHajdamowicz, 2023-02-03:
Sure, let's move there.

closed 2023-02-03 by KrzysztofHajdamowicz