wmbusmeters/wmbusmeters #837: Addon configuration expects meters as string, instead of JSON

KrzysztofHajdamowicz, 2023-02-03:
### Type of request

Bug

### HAOS version

9.5

### wmbusmeters version

1.11.0

### Your message goes here

Freshly installed wmbusmeters addon, `mqtt` section looks like proper code input field, but `conf` and `meters` sections expect strings.  
`conf` has to be prepared in external editor, with `\n` at end of each line, then compressing it to single line to paste it here.  
What's worse, `meters` is a drop-down, not input.  
This renders configuring wmbusmeters not possible.
[image]

KrzysztofHajdamowicz, 2023-02-03:
Quick look at Firefox Developer tools, mqtt section is declared as `<ha-code-editor mode="yaml" />`, where conf section was declared as `<input class="mdc-text-field__input" type="text" />`

KrzysztofHajdamowicz, 2023-02-03:
I had some spare time and solved this via #838 .
Please, can you merge it with upstream?

BIBOLV, 2023-02-24:
PR merged. Tnx!

closed 2023-02-24 by BIBOLV